evcc-io/evcc #10752: KIA zeigt Position und Kilometerstand aber keinen SOC

schenlap, 2023-11-12:
### Describe the bug

Bei meinem KIA EV6 wird zwar der Kilometerstand und die Position angezeigt, nicht aber der SOC.
Issue wurde auf Wunsch in https://github.com/evcc-io/evcc/discussions/10501 erzeugt.

### Steps to reproduce

1. KIA EV6 anstecken
2. `evcc vehicle` starten
3.
```
..................Soc: i/o timeout
Capacity:              80.0kWh
Charge status:         must retry
Range:                 must retry
Odometer:              28383km  --> korrekt und aktuell
Finish time:           must retry
Position:              4x.xx4047,1x.xx8072  --> korrekt und aktuell
Target Soc:            must retry
OnIdentified:          Mode:pv, MinSoc_:15, TargetSoc:80
Features:              []
```
Der HV-Akku sollte 82% haben. "evStatus":{"batteryCharge":false,"batteryStatus":82,"batteryPlugin":0, -> OK

Kann es vielleicht ein Problem machen dass vehicleLocation.time"20231026094249 oder vehicleStatus.time:20231026095624 vor der Uhrzeit von evcc liegen? TRACE 2023/10/26 09:03:33 ? Das hst vermutlich was mit Zeitzonen zu tun. Wie alt dürften die Werte maximal sein dann nicht nochmals gepollt wird?

### Configuration details

```yaml
aus tempalte:

vehicles:
  - name: EV6
    type: template
    template: kia
    user: xxxx
    password: # xxx
    vin: xxx
```

### Log details

```text
Das komplette log kann hier nicht hochgeladen werden wegen der Beschränkung auf 65536 Zeichen. Ich habe es unter https://pastebin.com/wXvyjV79 abgelegt.

Geküzte Version (einige Retries entfernt):
 evcc version 0.121.1

[main  ] FATAL 2023/10/26 09:03:17 kia not found
root@beaglebone:/data/evcc# ./evcc vehicle --log trace
[main  ] INFO 2023/10/26 09:03:29 evcc 0.121.1
[main  ] INFO 2023/10/26 09:03:29 using config file: /data/evcc/evcc.yaml
[kia   ] TRACE 2023/10/26 09:03:31 POST https://prd.eu-ccapi.kia.com:8080/api/v1/spa/notifications/register
[kia   ] TRACE 2023/10/26 09:03:31 {"pushRegId":"XXXXX91AB7CCFA733FF5BC89312F7B50AFFB1E65C93A1831D9EAA8102856XXXX","pushType":"APNS","uuid":"e0c053fc-adc1-491d-8b8f-5cf75b02a20e"}
--
{"retCode":"S","resCode":"0000","resMsg":{"deviceId":"xxxx6048-66fb-411a-83e9-8251c37dxxxx"},"msgId":"xxxx9e3c-c9b8-4960-9b43-b3426ba7xxxx"}
[kia   ] TRACE 2023/10/26 09:03:31 GET https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/authorize?response_type=code&state=test&client_id=xxxx5c00-0a2f-4c64-bcb4-2cfb1500xxxx&redirect_uri=https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/redirect
[kia   ] TRACE 2023/10/26 09:03:32 <a href="https://prd.eu-ccapi.kia.com:8080/web/v1/user/authorize?cache=reset">Found</a>.
[kia   ] TRACE 2023/10/26 09:03:32 GET https://prd.eu-ccapi.kia.com:8080/web/v1/user/authorize?cache=reset
[kia   ] TRACE 2023/10/26 09:03:32 <!doctype html><html lang="en"><head><meta charset="utf-8"><meta name="viewport" content="width=device-width,initial-scale=1,shrink-to-fit=no,maximum-scale=1,user-scalable=0"><meta name="theme-color" content="#ffffff"><link href="https://fonts.googleapis.com/css?family=Nanum+Gothic:400,700,800&amp;subset=korean" rel="stylesheet"><link href="https://fonts.googleapis.com/icon?family=Material+Icons" rel="stylesheet"><link rel="manifest" href="/web/v1/user/manifest.json"><link rel="icon" href="/web/v1/user/favicon_kia.ico" type="image/x-icon"><!--[if lt IE 10]><script src="/web/v1/user/fetch.js"></script><![endif]--><title></title><link href="/web/v1/user/static/css/main.6be91679.chunk.css" rel="stylesheet"></head><body><noscript>You need to enable JavaScript to run this app.</noscript><div id="root"></div><script>!function(f){function e(e){for(var o,r,t=e[0],n=e[1],a=e[2],c=0,l=[];c<t.length;c++)r=t[c],i[r]&&l.push(i[r][0]),i[r]=0;for(o in n)Object.prototype.hasOwnProperty.call(n,o)&&(f[o]=n[o]);for(p&&p(e);l.length;)l.shift()();return d.push.apply(d,a||[]),u()}function u(){for(var e,o=0;o<d.length;o++){for(var r=d[o],t=!0,n=1;n<r.length;n++){var a=r[n];0!==i[a]&&(t=!1)}t&&(d.splice(o--,1),e=s(s.s=r[0]))}return e}var r={},i={34:0},d=[];function s(e){if(r[e])return r[e].exports;var o=r[e]={i:e,l:!1,exports:{}};return f[e].call(o.exports,o,o.exports,s),o.l=!0,o.exports}s.e=function(a){var e=[],r=i[a];if(0!==r)if(r)e.push(r[2]);else{var o=new Promise(function(e,o){r=i[a]=[e,o]});e.push(r[2]=o);var t,c=document.createElement("script");c.charset="utf-8",c.timeout=120,s.nc&&c.setAttribute("nonce",s.nc),c.src=s.p+"static/js/"+({0:"Locale0",1:"Locale1",2:"Locale10",3:"Locale11",4:"Locale12",5:"Locale13",6:"Locale14",7:"Locale15",8:"Locale16",9:"Locale17",10:"Locale18",11:"Locale19",12:"Locale2",13:"Locale20",14:"Locale21",15:"Locale22",16:"Locale23",17:"Locale24",18:"Locale25",19:"Locale26",20:"Locale27",21:"Locale28",22:"Locale29",23:"Locale3",24:"Locale4",25:"Locale5",26:"Locale6",27:"Locale7",28:"Locale8",29:"Locale9",30:"agreement_cert",31:"agreement_genesis",33:"resident_registration"}[a]||a)+"."+{0:"6069d477",1:"23f6bf0c",2:"d1cd04d2",3:"ae196e4c",4:"d5d266aa",5:"67c7f8fe",6:"d78a73cc",7:"7d132977",8:"1a623d1c",9:"9bb00fa6",10:"ff6a9f5a",11:"353d6178",12:"f918e71d",13:"653161cd",14:"6a13f103",15:"7425ce6c",16:"e73c8442",17:"4fd5857e",18:"11b19179",19:"65d7ceb2",20:"025a01a6",21:"620e141a",22:"ff6c3d4a",23:"6e838d5d",24:"e2e978df",25:"8065f6cb",26:"74dee2bf",27:"d25b445d",28:"95a5dc96",29:"fac04a97",30:"1388835f",31:"dd055c85",33:"74f07063"}[a]+".chunk.js",t=function(e){c.onerror=c.onload=null,clearTimeout(l);var o=i[a];if(0!==o){if(o){var r=e&&("load"===e.type?"missing":e.type),t=e&&e.target&&e.target.src,n=new Error("Loading chunk "+a+" failed.\n("+r+": "+t+")");n.type=r,n.request=t,o[1](n)}i[a]=void 0}};var l=setTimeout(function(){t({type:"timeout",target:c})},12e4);c.onerror=c.onload=t,document.head.appendChild(c)}return Promise.all(e)},s.m=f,s.c=r,s.d=function(e,o,r){s.o(e,o)||Object.defineProperty(e,o,{enumerable:!0,get:r})},s.r=function(e){"undefined"!=typeof Symbol&&Symbol.toStringTag&&Object.defineProperty(e,Symbol.toStringTag,{value:"Module"}),Object.defineProperty(e,"__esModule",{value:!0})},s.t=function(o,e){if(1&e&&(o=s(o)),8&e)return o;if(4&e&&"object"==typeof o&&o&&o.__esModule)return o;var r=Object.create(null);if(s.r(r),Object.defineProperty(r,"default",{enumerable:!0,value:o}),2&e&&"string"!=typeof o)for(var t in o)s.d(r,t,function(e){return o[e]}.bind(null,t));return r},s.n=function(e){var o=e&&e.__esModule?function(){return e.default}:function(){return e};return s.d(o,"a",o),o},s.o=function(e,o){return Object.prototype.hasOwnProperty.call(e,o)},s.p="/web/v1/user/",s.oe=function(e){throw console.error(e),e};var o=window.webpackJsonp=window.webpackJsonp||[],t=o.push.bind(o);o.push=e,o=o.slice();for(var n=0;n<o.length;n++)e(o[n]);var p=t;u()}([])</script><script src="/web/v1/user/static/js/35.dae7275d.chunk.js"></script><script src="/web/v1/user/static/js/main.7c56340a.chunk.js"></script></body></html>
[kia   ] TRACE 2023/10/26 09:03:32 POST https://prd.eu-ccapi.kia.com:8080/api/v1/user/language
[kia   ] TRACE 2023/10/26 09:03:32 {"lang":"en"}
[kia   ] TRACE 2023/10/26 09:03:32 GET https://prd.eu-ccapi.kia.com:8080/api/v1/user/integrationinfo
[kia   ] TRACE 2023/10/26 09:03:32 {"userId":"xxxxe948-a128-49a3-89e5-b865dabbxxxx","serviceId":"xxxx5c00-0a2f-4c64-bcb4-2cfb1500xxxx","serviceName":"UVO"}
[kia   ] TRACE 2023/10/26 09:03:32 GET https://eu-account.kia.com/auth/realms/eukiaidm/protocol/openid-connect/auth?client_id=572e0304-5f8d-4b4c-9dd5-41aa84eed160&scope=openid+profile+email+phone&response_type=code&hkid_session_reset=true&redirect_uri=https://prd.eu-ccapi.kia.com:8080/api/v1/user/integration/redirect/login&ui_locales=en&state=xxxx5c00-0a2f-4c64-bcb4-2cfb1500xxxx:xxxxe948-a128-49a3-89e5-b865dabbxxxx
[kia   ] TRACE 2023/10/26 09:03:32 <!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN"  "http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd">
    <html xmlns="http://www.w3.org/1999/xhtml" class="login-pf"
                    lang="en"
    >

[... 178 lines cut ...]
[kia   ] TRACE 2023/10/26 09:04:36 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/5e8256d5-54bf-4518-bc9e-46d99b90xxxx/status/latest
[kia   ] TRACE 2023/10/26 09:04:36 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":4x.xx4047,"lon":1x.xx8072,"alt":0,"type":0},"head":341,"speed":{"value":0,"unit":0},"accuracy":{"hdop":0,"pdop":0},"time":"20231026095843"},"vehicleStatus":{"airCtrlOn":false,"engine":false,"doorLock":false,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"02H","unit":0,"hvacTempType":1 [... cut ...] t":1}}},"msgId":"4099bbea-24c1-4dfc-a57b-55925ad2cfa7"}
..................Soc: i/o timeout
Capacity:              80.0kWh
Charge status:         must retry
Range:                 must retry
Odometer:              28383km
Finish time:           must retry
Position:              4x.xx4047,1x.xx8072
Target Soc:            must retry
OnIdentified:          Mode:pv, MinSoc_:15, TargetSoc:80
Features:              []
```

### What type of operating system are you running?

Linux

### Version

 0.121.1

andig, 2023-11-12:
Beim Odometer akzeptieren wir alte Werte. Beim Soc wollen wir einen aktuellen habe. Laut api:

    "time": "20231026095624",

Das ist steinalt. Daher wird refreshed. Allerdings kommt bei Dir niemals ein aktueller Wert.

Damit bleibt die Frage woran evcc erkennen soll, dass der Wert ok ist? Wie macht die Kia App das?

schenlap, 2023-11-12:
Die Zeit von evcc war von diesem Log `[main  ] INFO 2023/10/26 09:03:29` also ist die api Zeit 53 Minuten in der Zukunft. Ich kann gerne nochmal einen neuren Log liefern. Da dies aber nicht wirklich "mein" Fahrzeug ist sondern nur eine Installation die ich bei jemanden gemacht habe, kann es etwas dauern. Der Soc Wert wird aber aktuell auch noch nicht angezeigt, also wird das Problem das selbe sein.

andig, 2023-11-12:
Schick bitte mal Zugangsdaten an info@evcc.io, alles Andere with zu umständlich.

closed 2023-11-12 by andig