evcc-io/evcc #10754: Virtuelle Wallbox für BMW API

Nekronomekron, 2023-11-12:
Über die BMW Connected Drive API kann die Ladung des Fahrzeuges wie bei einer Wallbox gesteuert werden. Die Ladebegrenzung kann ebenfalls gesteuert werden, von 6-15A. Somit lässt sich mit der BMW API eine einfache virtuelle Wallbox umsetzen, ohne Phasenumschaltung.

Die BMW API wird bereits für die Fahrzeuge verwendet, es müsste „nur“ noch eine entsprechende Wallbox erstellt werden. So könnte man EVCC mit jedem BMW auch ohne separate Wallbox verwenden, rein mit dem beiliegenden Ladegerät.

andig, 2023-11-12:
Das ginge mit der Powerwall zusammen. Sonst fehlt uns die Verbindungsinformation dass das Fahrzeug an genau diesem Ladepunkt hängt. Wenn Du eine hast könnten wir das anschauen, ansonsten wäre das wontfix.

Nekronomekron, 2023-11-12:
Der Ladepunkt wäre das Auto selbst. Es wäre also eine 1:1 Beziehung zwischen Ladepunkt und Fahrzeug, da das Fahrzeug selbst das Ladegerät steuert. Daher wäre es theoretisch unerheblich an welchem Netzteil das Auto physisch steckt, die Ansteuerung des Netzteils erfolgt dann einfach durch das Auto selbst.
So mach ich das aktuell mit einem einfachen Python Script in Homeassisstant. In der API setze ich die Ladegrenze und den Ladezustand, den Test erledigt das Fahrzeug. Dabei ist es unerheblich an welchem „echten“ Ladepunkt das Fahrzeug hängt. Der Ladepunkt ist quasi das Fahrzeug selbst.

andig, 2023-11-12:
> Daher wäre es theoretisch unerheblich an welchem Netzteil das Auto physisch steckt, die Ansteuerung des Netzteils erfolgt dann einfach durch das Auto selbst.

Das ist schon relevant, da es auch ein Ladepunkt mit Überschuss sein muss. Das Szenario "virtuelle Fahrzeuge" wollen wir so nicht unterstützen, daher wontfix.

closed 2023-11-12 by andig