evcc-io/evcc #10756: After upgrade to 0.122.1 EVCC will not start

JHPWBN, 2023-11-12:
As mentioned in the title the add-on does not start. When I open it in the Side Bar I get the message that it is not running.
I can Start it as much as I want, but get the same message again.

However, if I go into my browser and open port 7070 of Homeassistant, I get the EVCC UI, showing me the previous release version.
See screenshot.

Help urgently needed!!
[image]

thizmo, 2023-11-12:
Hi, same problem here.

After upgrade, no start anymore.
This ist the error i got while trying:

`Can't create container from addon_49686a9f_evcc: 400 Client Error for http+docker://localhost/v1.43/containers/create?name=addon_49686a9f_evcc: Bad Request ("invalid mount config for type "bind": bind source path does not exist: /mnt/data/supervisor/addon_configs/49686a9f_evcc")`

Thx in advance.

JHPWBN, 2023-11-12:
Hi Thizmo, how do I get to the addon_config directory?
I installed SAMBA but do not know how to access it from my iMac

thizmo, 2023-11-12:
> Hi Thizmo, how do I get to the addon_config directory? I installed SAMBA but do not know how to access it from my iMac
Hi JHPWBN,

i have the Terminal & SSH enabled/installed on the system. This way i have direct access to the system itself.

HTH.

JHPWBN, 2023-11-12:
Thxs, my add-on_config directory is totally empty

JHPWBN, 2023-11-12:
are you also on HA Core 2023.11.2?

thizmo, 2023-11-12:
> are you also on HA Core 2023.11.2?

Yes i am. Since today. :-)

thizmo, 2023-11-12:
> Thxs, my add-on_config directory is totally empty
> 
> for me this is a major issue, it took me quite some time to get evcc running , decided to become a sponsor and now this

Becoming a sponsor should have nothing to do with getting evcc running. ;-)
Becoming a sponsor means you believe in the idea of developing a great software. Not only for a few month. But for years which is sustainable. 
And with that you get a great community, helping and developing.
Just my thoughts. ;-)

JHPWBN, 2023-11-12:
I am trying to understand the BREAKING CHANGE comment below from the read.me, which is very cryptic to me: U have any idea where I find "container-ID" and "addon's slug" ?
!! BREAKING CHANGE with version 0.122.2 !!

/config directs to addons_config/[container-ID]_evcc (/addon_configs/<your addon's slug>) in Home Assistant. To access this new share in homeassistant you have to install at least verion 11.0.0 of Samba Share Copy your config from /config (in Samba share 11.0.0 /config is named /homeassistant) to addons_config/[container-ID]_evcc

JHPWBN, 2023-11-12:
> Becoming a sponsor should have nothing to do with getting evcc running. ;-) Becoming a sponsor means you believe in the idea of developing a great software. Not only for a few month. But for years [...]

Agree

StefanSchoof, 2023-11-12:
I think you have the same problem as in https://github.com/evcc-io/evcc/discussions/10744