evcc-io/evcc #10764: Battery Control: Solaredge

premultiply, 2023-11-13:
### Discussed in https://github.com/evcc-io/evcc/discussions/8498

<div type='discussions-op-text'>

<sup>Originally posted by **Gifford47** June 16, 2023</sup>
**Is your feature request related to a problem? Please describe.**
I wouldn't like to use my house battery for the ev. Therefore i want to "control" the energy used by the ev. Currently i have no suitable solution to restrict my hems to use the house battery. 

**Describe the solution you'd like**
A possibility to change the hems/battery profile (f.e.: maximize export mode, self consumption, off, ...). I dont know if evcc currently needs this feature, but you could also use it with my smarthome system. F.e.: if evcc has set a target soc and a target time, i could change the battery profile with my smarthome from self consumption to off while loading. I know that the house isn't supported by battery also, but that's the compromise i go with. In my case evcc starts loading in the morning, so that`s not a big deal.
I great option would be to change the battery profile by the evcc api (like all the other values/registers that are currently supported).

**Describe alternatives you've considered**
On googling i found some interesting topics on homeassistant community. For example:

[Setting battery charge/discharge profile via HA integration](https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/36)
[Add storedge control via HA integration](https://github.com/binsentsu/home-assistant-solaredge-modbus/pull/38)

and here is the solaredge [Technical Note – Power Control Protocol for
SolarEdge Inverters](https://www.photovoltaikforum.com/core/attachment/88445-power-control-open-protocol-for-solaredge-inverters-pdf/)
on Page 17 you find the charge/discharge modes from 0-7

**Additional context**

</div>

premultiply, 2023-11-13:
Das SE-API via Modbus ist wirklich einfach und sicher und bietet alle Funktionen die wir brauchen.
Inkl. automatischem Fallback auf einen vorab eingestellten Default-Modus bei Ausfall der Steuerung.

Wäre IMHO vielleicht ein optimaler Testballon.

andig, 2023-11-13:
Wir lösen jetzt bitte erstmal einen einzelnen Case. Dann schauen wir weiter.

closed 2023-11-13 by andig