evcc-io/evcc #10765: Home Battery Control: Kostal Plenticore

deadrabbit87, 2023-11-13:
Deactivates the use of the house battery on the basis developed here https://github.com/evcc-io/evcc/pull/10553. 

Kostal provides the Modbus adress 1034 controlling the charge / discharge power of the battery. 
https://www.kostal-solar-electric.com/de-de/produkte/solar-wechselrichter/-/media/document-library-folder---kse/2020/12/15/13/38/ba_kostal-interface-description-modbus-tcp_sunspec_hybrid.pdf

After a set timeout, the inverter takes over control again:
[image]

deadrabbit87, 2023-11-13:
Okay, hatte ich nicht mitbekommen. Dann würde sich die minimum SoC Adresse dafür eignen.

premultiply, 2023-11-13:
Da die Doku so dünn ist lässt sich nicht genau sagen wie was genau wirkt.
Das müsstest du dann mal in einen PR gießen und ausprobieren.

deadrabbit87, 2023-11-13:
Genauer beschrieben ist das dann in der BA: 
https://www.kostal-solar-electric.com/de-de/download/download/-/media/document-library-folder---kse/2020/12/15/13/14/plenticore-plus_ba_de.pdf

Siehe z.B. Seite 170. Jetzt auch nicht mega detailliert, aber ich hätte gesagt, dass man damit arbeiten kann. 

Alleine kriege ich das noch nicht hin. Ich schau mal wie ihr das in https://github.com/evcc-io/evcc/issues/10764 macht und überlege mir dann was.

andig, 2023-11-13:
Wir lösen jetzt bitte erstmal einen einzelnen Case. Dann schauen wir weiter.

closed 2023-11-13 by andig