evcc-io/evcc #10771: Senec Systems APP Api

oakdesign, 2023-11-13:
As the latest Senec Home 4 System don't have a local interface (lala.cgi) at all and for other System with latest firmware update data such as Statistik have been removed. I made the effort to document the REST Services that are used by the current stable Senec APP v 4.3.3 and the upcoming version 4.4.0_beta01. 
The Dokumentation will be constantly updated. So just an information if it's ever planned to include Senec Home 4 Systems
https://documenter.getpostman.com/view/932140/2s9YXib2td

Heimdall-One, 2023-12-01:
> As the latest Senec Home 4 System don't have a local interface (lala.cgi) at all and for other System with latest firmware update data such as Statistik have been removed. I made the effort to [...]

So is it possible to access the Home 4 via API? My latest info was, that the API was removed?

andig, 2023-12-06:
> So is it possible to access the Home 4 via API? My latest info was, that the API was removed?

@oakdesign could you comment? Would you want to try a PR for a new template using the HTTP plugin?

andig, 2023-12-22:
No new information- closing.

closed 2023-12-22 by andig