evcc-io/evcc #10775: OpenWB: wait for initial mqtt value

andig, 2023-11-14:
Das Weglassen des Wartens auf den initialen Wert killt leider die openWB Anbindung:

```Nov 14 09:44:29 homebridge evcc[3722554]: [main  ] INFO 2023/11/14 09:44:29 evcc 0.122.1 (e9ab1ed1)
Nov 14 09:44:29 homebridge evcc[3722554]: [main  ] INFO 2023/11/14 09:44:29 using config file: /etc/evcc.yaml
Nov 14 09:44:29 homebridge evcc[3722554]: [main  ] INFO 2023/11/14 09:44:29 starting ui and api at :7070
Nov 14 09:44:29 homebridge evcc[3722554]: [db    ] INFO 2023/11/14 09:44:29 using sqlite database: /var/lib/evcc/evcc.db
Nov 14 09:44:29 homebridge evcc[3722554]: [mqtt  ] INFO 2023/11/14 09:44:29 connecting evcc-1697620767 at tcp://mqtt.xxx.de:1883
Nov 14 09:44:29 homebridge evcc[3722554]: [mqtt  ] DEBUG 2023/11/14 09:44:29 tcp://mqtt.xxx.de:1883 connected
Nov 14 09:44:29 homebridge evcc[3722554]: [openwb] INFO 2023/11/14 09:44:29 connecting evcc-605379446 at tcp://openwb.xxx.de:1883
Nov 14 09:44:29 homebridge evcc[3722554]: [openwb] DEBUG 2023/11/14 09:44:29 tcp://openwb.xxx.de:1883 connected
Nov 14 09:44:29 homebridge evcc[3722554]: [openwb] DEBUG 2023/11/14 09:44:29 tcp://openwb.xxx.de:1883 subscribe openWB/system/Timestamp
Nov 14 09:44:29 homebridge evcc[3722554]: [main  ] FATAL 2023/11/14 09:44:29 cannot create charger 'openwb2': cannot create charger 'openwb': loadpoint 2 is not configured
Nov 14 09:44:29 homebridge evcc[3722554]: [main  ] FATAL 2023/11/14 09:44:29 will attempt restart in: 5m0s
Nov 14 09:44:29 homebridge evcc[3722554]: [openwb] TRACE 2023/11/14 09:44:29 recv openWB/system/Timestamp: '1699951462'
Nov 14 09:44:29 homebridge evcc[3722554]: message repeated 2 times: [ [openwb] TRACE 2023/11/14 09:44:29 recv openWB/system/Timestamp: '1699951462']
Nov 14 09:44:29 homebridge evcc[3722554]: [openwb] TRACE 2023/11/14 09:44:29 recv openWB/lp/1/boolChargePointConfigured: '1'
Nov 14 09:44:29 homebridge evcc[3722554]: [openwb] TRACE 2023/11/14 09:44:29 recv openWB/lp/2/boolChargePointConfigured: '1'
Nov 14 09:44:32 homebridge evcc[3722554]: [openwb] TRACE 2023/11/14 09:44:32 recv openWB/system/Timestamp: '1699951472'
Nov 14 09:44:42 homebridge evcc[3722554]: [openwb] TRACE 2023/11/14 09:44:42 recv openWB/system/Timestamp: '1699951482'
Nov 14 09:44:52 homebridge evcc[3722554]: [openwb] TRACE 2023/11/14 09:44:52 recv openWB/system/Timestamp: '1699951492'
```

Grundsätzliche keine gute Lösung, dass beim Start Teilkomponenten nicht erzeugt werden, wenn noch Daten fehlen. Aber das zu ändern führt sicher weit über diesen PR hinaus.

_Originally posted by @MarkusGH in https://github.com/evcc-io/evcc/issues/10727#issuecomment-1809878073_

MarkusGH, 2023-11-14:
```
chargers:
- name: openwb1
  type: openwb
  phases1p3p: true
  broker: openwb.xxx.de # openWB IP
  id: 1                      # loadpoint ID
- name: openwb2
  type: openwb
  phases1p3p: true
  broker: openwb.xxx.de # openWB IP
  id: 2                      # loadpoint ID

```

andig, 2023-11-14:
Könntest du mir den Broker per Mail an info@evcc.io zukommen lassen? Der scheint public zu sein?

MarkusGH, 2023-11-14:
> Könntest du mir den Broker per Mail an [info@evcc.io](mailto:info@evcc.io) zukommen lassen? Der scheint public zu sein?

Ist er leider nicht - das ist mein internes Netz.
Kriege ich auf die Schnelle nicht aufgesperrt.

Aber das ist ein ganz normaler Raspi mit openWB drauf.