evcc-io/evcc #10789: charger disable: 403 Forbidden - Tesla Wall Charger twc3

nick2086, 2023-11-15:
### Describe the bug

After connecting Tesla Model Y to the Tesla Wall Charger (twc3) I receive an error in the log:

[lp-1  ] ERROR 2023/11/15 23:02:26 charger disable: 403 Forbidden

and in the UI:

FF-Charge station: charger disable: 403 Forbidden

The car begins to charge immediately.

I have reset evcc (many times), reset the twc3 with a power cycle; however, the issue persists.

### Steps to reproduce

1. Check, and double check evcc.yaml. All appears correctly configured.
2. I can browse to the Tesla Wall Charger in a browser, and receive the following response:
3. Part Number
1529455-02-E
Serial Number
PGT23061059872
Tesla Wall Connector
To edit the system's configuration, connect to the Wall Connector's access point. For detailed instructions, visit .
4. Connect the only Tesla vehicle configured.
5. Receive the error as above.

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
[... 184 lines cut ...]
[lp-1  ] DEBUG 2023/11/15 23:09:12 pv timer inactive
[lp-1  ] DEBUG 2023/11/15 23:09:12 charger: guard elapse
[lp-1  ] DEBUG 2023/11/15 23:09:12 guard timer inactive
[lp-1  ] INFO 2023/11/15 23:09:12 start charging ->
[lp-1  ] DEBUG 2023/11/15 23:09:12 wake-up timer: stop
[lp-1  ] DEBUG 2023/11/15 23:09:12 vehicle status: C
[lp-1  ] DEBUG 2023/11/15 23:09:12 vehicle soc: 62%
[lp-1  ] DEBUG 2023/11/15 23:09:12 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/11/15 23:09:12 vehicle range: 322km
[lp-1  ] DEBUG 2023/11/15 23:09:12 pv charge current: 0A = 0A + -35.1A (8065W @ 1p)
[lp-1  ] DEBUG 2023/11/15 23:09:12 site power 8065W >= 0W disable threshold
[lp-1  ] DEBUG 2023/11/15 23:09:12 pv disable in 0s
[lp-1  ] DEBUG 2023/11/15 23:09:12 pv disable timer elapsed
[lp-1  ] ERROR 2023/11/15 23:09:12 charger disable: 403 Forbidden
[site  ] DEBUG 2023/11/15 23:09:37 ----
[lp-1  ] DEBUG 2023/11/15 23:09:43 charge power: 7137W
```

### What type of operating system are you running?

Windows

### Version

0.122.1

andig, 2023-11-15:
Enable/disable läuft hier über das Fahrzeug. Bitte trace Log für den Tesla.

nick2086, 2023-11-15:
Hoping this is what you wanted, apologies is not.

I've redacted some information, and replaced with ***

[main  ] INFO 2023/11/15 23:40:26 evcc 0.122.1
[main  ] INFO 2023/11/15 23:40:26 using config file: C:\Users\nickj\Desktop\evcc\evcc.yaml
[main  ] INFO 2023/11/15 23:40:26 starting ui and api at :7070
[tesla ] TRACE 2023/11/15 23:40:26 POST https://auth.tesla.com/oauth2/v3/token
[tesla ] TRACE 2023/11/15 23:40:27 client_id=ownerapi&grant_type=refresh_token&refresh_token=***
--
{"access_token":"***","expires_in":28800,"token_type":"Bearer"}
[tesla ] TRACE 2023/11/15 23:40:27 GET https://owner-api.teslamotors.com/api/1/vehicles
[tesla ] TRACE 2023/11/15 23:40:28 {"response":[{"id":***,"vehicle_id":***,"vin":"***","color":null,"access_type":"OWNER","display_name":"Rosso","option_codes":null,"granular_access":{"hide_private":false},"tokens":["***"],"state":"online","in_service":false,"id_s":"***","calendar_enabled":true,"api_version":63,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false}],"count":1}
[site  ] INFO 2023/11/15 23:40:28 site config:
[site  ] INFO 2023/11/15 23:40:28   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/11/15 23:40:28     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/11/15 23:40:28     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/11/15 23:40:28   vehicles:
[site  ] INFO 2023/11/15 23:40:28     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/11/15 23:40:28 loadpoint 1:
[lp-1  ] INFO 2023/11/15 23:40:28   mode:        pv
[lp-1  ] INFO 2023/11/15 23:40:28   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/11/15 23:40:28   meters:      charge ✓
[lp-1  ] INFO 2023/11/15 23:40:28     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2023/11/15 23:40:28 phase timer inactive
[lp-1  ] DEBUG 2023/11/15 23:40:28 pv timer inactive
[lp-1  ] DEBUG 2023/11/15 23:40:28 guard timer inactive
[lp-1  ] INFO 2023/11/15 23:40:28 vehicle updated: unknown -> Rosso
[site  ] DEBUG 2023/11/15 23:40:28 ----
[lp-1  ] DEBUG 2023/11/15 23:40:28 charge power: 7070W
[site  ] DEBUG 2023/11/15 23:40:32 pv power: 0W
[site  ] DEBUG 2023/11/15 23:40:34 grid power: 8016W
[site  ] DEBUG 2023/11/15 23:40:34 site power: 8016W
[tesla ] TRACE 2023/11/15 23:40:34 GET https://owner-api.teslamotors.com/api/1/vehicles/3744642368904898/vehicle_data
[tesla ] TRACE 2023/11/15 23:40:35 {"response":{"id":***,"user_id":***,"vehicle_id":***,"vin":"***","color":null,"access_type":"OWNER","granular_access":{"hide_private":false},"tokens":["***","***"],"state":"online","in_service":false,"id_s":"***","calendar_enabled":true,"api_version":63,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false,"charge_state":{"battery_heater_on":false,"battery_level":63,"battery_range":202.88,"charge_amps":32,"charge_current_request":32,"charge_current_request_max":32,"charge_enable_request":true,"charge_energy_added":0.4,"charge_limit_soc":80,"charge_limit_soc_max":100,"charge_limit_soc_min":50,"charge_limit_soc_std":80,"charge_miles_added_ideal":1.5,"charge_miles_added_rated":1.5,"charge_port_cold_weather_mode":false,"charge_port_color":"\u003cinvalid\u003e","charge_port_door_open":true,"charge_port_latch":"Engaged","charge_rate":27.1,"charger_actual_current":32,"charger_phases":1,"charger_pilot_current":32,"charger_power":7,"charger_voltage":215,"charging_state":"Charging","conn_charge_cable":"IEC","est_battery_range":208.58,"fast_charger_brand":"\u003cinvalid\u003e","fast_charger_present":false,"fast_charger_type":"ACSingleWireCAN","ideal_battery_range":202.88,"max_range_charge_counter":0,"minutes_to_full_charge":120,"not_enough_power_to_heat":null,"off_peak_charging_enabled":false,"off_peak_charging_times":"all_week","off_peak_hours_end_time":360,"preconditioning_enabled":false,"preconditioning_times":"weekdays","scheduled_charging_mode":"Off","scheduled_charging_pending":false,"scheduled_charging_start_time":null,"scheduled_charging_start_time_app":0,"scheduled_departure_time":1699907400,"scheduled_departure_time_minutes":450,"supercharger_session_trip_planner":false,"time_to_full_charge":2.0,"timestamp":1700052035976,"trip_charging":false,"usable_battery_level":63,"user_charge_enable_request":null},"climate_state":{"allow_cabin_overheat_protection":true,"auto_seat_climate_left":false,"auto_seat_climate_right":false,"auto_steering_wheel_heat":false,"battery_heater":false,"battery_heater_no_power":null,"bioweapon_mode":false,"cabin_overheat_protection":"On","cabin_overheat_protection_actively_cooling":false,"climate_keeper_mode":"off","cop_activation_temperature":"High","defrost_mode":0,"driver_temp_setting":20.5,"fan_status":0,"hvac_auto_request":"On","inside_temp":21.5,"is_auto_conditioning_on":false,"is_climate_on":false,"is_front_defroster_on":false,"is_preconditioning":false,"is_rear_defroster_on":false,"left_temp_direction":0,"max_avail_temp":28.0,"min_avail_temp":15.0,"outside_temp":20.5,"passenger_temp_setting":20.5,"remote_heater_control_enabled":false,"right_temp_direction":0,"seat_heater_left":0,"seat_heater_rear_center":0,"seat_heater_rear_left":0,"seat_heater_rear_right":0,"seat_heater_right":0,"side_mirror_heaters":false,"steering_wheel_heat_level":0,"steering_wheel_heater":false,"supports_fan_only_cabin_overheat_protection":true,"timestamp":1700052035976,"wiper_blade_heater":false},"drive_state":{"active_route_traffic_minutes_delay":0.0,"gps_as_of":1700051956,"heading":3,"l [... cut ...]
[lp-1  ] DEBUG 2023/11/15 23:40:35 pv disable timer elapsed
[tesla ] TRACE 2023/11/15 23:40:35 POST https://owner-api.teslamotors.com/api/1/vehicles/3744642368904898/command/charge_stop
[tesla ] TRACE 2023/11/15 23:40:35 {"response":null,"error":"Tesla Vehicle Command Protocol required, please refer to the documentation here: https://developer.tesla.com/docs/fleet-api#2023-10-09-rest-api-vehicle-commands-endpoint-deprecation-warning","error_description":""}
[lp-1  ] ERROR 2023/11/15 23:40:35 charger disable: 403 Forbidden

nick2086, 2023-11-15:
Thank you for your update.

When I searched for this error after it occurred I could find nothing about the error, or evcc no longer working with TWC3. 

Does that mean the evcc won't now work for any Tesla vehicles using a TWC3?

I thought there would have been a lot of comments about this?

andig, 2023-11-16:
Closing as duplicate of https://github.com/evcc-io/evcc/issues/10460

closed 2023-11-16 by andig