evcc-io/evcc #10796: KIA Connect , Ladeziel zu variabel

theoldphilip, 2023-11-16:
### Describe the bug

Im Dialog Ladeziel (10 - 100 %) wird ja auch die von KIA Connect ermittelte Reichweite angezeigt.
Mein XCeed PEHV hat im Idealfall bei 100 % SoC rund 55 KM. Leider wird dieser Wert von EVCC immer 
wieder verändert. 

Manchmal sind 100 % 20 KM manchmal sind 100 % 36 KM.
Die Batteriekapazität ist in der Config auf 10 KW angegeben.

Gibt es eine Möglichkeit die 100 % auf 55 KM zu fixieren ?

Gruß Philip 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Scheinbar abhängig von der Aktualisierung der KIA Connect Schnittstelle
```

### Log details

```text
..
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.122.1

andig, 2023-11-16:
Leider nicht. Da eher optisches Problem an dieser Stelle wontfix.

closed 2023-11-16 by andig

VolkerK62, 2023-11-16:
Die aktuelle Reichweite kommt doch von der Fahrzeug API, oder?
evcc rechnet doch nur bis 100% hoch (Dreisatz)