evcc-io/evcc #10810: UI for battery control

naltatis, 2023-11-17:
Make battery control https://github.com/evcc-io/evcc/pull/10553 visible in the UI.

```yaml
site:
  batteryDischargeControl: true
```

- [ ] Add a toggle switch to activate/deactivate (🧪 experimental)
- [ ] Add status indicator for battery mode (normal, hold, charge)
- [ ] API: add status to site, add endpoint for (de)activating @andig

andig, 2023-11-17:
@naltatis den Status gibts doch schon? Oder welchen Status meinst Du?

naltatis, 2023-11-17:
Die gepublishte Info an der Site zum aktuellen Batteriemodus

andig, 2023-11-17:
Du verwirrst mich: https://github.com/evcc-io/evcc/pull/10553/files#diff-b55414d3499ab543964db1d26fc3dc2dca53bb6005bac7481f0fbf95c505601aR48

naltatis, 2023-11-17:
> Du verwirrst mich: https://github.com/evcc-io/evcc/pull/10553/files#diff-b55414d3499ab543964db1d26fc3dc2dca53bb6005bac7481f0fbf95c505601aR48

Ah, den hatte ich übersehen. Danke.

closed 2023-11-22 by naltatis