evcc-io/evcc #10822: guest vehicle reported as connected even if there is none

modwadd, 2023-11-18:
### Describe the bug

Using evcc v0.122.1 on Home Asisstent (not verified on diferent system) reports a connected vehicle.
[image]
even if no one is connected!

### Steps to reproduce

1.start evcc, and wait
2.start evcc, connect car e.g. Zoe, disconnect car, wait

pulsares wallbox is V3 software. Sleep disabled. connectied via wafeshare rs485 to eth gateway.
Connectione verified, current is changing in PV mode.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: yy

interval: 30s # control cycle interval war 10

# mqtt message broker
mqtt:
  broker: 192.168.yy:1883
  topic: evcc # root topic for publishing, set empty to disable
  user:
  password:

meters:
- name: grid1
  type: custom
  power: #Power # default value, optionally override # Leistung (W)
    source: mqtt
    topic: Solax/gridimport
    timeout: 70s
    scale: -1
  energy: #Sum # default value, optionally override # Energiemenge (kWh)
    source: mqtt
    topic: Solax/grid_energy #
    timeout: 70s
    scale: -1
- name: pv # Leistung PV
  type: custom
  power:
    source: mqtt
    topic: Solax/pvleistung
    timeout: 70s
- name: battery
  type: custom
  capacity: 11 # Akkukapazität in kWh # Optional
  power:
    source: mqtt
    topic: Solax/batterieladung
    timeout: 70s
    scale: -1
  soc: # Batterie SOC (%)
    source: mqtt
    topic: Solax/batterysoc

chargers:
- name: my_charger
  type: template
  template: pulsares 
  # RS485 via TCP/IP (Modbus RTU)
  modbus: tcpip #rs485tcpip
  id: 2
  host: 192.168.xx.yy # Hostname
  port: yy # Port 

vehicles:
- type: template
  template: renault 
  title: Zoe  
  user: yyy  
  password: yyy  
  vin:
  capacity: 50  
  phases: 3
  name: ev4
- type: template
  template: bmw 
  title: x3  
  user: xxx  
  password: xxx
  vin:  
  capacity: 11  
  phases: 1
  name: ev5

loadpoints:
- title: Garage
  charger: my_charger
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv
    battery:
    - battery
```

### Log details

```text
not yet avaiable
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.122.1

andig, 2023-11-18:
We need the output of 

    evcc charger

For this case. Closing until available.

modwadd, 2023-11-18:
I moved the setup to a debian PC:
kai@debian:~$ evcc charger
[main  ] INFO 2023/11/18 18:39:42 evcc 0.122.1
[main  ] INFO 2023/11/18 18:39:42 using config file: /home/kai/evcc.yaml
[db    ] INFO 2023/11/18 18:39:42 using sqlite database: /home/kai/.evcc/evcc.db
[mqtt  ] INFO 2023/11/18 18:39:42 connecting evcc-879769307 at tcp://192.168.13.30:1883
[mqtt  ] DEBUG 2023/11/18 18:39:42 tcp://192.168.13.30:1883 connected
Charge status: B
Enabled:       false

modwadd, 2023-11-18:
evcc -c evcc.yaml
kommen diese ausgaben:
[site  ] DEBUG 2023/11/18 18:47:53 ----
[lp-1  ] DEBUG 2023/11/18 18:47:53 charge power: 0W
[site  ] DEBUG 2023/11/18 18:47:53 pv power: 0W
[site  ] DEBUG 2023/11/18 18:47:53 battery soc: 15%
[site  ] DEBUG 2023/11/18 18:47:53 battery power: 858W
[site  ] DEBUG 2023/11/18 18:47:53 grid power: -0W
[site  ] DEBUG 2023/11/18 18:47:53 site power: 858W
[lp-1  ] DEBUG 2023/11/18 18:47:53 charger status: B
[main  ] DEBUG 2023/11/18 18:47:53 vehicle status: A (Zoe)
[main  ] DEBUG 2023/11/18 18:47:53 vehicle status: A (x3)
[lp-1  ] DEBUG 2023/11/18 18:47:53 pv charge current: 0A = 0A + -3.73A (858W @ 1p)

modwadd, 2023-11-18:
Anzeige des Pulsares tools:

modwadd, 2023-11-18:
so ausgabe nochmal mit trace:
`
kai@debian:~$ evcc charger
[main  ] INFO 2023/11/18 19:07:57 evcc 0.122.1
[main  ] INFO 2023/11/18 19:07:57 using config file: /home/kai/evcc.yaml
[db    ] INFO 2023/11/18 19:07:57 using sqlite database: /home/kai/.evcc/evcc.db
[db    ] TRACE 2023/11/18 19:07:57 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/11/18 19:07:57 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/11/18 19:07:57 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/18 19:07:57 SELECT * FROM `settings` 11 <nil>
[mqtt  ] INFO 2023/11/18 19:07:57 connecting evcc-1316698305 at tcp://192.168.13.30:1883
[mqtt  ] DEBUG 2023/11/18 19:07:58 tcp://192.168.13.30:1883 connected
[db    ] TRACE 2023/11/18 19:07:58 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/11/18 19:07:58 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/11/18 19:07:58 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/11/18 19:07:58 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/11/18 19:07:58 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/18 19:07:58 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/11/18 19:07:58 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/11/18 19:07:58 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/18 19:07:58 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2023/11/18 19:07:58 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/11/18 19:07:58 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2023/11/18 19:07:58 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%   device_id       %") -1 <nil>
[db    ] TRACE 2023/11/18 19:07:58 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[pulsares] TRACE 2023/11/18 19:07:58 modbus: send 00 01 00 00 00 06 02 03 00 5d 00 01
[pulsares] TRACE 2023/11/18 19:07:58 modbus: recv 00 01 00 00 00 05 02 03 02 00 00
[pulsares] TRACE 2023/11/18 19:07:58 modbus: send 00 02 00 00 00 06 02 03 00 61 00 01
[pulsares] TRACE 2023/11/18 19:07:58 modbus: recv 00 02 00 00 00 05 02 03 02 00 00
[pulsares] TRACE 2023/11/18 19:07:58 modbus: send 00 03 00 00 00 06 02 03 00 1f 00 01
[pulsares] TRACE 2023/11/18 19:07:59 modbus: recv 00 03 00 00 00 05 02 03 02 00 01
[pulsares] TRACE 2023/11/18 19:07:59 modbus: send 00 04 00 00 00 06 02 03 00 5d 00 01
[pulsares] TRACE 2023/11/18 19:07:59 modbus: recv 00 04 00 00 00 05 02 03 02 00 00
Charge status: B
Enabled:       false
`

andig, 2023-11-18:
Bitte lesbar formatieren. Danke 🙏

modwadd, 2023-11-18:
wenn ich genauer wüsstet was ihr benötigt?
den modbus?

modwadd, 2023-11-18:
kai@debian:~$ evcc charger
[main  ] INFO 2023/11/18 19:55:50 evcc 0.122.1
[main  ] INFO 2023/11/18 19:55:50 using config file: /home/kai/evcc.yaml
[db    ] INFO 2023/11/18 19:55:50 using sqlite database: /home/kai/.evcc/evcc.db
[db    ] TRACE 2023/11/18 19:55:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/11/18 19:55:50 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/11/18 19:55:50 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/18 19:55:50 SELECT * FROM `settings` 11 <nil>
[mqtt  ] INFO 2023/11/18 19:55:50 connecting evcc-1100666892 at tcp://192.168.13.30:1883
[mqtt  ] DEBUG 2023/11/18 19:55:50 tcp://192.168.13.30:1883 connected
[db    ] TRACE 2023/11/18 19:55:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/11/18 19:55:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/11/18 19:55:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/11/18 19:55:50 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/11/18 19:55:50 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/18 19:55:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/11/18 19:55:50 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/11/18 19:55:50 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/18 19:55:50 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2023/11/18 19:55:50 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/11/18 19:55:50 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2023/11/18 19:55:50 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%   device_id       %") -1 <nil>
[db    ] TRACE 2023/11/18 19:55:50 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[pulsares] TRACE 2023/11/18 19:55:50 modbus: send 00 01 00 00 00 06 02 03 00 5d 00 01
[pulsares] TRACE 2023/11/18 19:55:50 modbus: recv 00 01 00 00 00 05 02 03 02 00 00
[pulsares] TRACE 2023/11/18 19:55:50 modbus: send 00 02 00 00 00 06 02 03 00 61 00 01
[pulsares] TRACE 2023/11/18 19:55:51 modbus: recv 00 02 00 00 00 05 02 03 02 00 00
[pulsares] TRACE 2023/11/18 19:55:51 modbus: send 00 03 00 00 00 06 02 03 00 1f 00 01
[pulsares] TRACE 2023/11/18 19:55:51 modbus: recv 00 03 00 00 00 05 02 03 02 00 01
[pulsares] TRACE 2023/11/18 19:55:51 modbus: send 00 04 00 00 00 06 02 03 00 5d 00 01
[pulsares] TRACE 2023/11/18 19:55:51 modbus: recv 00 04 00 00 00 05 02 03 02 00 00
Charge status: B
Enabled:       false

> sorry, so besser

Manu7788, 2023-11-18:
```
[pulsares] TRACE 2023/11/18 19:25:26 modbus: send 00 14 00 00 00 06 02 03 00 1f 00 01
[pulsares] TRACE 2023/11/18 19:25:26 modbus: recv 00 14 00 00 00 05 02 03 02 00 01
[lp-1  ] DEBUG 2023/11/18 19:25:26 charger status: B
```

Hier kommt aus Register 1F eine 1, was bedeutet, dass der LÖadevorgang aktuell gesperrt ist. Für die Info bezüglich Fahrzeug an oder abgesteckt ist Register 1B notwendig.
Siehe: https://www.pulsares.shop/medien/dokumente/Modbus_Map_EV_EasyCharge.pdf

premultiply, 2023-11-19:
> ```
> Charge status: B
> Enabled: false
> ```

Stimmt doch, oder? 1 ist Status B1.
Verbunden, keine Ladefreigabe.

modwadd, 2023-11-19:
Aus meiner Sicht ist es durch mein szenario dazu gekommen das in der Wallbox der Status Ladestecker und der Status Ladevorgang out of sync sind. Der Ladestecker ist zweifelsfrei abgezogen, somit stimmt Status Ladestecker(1d). 
Da kein Stecker somit auch kein Ladevorgang. Somit müsste die WB den status in Status Ladevorgang (1f) auf 0 zurücksetzen, auch wenn ladestrom auf 0 ist.

Allerding dürfte evcc auch robuster sein und auch den Status Ladestecker auswerten.
Nur was ist dann richtig der Status Ladestecker (1d) oder der Status Ladevorgang (1f)

Ich sehe hier die WB software in der besseren situation den Fehler zu vermeiden.

andig, 2023-11-19:
@Manu7788 @premultiply bitte PR anschauen. Ich denke das war einfach völlig falsch?

modwadd, 2023-11-19:
Ich habe nur gesehen wie ich die evcc version auf nighly umstellen kann, sagt bescheid sobald die änderung dort testbar sein sollte. Dann mache ich das gerne.

closed 2023-11-20 by andig

premultiply, 2023-11-21:
Ich denke da hat es einen Bug in der Firmware.
Die Implementierung war eigentlich einfach und richtig (nach Doku).

modwadd, 2023-11-21:
Testted with nighly build, now OK.
Pulsares will improve documentation.

Manu7788, 2023-11-21:
> Ich denke da hat es einen Bug in der Firmware. Die Implementierung war eigentlich einfach und richtig (nach Doku).

Der Zustand 0 (Modbus-Map) war schlecht von mir formuliert und hat für Missverständnis gesorgt. Werd ich bei dem nächsten Update anpassen.