evcc-io/evcc #10824: Falsche PV Werte von Goodwe-Hybrid WR über Modbus TCPIP

Ghost7-hassio, 2023-11-18:
### Describe the bug

Der Goodwe Wechselrichter ist mittels Lan ins Netzwerk eingebunden und über das Goodwe Hybrid Template in evcc hinzugefügt. Die ausgelesenen Werte für PV Power stimmen nicht und sind aus unerfindlichen Gründen unrealistisch. In den Logs ist nur der Wert pv power falsch. Hierdurch wird auch im evcc Frontend In und Out komplett falsch ausgegeben. Die Integration in Home Assistant gibt die korrekten Werte aus.

### Steps to reproduce

1. Goodwe Wechselrichter über Lan in das Netzwerk einbinden.
2. Goodwe Wechselrichter mittel Goode-Hybrid Template in evc config einbinden.
3. Logs auslesen/Frontend auswerten

[image]

### Configuration details

```yaml
meters:
- name: my_grid
  type: template
  template: goodwe-hybrid
  usage: grid
  modbus: tcpip
  id: 247
  host: 192.168.178.71
  port: 502

- name: my_pv
  type: template
  template: goodwe-hybrid
  usage: pv
  modbus: tcpip
  id: 247
  host: 192.168.178.71
  port: 502

- name: my_battery
  type: template
  template: goodwe-hybrid
  usage: battery
  modbus: tcpip
  id: 247
  host: 192.168.178.71
  port: 502

site:
  title: Home # display name for UI
  meters:
    grid: my_grid
    pv: my_pv
```

### Log details

```text
ite  ] DEBUG 2023/11/18 15:57:45 pv power: 8589934633W
[site  ] DEBUG 2023/11/18 15:57:46 battery soc: 0%
[site  ] DEBUG 2023/11/18 15:57:46 battery power: 0W
[site  ] DEBUG 2023/11/18 15:57:46 grid power: 376W
[site  ] DEBUG 2023/11/18 15:57:46 site power: 376W
[lp-1  ] DEBUG 2023/11/18 15:57:46 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/11/18 15:57:46 charge total import: 4452.485kWh
[lp-1  ] DEBUG 2023/11/18 15:57:46 charger status: A
[site  ] DEBUG 2023/11/18 15:58:15 ----
[lp-1  ] DEBUG 2023/11/18 15:58:15 charge power: 0W
[lp-2  ] DEBUG 2023/11/18 15:58:15 charge power: 0W
[site  ] DEBUG 2023/11/18 15:58:16 pv power: 8589934621W
[site  ] DEBUG 2023/11/18 15:58:16 battery soc: 0%
[site  ] DEBUG 2023/11/18 15:58:16 battery power: 0W
[site  ] DEBUG 2023/11/18 15:58:16 grid power: 249W
[site  ] DEBUG 2023/11/18 15:58:16 site power: 249W
[lp-2  ] DEBUG 2023/11/18 15:58:16 charge currents: [0.008 0.008 0.008]A
[lp-2  ] DEBUG 2023/11/18 15:58:16 charge total import: 4812.905kWh
[lp-2  ] DEBUG 2023/11/18 15:58:16 charger status: B
[lp-2  ] DEBUG 2023/11/18 15:58:16 targetSoc reached: 82.0% > 80%
[site  ] DEBUG 2023/11/18 15:58:45 ----
[lp-1  ] DEBUG 2023/11/18 15:58:45 charge power: 0W
[lp-2  ] DEBUG 2023/11/18 15:58:45 charge power: 0W
[site  ] DEBUG 2023/11/18 15:58:46 pv power: 8589934621W
[site  ] DEBUG 2023/11/18 15:58:46 battery soc: 0%
[site  ] DEBUG 2023/11/18 15:58:46 battery power: 0W
[site  ] DEBUG 2023/11/18 15:58:46 grid power: 410W
[site  ] DEBUG 2023/11/18 15:58:46 site power: 410W
[lp-1  ] DEBUG 2023/11/18 15:58:46 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/11/18 15:58:46 charge total import: 4452.485kWh
[lp-1  ] DEBUG 2023/11/18 15:58:46 charger status: A
[site  ] DEBUG 2023/11/18 15:59:15 ----
[lp-1  ] DEBUG 2023/11/18 15:59:15 charge power: 0W
[lp-2  ] DEBUG 2023/11/18 15:59:15 charge power: 0W
[site  ] DEBUG 2023/11/18 15:59:16 pv power: 8589934631W
[site  ] DEBUG 2023/11/18 15:59:16 battery soc: 0%
[site  ] DEBUG 2023/11/18 15:59:16 battery power: 0W
[site  ] DEBUG 2023/11/18 15:59:16 grid power: 514W
[site  ] DEBUG 2023/11/18 15:59:16 site power: 514W
[lp-2  ] DEBUG 2023/11/18 15:59:16 charge currents: [0.008 0.008 0.008]A
[lp-2  ] DEBUG 2023/11/18 15:59:16 charge total import: 4812.905kWh
[lp-2  ] DEBUG 2023/11/18 15:59:16 charger status: B
[lp-2  ] DEBUG 2023/11/18 15:59:16 targetSoc reached: 82.0% > 80%
[site  ] DEBUG 2023/11/18 15:59:45 ----
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.122.1

andig, 2023-11-18:
Bitte

    evcc meter

Ghost7-hassio, 2023-11-18:
Meinst du in der config soll ich folgendes konfigurieren?

log: trace
levels:
  evcc meter: trace

Oder gibts unter Home Assistant die Möglichkeit den Befehl per Konsole auszuführen?

andig, 2023-11-19:
Bitte den Befehl auf der Kommandozeile. HA brauchts dafür nicht.

Ghost7-hassio, 2023-11-19:
Sorry, wenn ich dumm fragen muss, aber evcc läuft bei mir in HA als Addon. Ich dachte, dass dadurch das Ausführen von Kommandozeilenbefehlen nicht möglich ist?

premultiply, 2023-11-20:
Dann halt evcc auf dem Rechner ausführen mit dem du das gerade hier schreibst?

Ghost7-hassio, 2023-11-21:
Da ist natürlich was dran... Einmal ausgeführt:

### Log details

```text
[main  ] INFO 2023/11/21 22:40:30 evcc 0.122.1
[main  ] INFO 2023/11/21 22:40:30 using config file: C:\Users\Downloads\evcc\evcc.yaml
[db    ] INFO 2023/11/21 22:40:31 using sqlite database: C:\Users\.evcc\evcc.db
[db    ] TRACE 2023/11/21 22:40:31 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%   device_id       %") -1 <nil>
[db    ] TRACE 2023/11/21 22:40:31 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
my_grid
-------
[modbus] TRACE 2023/11/21 22:40:31 modbus: send 00 01 00 00 00 06 f7 03 8c b9 00 02
[modbus] TRACE 2023/11/21 22:40:31 modbus: recv 00 01 00 00 00 07 f7 03 04 ff ff f6 c1
[modbus] TRACE 2023/11/21 22:40:31 modbus: send 00 02 00 00 00 06 f7 03 8c b1 00 02
[modbus] TRACE 2023/11/21 22:40:31 modbus: recv 00 02 00 00 00 07 f7 03 04 4a 4b bb c8
Power:  2367W
Energy: 333797.0kWh

my_pv
-----
[modbus] TRACE 2023/11/21 22:40:31 modbus: send 00 03 00 00 00 06 f7 03 89 21 00 02
[modbus] TRACE 2023/11/21 22:40:31 modbus: recv 00 03 00 00 00 07 f7 03 04 00 00 00 00
[modbus] TRACE 2023/11/21 22:40:31 modbus: send 00 04 00 00 00 06 f7 03 89 25 00 02
[modbus] TRACE 2023/11/21 22:40:31 modbus: recv 00 04 00 00 00 07 f7 03 04 00 00 00 00
[modbus] TRACE 2023/11/21 22:40:31 modbus: send 00 05 00 00 00 06 f7 03 89 29 00 02
[modbus] TRACE 2023/11/21 22:40:31 modbus: recv 00 05 00 00 00 07 f7 03 04 ff ff ff ff
[modbus] TRACE 2023/11/21 22:40:31 modbus: send 00 06 00 00 00 06 f7 03 89 2d 00 02
[modbus] TRACE 2023/11/21 22:40:31 modbus: recv 00 06 00 00 00 07 f7 03 04 ff ff ff ff
[modbus] TRACE 2023/11/21 22:40:31 modbus: send 00 07 00 00 00 06 f7 03 89 77 00 02
[modbus] TRACE 2023/11/21 22:40:31 modbus: recv 00 07 00 00 00 07 f7 03 04 00 00 6c cf
Power:  8589934590W
Energy: 2785.5kWh
```

premultiply, 2023-11-22:
💡MPPT 3&4 liefern hier NaN-Werte.

PR #10865 (2023-11-22): goodwe-hybrid: fix NaN on MPPTs
Fixes #10824
commit 7ee92979 (2023-11-22): goodwe-hybrid: fix NaN on MPPTs (#10865)

Fixes #10824
closed 2023-11-22 by andig