evcc-io/evcc #10831: EVC04 - Nach Update auf 0.122.1 startet und stoppt die Ladung ständig

chrostek, 2023-11-19:
### Describe the bug

Nach Update auf 0.122.1 startet und stoppt die Ladung ständig (PV Mode mit gesetzter Preisgrenze mit smartem Netzladen (Tibber))
Auf Version 0.121.5 ist noch soweit alles stabil.

### Steps to reproduce

1. PV Modus
2. Preisgrenze setzen, die zum jetzigen Zeitpunkt nicht erfüllt ist - er sollte also nicht laden
3. Ladevorgang startet und stoppt ständig

### Configuration details

```yaml
network:
  schema: http
  host: sc-evcc.chrostek.network
  port: 7070

log: info
levels:
  cache: error

interval: 30s

sponsortoken: xxxxxxxxxx

meters:
- type: template
  template: tibber-pulse
  usage: grid
  token: xxxxxxxxxx
  homeid: xxxxxxxxxx
  name: grid1

- type: custom
  name: pv1
  power:
    source: mqtt
    topic: mbmd/sdm1-9/meter/powerl1
    scale: -1
  currents:
  - source: mqtt
    topic: mbmd/sdm1-9/meter/currentl1
    scale: -1
  - source: mqtt
    topic: mbmd/sdm1-9/meter/currentl2
    scale: -1
  - source: mqtt
    topic: mbmd/sdm1-9/meter/currentl3
    scale: -1
  energy:
    source: mqtt
    topic: mbmd/sdm1-9/meter/export

chargers:
- type: template
  template: vestel
  id: 255
  host: 172.30.2.230
  port: 502
  modbus: tcpip
  name: wallbox3

vehicles:
- type: template
  template: smart
  title: xxxxxxxxxx
  user: xxxxxxxxxx
  password: 'xxxxxxxxxx'
  vin: xxxxxxxxxx
  capacity: 17.6
  name: ev2

loadpoints:
- title: Wallbox xxxxxxxxxx
  charger: wallbox3
  vehicle: ev2
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 32
  resetOnDisconnect: true

site:
  title: xxxxxxxxxx
  meters:
    grid: grid1
    pv: pv1

mqtt:
  broker: 172.30.20.23:1883
  topic: mbmd
  user: xxxxxxxxxx
  password: xxxxxxxxxx
  clientid: evcc

tariffs:
  grid:
    type: tibber
    token: xxxxxxxxxx
    homeid: xxxxxxxxxx

influx:
  url: http://172.30.20.24:8086
  database: evcc
  user: evcc
  password: xxxxxxxxxx
```

### Log details

```text
[lp-1  ] INFO 2023/11/19 11:46:13 car connected
[lp-1  ] INFO 2023/11/19 11:46:13 start charging ->
[lp-1  ] INFO 2023/11/19 11:46:22 stop charging <-
[lp-1  ] INFO 2023/11/19 11:46:52 start charging ->
[lp-1  ] WARN 2023/11/19 11:47:22 charger out of sync: expected disabled, got enabled

evcc charger --log trace
[main  ] INFO 2023/11/19 11:48:18 evcc 0.122.1
[main  ] INFO 2023/11/19 11:48:18 using config file: /etc/evcc.yaml
[db    ] INFO 2023/11/19 11:48:18 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2023/11/19 11:48:18 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/11/19 11:48:18 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table"
desc 1 <nil>
[db    ] TRACE 2023/11/19 11:48:18 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/19 11:48:18 SELECT * FROM `settings` 0 <nil>
[mqtt  ] INFO 2023/11/19 11:48:18 connecting evcc at tcp://172.30.20.23:1883
[mqtt  ] DEBUG 2023/11/19 11:48:18 tcp://172.30.20.23:1883 connected
[db    ] TRACE 2023/11/19 11:48:18 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/11/19 11:48:18 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/11/19 11:48:18 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/11/19 11:48:18 CREATE TABLE `configs` (`id` integer PRIMARY KEY AUTOINCREMENT,`class` integer,`type` text) 0 <nil>
[db    ] TRACE 2023/11/19 11:48:18 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/11/19 11:48:18 CREATE TABLE `config_details` (`config_id` integer,`key` text,`value` text,CONSTRAINT `fk_configs_details` FOREIGN KEY (`config_id`) REFERENCES `configs`(`id`) ON DELETE CASCADE ON UPDATE CASCADE) 0 <nil>
[db    ] TRACE 2023/11/19 11:48:18 CREATE INDEX `idx_unique` ON `config_details`(`config_id`,`key`) 0 <nil>
[db    ] TRACE 2023/11/19 11:48:18 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT       fk_devices_details      %") -1 <nil>
[db    ] TRACE 2023/11/19 11:48:18 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%  device_id       %") -1 <nil>
[db    ] TRACE 2023/11/19 11:48:18 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[vestel] TRACE 2023/11/19 11:48:18 modbus: send 00 01 00 00 00 06 ff 03 07 d2 00 01
[vestel] TRACE 2023/11/19 11:48:18 modbus: recv 00 01 00 00 00 05 ff 03 02 00 00
[vestel] TRACE 2023/11/19 11:48:18 modbus: send 00 02 00 00 00 06 ff 04 03 fc 00 02
[vestel] TRACE 2023/11/19 11:48:18 modbus: recv 00 02 00 00 00 07 ff 04 04 00 00 23 91
[vestel] TRACE 2023/11/19 11:48:18 modbus: send 00 03 00 00 00 06 ff 04 04 0c 00 02
[vestel] TRACE 2023/11/19 11:48:18 modbus: recv 00 03 00 00 00 07 ff 04 04 00 00 0a 50
[vestel] TRACE 2023/11/19 11:48:18 modbus: send 00 04 00 00 00 06 ff 04 03 f0 00 01
[vestel] TRACE 2023/11/19 11:48:18 modbus: recv 00 04 00 00 00 05 ff 04 02 35 73
[vestel] TRACE 2023/11/19 11:48:18 modbus: send 00 05 00 00 00 06 ff 04 03 f2 00 01
[vestel] TRACE 2023/11/19 11:48:18 modbus: recv 00 05 00 00 00 05 ff 04 02 34 e7
[vestel] TRACE 2023/11/19 11:48:18 modbus: send 00 06 00 00 00 06 ff 04 03 f4 00 01
[vestel] TRACE 2023/11/19 11:48:18 modbus: recv 00 06 00 00 00 05 ff 04 02 35 10
[vestel] TRACE 2023/11/19 11:48:18 modbus: send 00 07 00 00 00 06 ff 04 03 f6 00 01
[vestel] TRACE 2023/11/19 11:48:18 modbus: recv 00 07 00 00 00 05 ff 04 02 00 e2
[vestel] TRACE 2023/11/19 11:48:18 modbus: send 00 0b 00 00 00 06 ff 04 03 e9 00 01
[vestel] TRACE 2023/11/19 11:48:18 modbus: recv 00 0b 00 00 00 05 ff 04 02 00 01
[vestel] TRACE 2023/11/19 11:48:18 modbus: send 00 0c 00 00 00 06 ff 04 03 ec 00 01
[vestel] TRACE 2023/11/19 11:48:18 modbus: recv 00 0c 00 00 00 05 ff 04 02 00 03
[vestel] TRACE 2023/11/19 11:48:18 modbus: send 00 0d 00 00 00 06 ff 04 03 e9 00 01
[vestel] TRACE 2023/11/19 11:48:18 modbus: recv 00 0d 00 00 00 05 ff 04 02 00 01
[vestel] TRACE 2023/11/19 11:48:18 modbus: send 00 0e 00 00 00 06 ff 04 05 de 00 02
[vestel] TRACE 2023/11/19 11:48:18 modbus: recv 00 0e 00 00 00 07 ff 04 04 00 00 01 13
[vestel] TRACE 2023/11/19 11:48:18 modbus: send 00 0f 00 00 00 06 ff 04 05 e4 00 02
[vestel] TRACE 2023/11/19 11:48:18 modbus: recv 00 0f 00 00 00 07 ff 04 04 00 00 00 e2
Power:          9105W
Energy:         264.0kWh
Current L1..L3: 13.7A 13.5A 13.6A
Voltage L1..L3: 226V 226V 225V
Charge status:  C
Enabled:        true
Charged:        0.3kWh
Duration:       3m46s
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.122.1

StefanSchoof, 2023-11-19:
Klingt ähnlich wie #10735

chrostek, 2023-11-19:
Stimmt. Ich hatte nur in den issues gesucht, damit ich kein Duplikat anlege, aber in den discussions habe ich nicht geschaut 🙈

andig, 2023-11-19:
> Klingt ähnlich wie https://github.com/evcc-io/evcc/discussions/10735

Der letzte Stand scheint der zu sein: https://github.com/evcc-io/evcc/discussions/10735#discussioncomment-7574970, geht also.

/cc @premultiply

chrostek, 2023-11-19:
den Kommentar habe ich auch gelesen - aber ich finde im Text davor keine Lösung oder Änderung an der Config - also keine Ahnung, warum es bei @ogheinz77 jetzt geht. bei mir geht es definitiv nicht. Sobald ich das Fahrzeug anklemme, fängt er an zu laden - auch wenn er das nicht sollte. Der Zustand wechselt dann ständig zwischen Laden, Warten, nicht Laden, etc - aber es fließt dauerhaft Leistung - er lädt also einfach.

andig, 2023-11-19:
Historie: https://github.com/evcc-io/evcc/commits/master/charger/vestel.go

chrostek, 2023-11-19:
Damit ging es auch nicht, dann hatte ich den Bug:
https://github.com/evcc-io/evcc/issues/10652
ohne den commit ist es auch bei mir nicht mehr gestartet

0163deralex, 2023-11-20:
Bei mir klappt auch nix mehr richtig. Der pv Modus lädt mitten in der Nacht, AUS reagiert gar nicht, alle anderen Modi sind für mich grad uninteressant.
Kommt da ein Update oder muß ich manuell irgendwas fixen? Bin ein wenig traurig, da es bisher richtig perfekt funktioniert hat. Dann kam dieser out of sync fix und alles war anders 😔

andig, 2023-11-20:
Meckern nutzt ja nix. Gibts ein Modbus Log einer alten Version an dem man erkennen kann was da anders war?

> Dann kam dieser out of sync fix und alles war anders 😔

Was meinst Du damit konkret? Welche Version geht, welche dann nicht mehr? Welcher Chance ist schuld?

chrostek, 2023-11-20:
letzte funktionierende Version: 0.121.5
kein Programmstart möglich mit: 0.122.0 ( siehe https://github.com/evcc-io/evcc/issues/10652 )
Programmstart möglich, aber unkontrollierbares Ladeverhalten mit: 0.122.1

ich habe gerade kein Auto dran hängen, reichen auch Logs ohne Auto?

(evcc charger --log trace)

Version 0.121.5:

> [main  ] INFO 2023/11/20 18:49:52 evcc 0.121.5
[main  ] INFO 2023/11/20 18:49:52 using config file: /etc/evcc.yaml
[db    ] INFO 2023/11/20 18:49:52 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2023/11/20 18:49:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/11/20 18:49:52 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table"
desc 1 <nil>
[db    ] TRACE 2023/11/20 18:49:52 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/20 18:49:52 SELECT * FROM `settings` 0 <nil>
[mqtt  ] INFO 2023/11/20 18:49:52 connecting evcc at tcp://172.30.20.23:1883
[mqtt  ] DEBUG 2023/11/20 18:49:52 tcp://172.30.20.23:1883 connected
[db    ] TRACE 2023/11/20 18:49:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/11/20 18:49:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/11/20 18:49:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/11/20 18:49:52 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/11/20 18:49:52 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/20 18:49:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/11/20 18:49:52 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/11/20 18:49:52 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/20 18:49:52 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT       fk_configs_details      %") -1 <nil>
[db    ] TRACE 2023/11/20 18:49:52 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/11/20 18:49:52 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT       fk_devices_details      %") -1 <nil>
[db    ] TRACE 2023/11/20 18:49:52 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%  device_id       %") -1 <nil>
[db    ] TRACE 2023/11/20 18:49:52 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[vestel] TRACE 2023/11/20 18:49:52 modbus: send 00 01 00 00 00 06 ff 04 03 fc 00 02
[vestel] TRACE 2023/11/20 18:49:52 modbus: recv 00 01 00 00 00 07 ff 04 04 00 00 00 04
[vestel] TRACE 2023/11/20 18:49:52 modbus: send 00 02 00 00 00 06 ff 04 04 0c 00 02
[vestel] TRACE 2023/11/20 18:49:52 modbus: recv 00 02 00 00 00 07 ff 04 04 00 00 0a 69
[vestel] TRACE 2023/11/20 18:49:52 modbus: send 00 03 00 00 00 06 ff 04 03 f0 00 01
[vestel] TRACE 2023/11/20 18:49:52 modbus: recv 00 03 00 00 00 05 ff 04 02 00 31
[vestel] TRACE 2023/11/20 18:49:52 modbus: send 00 04 00 00 00 06 ff 04 03 f2 00 01
[vestel] TRACE 2023/11/20 18:49:52 modbus: recv 00 04 00 00 00 05 ff 04 02 00 12
[vestel] TRACE 2023/11/20 18:49:52 modbus: send 00 05 00 00 00 06 ff 04 03 f4 00 01
[vestel] TRACE 2023/11/20 18:49:52 modbus: recv 00 05 00 00 00 05 ff 04 02 00 12
[vestel] TRACE 2023/11/20 18:49:52 modbus: send 00 06 00 00 00 06 ff 04 03 f6 00 01
[vestel] TRACE 2023/11/20 18:49:52 modbus: recv 00 06 00 00 00 05 ff 04 02 00 df
[vestel] TRACE 2023/11/20 18:49:52 modbus: send 00 07 00 00 00 06 ff 04 03 f8 00 01
[vestel] TRACE 2023/11/20 18:49:52 modbus: recv 00 07 00 00 00 05 ff 04 02 00 e0
[vestel] TRACE 2023/11/20 18:49:52 modbus: send 00 08 00 00 00 06 ff 04 03 fa 00 01
[vestel] TRACE 2023/11/20 18:49:52 modbus: recv 00 08 00 00 00 05 ff 04 02 00 de
[vestel] TRACE 2023/11/20 18:49:52 modbus: send 00 09 00 00 00 06 ff 04 03 ec 00 01
[vestel] TRACE 2023/11/20 18:49:52 modbus: recv 00 09 00 00 00 05 ff 04 02 00 01
[vestel] TRACE 2023/11/20 18:49:52 modbus: send 00 0a 00 00 00 06 ff 03 13 8c 00 01
[vestel] TRACE 2023/11/20 18:49:52 modbus: recv 00 0a 00 00 00 05 ff 03 02 00 20
[vestel] TRACE 2023/11/20 18:49:52 modbus: send 00 0b 00 00 00 06 ff 04 05 de 00 02
[vestel] TRACE 2023/11/20 18:49:52 modbus: recv 00 0b 00 00 00 07 ff 04 04 00 00 00 00
[vestel] TRACE 2023/11/20 18:49:52 modbus: send 00 0c 00 00 00 06 ff 04 05 e4 00 02
[vestel] TRACE 2023/11/20 18:49:52 modbus: recv 00 0c 00 00 00 07 ff 04 04 00 01 1e 9c
Power:          4W
Energy:         266.5kWh
[... cut ...]
[vestel] TRACE 2023/11/20 18:51:10 modbus: send 00 0c 00 00 00 06 ff 04 05 de 00 02
[vestel] TRACE 2023/11/20 18:51:10 modbus: recv 00 0c 00 00 00 07 ff 04 04 00 00 00 00
[vestel] TRACE 2023/11/20 18:51:10 modbus: send 00 0d 00 00 00 06 ff 04 05 e4 00 02
[vestel] TRACE 2023/11/20 18:51:10 modbus: recv 00 0d 00 00 00 07 ff 04 04 00 01 1e 9c
Power:          4W
Energy:         266.5kWh
Current L1..L3: 0.049A 0.018A 0.018A
Voltage L1..L3: 223V 224V 222V
Charge status:  A
Enabled:        false
Charged:        0.0kWh
Duration:       20h22m52s

0163deralex, 2023-11-20:
Ich mecker gar nicht, ich merke doch nur an 🙈

charger out of sync: expected disabled, got enabled

Das kam ja vor der 121.x häufiger. dann habe ich zwischendurch mitbekommen, dass das gefixt wurde.

Mit der 0.121.5 läuft es soweit super, die Version davor war wohl auch fehlerhaft. 
das Update auf die 0.122.1 habe ich jetzt zwei mal gemacht (erst dachte ich, ich wäre zu hektisch gewesen, weil der Ladevorgang nicht gestoppt hat, als ich von PV auf AUS wechselte. Und dann hat der Vorgang doch irgendwann gestoppt, allerdings, um ne Sekunde später wieder zu starten (egal, ob Aus, pv, pv+min) und zu stoppen und zu starten....

Jetzt bin ich also wieder zurück zur 0.121.5. und kann hoffentlich morgen vormittag ein paar Sonnenstrahlen einfangen. 

Ein Protokoll habe ich mir natürlich nicht gezogen, tut mir leid. 

Durch das zurück gehen auf die alte Version verliert man allerdings die Ladevorgänge, die mit der neuen Version durchgeführt wurden (doof, aber ertragbar).

Wichtig ist aber, dass die Box sonst pink leuchtet, wenn der PV Modus aktiv ist (dann hat der Ladevorgang auch wirklich erst mit stärkere Sonne gestartet). 
Jetzt blinkt die im regulären blau in der 122.1. Und der Ladevorgang startet, egal was in evcc eingestellt ist.

premultiply, 2023-11-20:
Klingt für mich irgendwie immer noch verdächtig nach Watchdog.

chrostek, 2023-11-20:
Kann ich irgendwas für euch testen, um der Ursache auf die Spur zu kommen? Von mir aus könnt ihr auch per SSH auf meine VM (ist dedizierte nur für evcc), wenn mir jemand nen Public SSH Key (authorized_keys) schickt, hinterlege ich den und mache nen portforward.

andig, 2023-11-20:
Hast du die Probleme auch? Bitte bei info@evcc.io melden, ssh wär super.

developer-stephan, 2023-11-21:
Ich kann es auch bestätigen. Mit Version 0.122.1 habe ich das gleiche Problem. Nutze momentan 0.121.5 womit es alles einwandfrei funktioniert.

ogheinz77, 2023-11-21:
Hallo,
bei mir funktioniert es im Moment sehr gut, allerdings hab ich auch (noch) kein Tibber am Start ;).

chrostek, 2023-11-21:
> Hast du die Probleme auch? Bitte bei [info@evcc.io](mailto:info@evcc.io) melden, ssh wär super.

ich hatte gestern Abend eine Mail geschickt - sobald ich eine Antwort mit ssh-pub-key erhalte, kannst du gerne auf meiner VM per SSH testen

developer-stephan, 2023-11-21:
> Hallo, bei mir funktioniert es im Moment sehr gut, allerdings hab ich auch (noch) kein Tibber am Start ;).

Klingt gut. Mehr Details wäre gut und somit könntest du auch andere helfen:

evcc - Version
Vestel EVC04 Wallbox - Version

ogheinz77, 2023-11-21:
Info:
Vestel EVC04 Firmware: v3.118.0
EVCC: 0.122.1

developer-stephan, 2023-11-21:
> Info: Vestel EVC04 Firmware: v3.118.0 EVCC: 0.122.1

Ach interessant. Danke! Meins hat die Wallbox Version v3.131.0. Vielleicht sollte ich es mal downgraden? Kann man das gefahrlos machen?

ogheinz77, 2023-11-21:
Hast du die  v3.131.0 vom Support?
Hatte vor ca. 2 Wochen da nachgefragt und die haben mir gesagt v3.118.0 ist aktuell.

developer-stephan, 2023-11-21:
> Hast du die v3.131.0 vom Support? Hatte vor ca. 2 Wochen da nachgefragt und die haben mir gesagt v3.118.0 ist aktuell.

Ich habe die Wallbox letzte Woche erworben, und in der Anleitung befand sich eine Anleitung für das Update. Nachdem ich mich auf der Vestel-Website mit meinem Passwort angemeldet habe, habe ich die Version 3.131.0 gefunden.

ogheinz77, 2023-11-21:
Echt auf er Vestel Webseite, hast du einen Link?

developer-stephan, 2023-11-21:
Release Infos: https://support.vestel-germany.de/hc/de/articles/12455856839826-EVC04-Release-Notes
Download: http://evc04update.vestel-germany.de/
Passwort: Vestel_2023

developer-stephan, 2023-11-21:
PS.: Ich sehe gerade. Anscheinend haben die 131 wieder gelöscht. Augen roll...

chrostek, 2023-11-21:
Danke für Link/Passwort - das wusste ich auch nicht - bisher musste man die Firmware immer per Mail anfordern. Auf meiner Wallbox ist aber seit ich evcc nutze die v3.118.0 drauf - an einer zu neuen/verbuggten Firmware, die jetzt scheinbar zurückgezogen wurde, kann es bei mir nicht liegen. Die Probleme sind ausschließlich auf Änderungen in evcc zurückzuführen, da sich die Firmware bei vestel in meinem Setup nie geändert hat.
Downgrade habe ich nie versucht.

developer-stephan, 2023-11-21:
Dann warte ich lieber ab bis Vestel eine neue Version herausbringt anstatt ein Downgrade zu riskieren. Daher bleibe ich bei:

evcc 0.121.5
Wallbox v3.131.0

developer-stephan, 2023-11-21:
Sorry 3.131.0 ist doch auf Vestel zu sehen.

[image]

ogheinz77, 2023-11-21:
Wenn ich das richtig verstehe kommt der Fehler nur mit Tibber?

developer-stephan, 2023-11-21:
Abwarten. Der Thread-Ersteller @chrostek hat auch noch nicht von seiner Wallbox-Version erwähnt. Mit mehr Infos über die verschiedene Konstellationen, können wir mehr ausschließen und näher an die Lösung. Lassen wir uns damit beginnen:

@developer-stephan 
evcc - 0.122.1
Wallbox - v3.131.0
Tibber: ja
**Ergebnis: ständiges Starten und Stoppen**

@ogheinz77 
Vestel EVC04 Firmware: v3.118.0
EVCC: 0.122.1
Tibber: nein
**Ergebnis: es läuft**

chrostek, 2023-11-21:
Nein. m.E. gibt es bei der modbus Ansteuerung des Laders doch gar keinen Unterschied, ob der Trigger ein PV Überschuss oder Smartes Laden ist. Andere hier hatten im PV Modus doch dasselbe Problem.

chrostek, 2023-11-21:
> Abwarten. Der Thread-Ersteller @chrostek hat auch noch nicht von seiner Wallbox-Version erwähnt. Mit mehr Infos über die verschiedene Konstellationen, können wir mehr ausschließen und näher an die Lösung. 

Ich hatte vor zwei Stunden doch geschrieben, dass ich auf v3.131.0 bin.

developer-stephan, 2023-11-21:
@chrostek  Sorry, ich habe es übersehen. Es scheint also an v3.131.0 zu liegen. Das bedeutet, dass evcc die Anpassungen für v3.131.0 vornehmen müsste damit es mit 0.122.1 wieder funktioniert.

0163deralex, 2023-11-21:
Meine Vestel evc04 hat:
HMI Softwarestand :	v3.89.0
Leistungsplatine Softwarestand :	1.0.119.0

Ich nutze Tibber und einen Tesla

Dann habe ich von SMA die Wechselrichter.

Oft schalte ich manuell auf 1p um, wenn die Sonne nicht so richtig scheint und ich habe evcc im Homeassistant installiert. 

Wie weiter oben von mir beschrieben, klappt evcc in der aktuellen Version 0.122.1 gar nicht mehr, in der 0.121.5 soweit prima.

developer-stephan, 2023-11-21:
Dann spiegelt sich das fehlerhaftes Verhalten eines der beiden im Change-Log wider:

Vestel: fix enabled detection
Vestel: fix failsafe register type

ogheinz77, 2023-11-22:
Wollte das noch mal zusammenfassen.
Alle bei dehnen es nicht geht haben Tibber am Start.  

@developer-stephan
evcc:  0.122.1
Vestel evc04 hat:  v3.131.0
Tibber: ja
Ergebnis: ständiges Starten und Stoppen - kommt nie in Lademodus

@0163deralex
evcc:  0.122.1
Vestel evc04 hat: v3.89.0
Tibber: ja
Ergebnis: ständiges Starten und Stoppen - kommt nie in Lademodus

@chrostek
evcc: 0.122.1
Vestel evc04 hat: 3.118.0
Tibber: ja
Ergebnis: ständiges Starten und Stoppen - kommt nie in Lademodus

@ogheinz77
Vestel EVC04 Firmware: v3.118.0
EVCC: 0.122.1
Tibber: nein (noch nicht)
Ergebnis: es läuft

premultiply, 2023-11-22:
Dann ist der Gegentest ja ganz einfach:
Tibber aus der Konfig rauswerfen und z. B. gegen ein custom-const-Meter ersetzen.

ogheinz77, 2023-11-22:
Das wäre Top

ThomasFrank266, 2023-11-23:
Hallo zusammen,
ich habe genauso das Problem, dass seit dem Update auf die evcc Version 0.122.1 das PV Überschuss Laden nicht mehr funktioniert. Er toggelt ständig zwischen min-Wert 1,4 KW (einphasig) und 7 kW hin und her. Ich möchte gerne auf die Version 0.121.5 zurück, bei der noch alles wunderbar funktioniert hat. Leider bin ich generell Linux  und evcc Newbie und finde einfach keine passende Anleitung, wie ich das anstelle (Schritt für Schritt für Dummies). Kann mir hier bitte von den Pros einer weiter helfen? EVCC läuft bei mir auf Ubuntu

chrostek, 2023-11-23:
unter Ubuntu:
apt install evcc=0.121.5

ThomasFrank266, 2023-11-23:
...danke für die schnelle Antwort. Ich habe den Dienst über sudo systemctl stop evcc beendet und dann apt install evcc=0.121.5 versucht.

--> Antwort: 

: Sperrdatei /var/lib/dpkg/lock-frontend konnte nicht geöffnet werden. - open (13: Keine Berechtigung)
E: Erlangen der Sperre für die Dpkg-Oberfläche (/var/lib/dpkg/lock-frontend) nicht möglich; sind Sie root?

developer-stephan, 2023-11-23:
> Hallo zusammen, ich habe genauso das Problem, dass seit dem Update auf die evcc Version 0.122.1 das PV Überschuss Laden nicht mehr funktioniert. Er toggelt ständig zwischen min-Wert 1,4 KW [...]

Es toggelt zwischen 1 und 3 Phasen? Geht das schon? Hat jemand Erfahrungen? Da die Sonnenausbeute im Winter nicht so stark ist, würde ich gerne ab und zu 1 Phase anstatt 3 Phasen nutzen. Habe im Wallbox Firmware und im evcc auf 1 Phase eingestellt. Trotzdem blieb es bei 3 Phasen.

chrostek, 2023-11-23:
> ...danke für die schnelle Antwort. Ich habe den Dienst über sudo systemctl stop evcc beendet und dann apt install evcc=0.121.5 versucht.
>
> --> Antwort:
>
> : Sperrdatei /var/lib/dpkg/lock-frontend [...]

okay, ich arbeite immer als root, du scheinbar nicht, dann also:
sudo apt install evcc=0.121.5

ThomasFrank266, 2023-11-23:
klasse, danke Dir...wie gesagt...am besten für Dummies. Muss ich die config jetzt auch nochmal neu machen? Beim aufrufen von 192.168.xxx.xxx:7070 komme ich nun nicht mehr auf die GUI

ThomasFrank266, 2023-11-23:
> > Hallo zusammen, ich habe genauso das Problem, dass seit dem Update auf die evcc Version 0.122.1 das PV Überschuss Laden nicht mehr funktioniert. Er toggelt ständig zwischen min-Wert 1,4 KW (einphasig) und 7 kW hin und her. Ich möchte gerne auf die Version 0.121.5 zurück, bei der noch alles wunderbar funktioniert hat. Leider bin ich generell Linux und evcc Newbie und finde einfach keine passende Anleitung, wie ich das anstelle (Schritt für Schritt für Dummies). Kann mir hier bitte von den Pros einer weiter helfen? EVCC läuft bei mir auf Ubuntu
> 
> Es toggelt zwischen 1 und 3 Phasen? Geht das schon? Hat jemand Erfahrungen? Da die Sonnenausbeute im Winter nicht so stark ist, würde ich gerne ab und zu 1 Phase anstatt 3 Phasen nutzen. Habe im Wallbox und im evcc auf 1 Phase eingestellt. Trotzdem blieb es bei 3 Phasen.

...nein, er springt zwischen 1,4Kw und ca 7kW bei einphasiger Einstellung über das Webinterphase der Vestel. Bei mir hatte die Umstellung auf 1 Phase funktioniert...positiver Nebeneffekt ist natürlich, dass die Regelstufen feiner werden, wenn 230V nicht mehr immer x3 für 3phasen als Regelstufe verwendet werden

chrostek, 2023-11-23:
> klasse, danke Dir...wie gesagt...am besten für Dummies. Muss ich die config jetzt auch nochmal neu machen? Beim aufrufen von 192.168.xxx.xxx:7070 komme ich nun nicht mehr auf die GUI

wenn du ihn vor dem downgrade mit

> sudo systemctl stop evcc

beendet hast, musst du ihn jetzt natürlich auch wieder mit

> sudo systemctl start evcc

starten. Wenn er vorher gestartet war, würde er sich nach dem Downgrade auch wieder selbst starten. An der Config muss man nichts ändern.

developer-stephan, 2023-11-23:
> ...nein, er springt zwischen 1,4Kw und ca 7kW bei einphasiger Einstellung über das Webinterphase der Vestel. Bei mir hatte die Umstellung auf 1 Phase funktioniert...positiver Nebeneffekt ist [...]

Vestel Webinterface: 1 Phase
evcc Config: auch 1 Phase? Oder auto / 3 Phasen?

ThomasFrank266, 2023-11-23:
> wenn du ihn vor dem downgrade mit
>
>
> beendet hast, musst du ihn jetzt natürlich auch wieder mit
>
>
> starten. Wenn er vorher gestartet war, würde er sich nach dem Downgrade auch wieder selbst starten. [...]

o.k, das habe ich gemacht, dennoch komme ich nicht mehr auf das UI. Habe in der Zwischenzeit die yaml neu gemacht ,da hat alles funktioninert, nur am Schluss kam diese Meldung, wo wohl der Hinweis zum Fehler drin steckt, dass ich nicht mehr auf die UI komme

[server] FATAL 2023/11/23 19:07:46 unlinkat /tmp/evcc: operation not permitted

ThomasFrank266, 2023-11-23:
> > > > Hallo zusammen, ich habe genauso das Problem, dass seit dem Update auf die evcc Version 0.122.1 das PV Überschuss Laden nicht mehr funktioniert. Er toggelt ständig zwischen min-Wert 1,4 KW (einphasig) und 7 kW hin und her. Ich möchte gerne auf die Version 0.121.5 zurück, bei der noch alles wunderbar funktioniert hat. Leider bin ich generell Linux und evcc Newbie und finde einfach keine passende Anleitung, wie ich das anstelle (Schritt für Schritt für Dummies). Kann mir hier bitte von den Pros einer weiter helfen? EVCC läuft bei mir auf Ubuntu
> > > 
> > > 
> > > Es toggelt zwischen 1 und 3 Phasen? Geht das schon? Hat jemand Erfahrungen? Da die Sonnenausbeute im Winter nicht so stark ist, würde ich gerne ab und zu 1 Phase anstatt 3 Phasen nutzen. Habe im Wallbox und im evcc auf 1 Phase eingestellt. Trotzdem blieb es bei 3 Phasen.
> > 
> > 
> > ...nein, er springt zwischen 1,4Kw und ca 7kW bei einphasiger Einstellung über das Webinterphase der Vestel. Bei mir hatte die Umstellung auf 1 Phase funktioniert...positiver Nebeneffekt ist natürlich, dass die Regelstufen feiner werden, wenn 230V nicht mehr immer x3 für 3phasen als Regelstufe verwendet werden
> 
> Vestel Webinterface: 1 Phase evcc Config: auch 1 Phase? Oder 0 (auto) / 3 Phasen?

Vestel Web 1 Phase / evcc Config meine steht auf 0 (Auto)

developer-stephan, 2023-11-23:
> o.k, das habe ich gemacht, dennoch komme ich nicht mehr auf das UI. Habe in der Zwischenzeit die yaml neu gemacht ,da hat alles funktioninert, nur am Schluss kam diese Meldung, wo wohl der Hinweis [...]

Es klingt wieder nach Berechtigung. Mit sudo gestartet?

ThomasFrank266, 2023-11-23:
> Es klingt wieder nach Berechtigung. Mit sudo gestartet?

ja, kann mir das gerade auch nicht erklären. Habe folgendes eingegeben
thomas@thomas-OptiPlex-3070:~$ sudo systemctl start evcc

PS: Vielen Dank für Deine Geduld

ThomasFrank266, 2023-11-23:
> ja, kann mir das gerade auch nicht erklären. Habe folgendes eingegeben thomas@thomas-OptiPlex-3070:~$ sudo systemctl start evcc
>
> PS: Vielen Dank für Deine Geduld

über die status Abfrage bekomme ich

sudo systemctl status evcc
● evcc.service - evcc
     Loaded: loaded (/lib/systemd/system/evcc.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Thu 2023-11-23 20:10:11 CET; 967ms ago
    Process: 11228 ExecStart=/usr/bin/evcc (code=exited, status=1/FAILURE)
   Main PID: 11228 (code=exited, status=1/FAILURE)
        CPU: 268ms

developer-stephan, 2023-11-23:
Hoffen wir nun darauf, dass jemand den Bug in Version 0.122.1 beheben kann :)

premultiply, 2023-11-23:
Derzeit bin ich eher geneigt hier einfach zuzumachen.

chrostek, 2023-11-23:
> Derzeit bin ich eher geneigt hier einfach zuzumachen.

?! Das Problem ist bei mehreren reproduzierbar. @andig hat seit gestern Nachmittag einen SSH Zugriff auf mein System und kann - wenn er Zeit findet - jederzeit an meiner Vestel Wallbox testen und debuggen. Ich hoffe doch sehr, dass wir der Lösung damit näher kommen. Aus anderen Issues liest man heraus, dass das Hauptproblem ist, dass der Code nicht getestet werden kann, weil keiner der Entwickler eine Wallbox dieses Typs zur Verfügung hat. Diesen Mangel haben wir ja jetzt denke ich beseitigt.

andig, 2023-11-24:
Da die Diskussion es geschafft hat sich kaum noch ums Issue zu drehen mache ich hier zu Fehlersuche läuft mit Hilfe von @chrostek weiter.

andig, 2023-11-27:
@chrostek klappt
@premultiply 2002==0. Da wäre Timeout jetzt 10s. Was passt daran nicht?

premultiply, 2023-11-27:
0s ist etwas wenig.
Dann läuft die Box immer im Failsafe-Modus?

andig, 2023-11-27:
Bei 0s ist der Default 20s laut Doku. Wir schreiben dann alle 10s den keep-alive Wert von 1 in Reg. 6000. Ich sehe keinen Fehler?

andig, 2023-11-27:
@chrostek bitte nochmal testen (aus master bauen oder auf nightly warten). Ich hab alle Timeouts nochmal halbiert da die Doku etwas- sagen wir- indifferent ist.

poozu, 2023-11-27:
Bei mir läuft die Box mit 121.5 problemlos, mit 122.1 allerdings nicht mehr und zeigt ebenfalls das Start/Stopp Verhalten:

@poozu
evcc: 0.122.1
Vestel evc04 hat: 3.118.0
Tibber: nein
Ergebnis: ständiges Starten und Stoppen - kommt nie in Lademodus

[4 more replies]

closed 2023-11-28 by andig