evcc-io/evcc #10832: VW Passat GTE Passat Offline

norbertv, 2023-11-19:
### Describe the bug

Seit letzter Woche wird mein VW Passat GTE immer als "Offline" angezeigt. Auch ein Upgrade auf Version 0.122.1 zeigt das gleiche Verhalten. Im Log wird nach dem Start des EVCC Service immer die Fehlermeldung "ERROR 2023/11/19 11:50:08 creating vehicle ev4 failed: cannot create vehicle 'template': cannot create vehicle 'vw': unexpected status: 401 (Unauthorized)" angezeigt. Ein Login auf der VW MyVolkswagen Seite und in der Volkswagen APP funktioniert ohne Probleme. Hier gab es auch nichts zu bestätigen, was sonst immer zu solchen Problemen geführt hat.

### Steps to reproduce

Neustart EVCC Service

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  vw: trace

# unique installation id
plant: xxxxxxxxxxx

interval: 10s # control cycle interval

sponsortoken: xxxxxxxxxxxxxxxxxxxxxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.0.58
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.0.58
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.0.58
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3

chargers:
- type: template
  template: heidelberg
  id: 1
  device: /dev/ttyUSB_WB
  baudrate: 19200
  comset: 8E1
  modbus: rs485serial
  name: wallbox5

vehicles:
[... 140 lines cut ...]
Nov 19 11:50:28 openhabian evcc[24716]: [lp-1  ] DEBUG 2023/11/19 11:50:28 charger status: B
Nov 19 11:50:28 openhabian evcc[24716]: [lp-1  ] DEBUG 2023/11/19 11:50:28 pv charge current: 0A = 0A + -0.109A (75W @ 3p)
Nov 19 11:50:38 openhabian evcc[24716]: [site  ] DEBUG 2023/11/19 11:50:38 ----
Nov 19 11:50:38 openhabian evcc[24716]: [lp-1  ] DEBUG 2023/11/19 11:50:38 charge power: 0W
Nov 19 11:50:38 openhabian evcc[24716]: [site  ] DEBUG 2023/11/19 11:50:38 pv power: 4579W
Nov 19 11:50:38 openhabian evcc[24716]: [site  ] DEBUG 2023/11/19 11:50:38 battery soc: 15%
Nov 19 11:50:38 openhabian evcc[24716]: [site  ] DEBUG 2023/11/19 11:50:38 battery power: 0W
Nov 19 11:50:38 openhabian evcc[24716]: [site  ] DEBUG 2023/11/19 11:50:38 grid power: 66W
Nov 19 11:50:38 openhabian evcc[24716]: [site  ] DEBUG 2023/11/19 11:50:38 site power: 166W
Nov 19 11:50:38 openhabian evcc[24716]: [lp-1  ] DEBUG 2023/11/19 11:50:38 charge voltages: [238 237 238]V
Nov 19 11:50:38 openhabian evcc[24716]: [lp-1  ] DEBUG 2023/11/19 11:50:38 detected connected phases: 3p
Nov 19 11:50:38 openhabian evcc[24716]: [lp-1  ] DEBUG 2023/11/19 11:50:38 charge currents: [0 0 0]A
Nov 19 11:50:38 openhabian evcc[24716]: [lp-1  ] DEBUG 2023/11/19 11:50:38 charge total import: 68.143kWh
Nov 19 11:50:38 openhabian evcc[24716]: [lp-1  ] DEBUG 2023/11/19 11:50:38 charger status: B
Nov 19 11:50:38 openhabian evcc[24716]: [lp-1  ] DEBUG 2023/11/19 11:50:38 pv charge current: 0A = 0A + -0.241A (166W @ 3p)
Nov 19 11:50:48 openhabian evcc[24716]: [site  ] DEBUG 2023/11/19 11:50:48 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.122.1

andig, 2023-11-19:
Bitte Zugangsdaten an info@evcc.io

TobiasHuber1980, 2023-11-19:
passt doch zu [#10820 ](https://github.com/evcc-io/evcc/discussions/10820#discussioncomment-7605416)

andig, 2023-11-19:
Absolut. Da wir nicht wissen, ob das für *alle* VW gilt erstmal wontfix damit wir sonst nix kaputt machen.

closed 2023-11-19 by andig

norbertv, 2023-11-19:
habe jetzt mal das Template id ausprobiert. War ein voller Erfolg !!!
Vielen Dank für die schnelle Hilfe

lordvalium, 2024-08-28:
Also mein GTE ist nicht offline aber ich sehe den SOC nicht in meinem evcc. An was kann das liegen?