evcc-io/evcc #10834: Tariff: Energinet returns HTTP Bad Request

MartinHave, 2023-11-19:
### Describe the bug

Currently, when enabling the `energinet` tariff type, no energy prices will be shown in the UI as the Energinet API returns HTTP Bad Request on requests to it.

This is due to an invalid format of the `start`/`end` query parameters in the current API call. 

Calling the API manually as follows: `https://api.energidataservice.dk/dataset/Elspotprices?offset=0&start=2023-11-19T13:00:00+01:00&end=2023-11-20T13:00:00+01:00&filter={%22PriceArea%22:[%22dk1%22]}&timezone=dk&limit=48`

Returns the following error:

```
Invalid datetime format for [end] - valid format: yyyy, yyyy-MM, yyyy-MM-dd, yyyy-MM-ddTHH:mm.
```

Removing the seconds and time zone parts yields the following successful result: `https://api.energidataservice.dk/dataset/Elspotprices?offset=0&start=2023-11-19T13:00&end=2023-11-20T13:00&filter={%22PriceArea%22:[%22dk1%22]}&timezone=dk&limit=48`

```json
{"total":24,"filters":"{\"PriceArea\":[\"dk1\"]}","limit":48,"dataset":"Elspotprices","records":[{"HourUTC":"2023-11-20T11:00:00","HourDK":"2023-11-20T12:00:00","PriceArea":"DK1","SpotPriceDKK":768.280029,"SpotPriceEUR":103.000000},{"HourUTC":"2023-11-20T10:00:00","HourDK":"2023-11-20T11:00:00","PriceArea":"DK1","SpotPriceDKK":790.650024,"SpotPriceEUR":106.000000},{"HourUTC":"2023-11-20T09:00:00","HourDK":"2023-11-20T10:00:00","PriceArea":"DK1","SpotPriceDKK":836.750000,"SpotPriceEUR":112.180000},{"HourUTC":"2023-11-20T08:00:00","HourDK":"2023-11-20T09:00:00","PriceArea":"DK1","SpotPriceDKK":883.440002,"SpotPriceEUR":118.440002},{"HourUTC":"2023-11-20T07:00:00","HourDK":"2023-11-20T08:00:00","PriceArea":"DK1","SpotPriceDKK":944.830017,"SpotPriceEUR":126.669998},{"HourUTC":"2023-11-20T06:00:00","HourDK":"2023-11-20T07:00:00","PriceArea":"DK1","SpotPriceDKK":955.419983,"SpotPriceEUR":128.089996},{"HourUTC":"2023-11-20T05:00:00","HourDK":"2023-11-20T06:00:00","PriceArea":"DK1","SpotPriceDKK":766.409973,"SpotPriceEUR":102.750000},{"HourUTC":"2023-11-20T04:00:00","HourDK":"2023-11-20T05:00:00","PriceArea":"DK1","SpotPriceDKK":599.929993,"SpotPriceEUR":80.430000},{"HourUTC":"2023-11-20T03:00:00","HourDK":"2023-11-20T04:00:00","PriceArea":"DK1","SpotPriceDKK":483.040009,"SpotPriceEUR":64.760002},{"HourUTC":"2023-11-20T02:00:00","HourDK":"2023-11-20T03:00:00","PriceArea":"DK1","SpotPriceDKK":447.760010,"SpotPriceEUR":60.029999},{"HourUTC":"2023-11-20T01:00:00","HourDK":"2023-11-20T02:00:00","PriceArea":"DK1","SpotPriceDKK":446.869995,"SpotPriceEUR":59.910000},{"HourUTC":"2023-11-20T00:00:00","HourDK":"2023-11-20T01:00:00","PriceArea":"DK1","SpotPriceDKK":467.529999,"SpotPriceEUR":62.680000},{"HourUTC":"2023-11-19T23:00:00","HourDK":"2023-11-20T00:00:00","PriceArea":"DK1","SpotPriceDKK":509.450012,"SpotPriceEUR":68.300003},{"HourUTC":"2023-11-19T22:00:00","HourDK":"2023-11-19T23:00:00","PriceArea":"DK1","SpotPriceDKK":597.090027,"SpotPriceEUR":80.050003},{"HourUTC":"2023-11-19T21:00:00","HourDK":"2023-11-19T22:00:00","PriceArea":"DK1","SpotPriceDKK":656.760010,"SpotPriceEUR":88.050003},{"HourUTC":"2023-11-19T20:00:00","HourDK":"2023-11-19T21:00:00","PriceArea":"DK1","SpotPriceDKK":629.390015,"SpotPriceEUR":84.379997},{"HourUTC":"2023-11-19T19:00:00","HourDK":"2023-11-19T20:00:00","PriceArea":"DK1","SpotPriceDKK":661.239990,"SpotPriceEUR":88.650002},{"HourUTC":"2023-11-19T18:00:00","HourDK":"2023-11-19T19:00:00","PriceArea":"DK1","SpotPriceDKK":774.390015,"SpotPriceEUR":103.820000},{"HourUTC":"2023-11-19T17:00:00","HourDK":"2023-11-19T18:00:00","PriceArea":"DK1","SpotPriceDKK":779.840027,"SpotPriceEUR":104.550003},{"HourUTC":"2023-11-19T16:00:00","HourDK":"2023-11-19T17:00:00","PriceArea":"DK1","SpotPriceDKK":708.750000,"SpotPriceEUR":95.019997},{"HourUTC":"2023-11-19T15:00:00","HourDK":"2023-11-19T16:00:00","PriceArea":"DK1","SpotPriceDKK":604.030029,"SpotPriceEUR":80.980003},{"HourUTC":"2023-11-19T14:00:00","HourDK":"2023-11-19T15:00:00","PriceArea":"DK1","SpotPriceDKK":502.070007,"SpotPriceEUR":67.309998},{"HourUTC":"2023-11-19T13:00:00","HourDK":"2023-11-19T14:00:00","PriceArea":"DK1","SpotPriceDKK":473.350006,"SpotPriceEUR":63.459999},{"HourUTC":"2023-11-19T12:00:00","HourDK":"2023-11-19T13:00:00","PriceArea":"DK1","SpotPriceDKK":462.829987,"SpotPriceEUR":62.049999}]}
```

### Steps to reproduce

1. Add tariff type: energinet to your configuration
3. Start EVCC
4. Observe the Bad Request log line

### Configuration details

```yaml
tariffs:
  currency: DKK
  grid:
    type: energinet
    region: dk1
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/11/19 13:45:10 evcc 0.122.1
[main  ] INFO 2023/11/19 13:45:10 using config file: /config/evcc.yaml
[main  ] INFO 2023/11/19 13:45:10 starting ui and api at :7070
[db    ] INFO 2023/11/19 13:45:11 using sqlite database: /data/evcc.db
[main  ] ERROR 2023/11/19 13:45:52 failed configuring grid tariff: cannot create tariff 'energinet': unexpected status: 400 (Bad Request)
[energinet] ERROR 2023/11/19 13:45:52 unexpected status: 400 (Bad Request)
[energinet] ERROR 2023/11/19 13:46:38 unexpected status: 400 (Bad Request)
[main  ] ERROR 2023/11/19 13:46:38 failed configuring feed-in tariff: cannot create tariff 'energinet': unexpected status: 400 (Bad Request)
[lp-1  ] DEBUG 2023/11/19 13:46:39 charge total import: 474.335kWh
[site  ] INFO 2023/11/19 13:46:39 site config:
[site  ] INFO 2023/11/19 13:46:39   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/11/19 13:46:39     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/11/19 13:46:39     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/11/19 13:46:39     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2023/11/19 13:46:39   vehicles:
[site  ] INFO 2023/11/19 13:46:39     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/11/19 13:46:39 loadpoint 1:
[lp-1  ] INFO 2023/11/19 13:46:39   mode:        pv
[lp-1  ] INFO 2023/11/19 13:46:39   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/11/19 13:46:39   meters:      charge ✓
[lp-1  ] INFO 2023/11/19 13:46:39     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/11/19 13:46:39 phase timer inactive
[lp-1  ] DEBUG 2023/11/19 13:46:39 pv timer inactive
[lp-1  ] DEBUG 2023/11/19 13:46:39 guard timer inactive
[site  ] DEBUG 2023/11/19 13:46:39 ----
[lp-1  ] DEBUG 2023/11/19 13:46:39 charge power: 0W
[site  ] DEBUG 2023/11/19 13:46:39 pv power: 168W
[site  ] DEBUG 2023/11/19 13:46:39 battery soc: 74%
[site  ] DEBUG 2023/11/19 13:46:39 battery power: 80W
[site  ] DEBUG 2023/11/19 13:46:39 grid power: 77W
[site  ] DEBUG 2023/11/19 13:46:39 site power: 257W
[lp-1  ] DEBUG 2023/11/19 13:46:39 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/11/19 13:46:39 charge total import: 474.335kWh
[lp-1  ] DEBUG 2023/11/19 13:46:39 charger status: B
[lp-1  ] DEBUG 2023/11/19 13:46:39 charge start energy: 474kWh
[lp-1  ] INFO 2023/11/19 13:46:39 car connected
[lp-1  ] DEBUG 2023/11/19 13:46:39 vehicle api refresh
[lp-1  ] DEBUG 2023/11/19 13:46:39 pv timer elapse
[lp-1  ] DEBUG 2023/11/19 13:46:39 pv timer inactive
[lp-1  ] DEBUG 2023/11/19 13:46:39 charger: guard elapse
[lp-1  ] DEBUG 2023/11/19 13:46:39 guard timer inactive
[lp-1  ] DEBUG 2023/11/19 13:46:39 charge total import: 474.335kWh
[lp-1  ] DEBUG 2023/11/19 13:46:39 charger vehicle id: F0F01FA34FCFDC
[main  ] ERROR 2023/11/19 13:46:39 vehicle status: asleep
[lp-1  ] DEBUG 2023/11/19 13:46:39 pv charge current: 0A = 0A + -0.372A (257W @ 3p)
[lp-1  ] DEBUG 2023/11/19 13:46:39 pv enable timer reset
[lp-1  ] DEBUG 2023/11/19 13:46:39 pv timer inactive
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

v0.122.1

andig, 2023-11-19:
Would be nice if we could do without the additional timezone. Wondering, if/why this worked before?

MartinHave, 2023-11-19:
> Would be nice if we could do without the additional timezone. Shouldn’t region suffice?
> 
> Wondering, if/why this worked before?

I can tell that this has worked for some time so it could be a change that Energinet added. I have not found any info to back

An-Wi, 2023-11-19:
Habe heute das gleiche Problem mit „grünstromindex“.
Allerdings bekomme ich ein status 500:
`[main  ] ERROR 2023/11/19 10:12:14 failed configuring co2 tariff: cannot create tariff ‘grünstromindex‘: unexpected status: 500 (Internal Server Error)`

andig, 2023-11-19:
@An-Wi da der Status vom Server kommt zwangsläufig ein anderes Problem.

andig, 2023-11-20:
Closed in https://github.com/evcc-io/evcc/commit/22859abd132902147662451445e860060cec9ce4

closed 2023-11-20 by andig