evcc-io/evcc #10837: connection to vw doesn't work anymore

Bullischorsch, 2023-11-19:
### Describe the bug

Since last night I realize, that my e-up! is shown as offline. 

### Steps to reproduce

Maybe there are issues on VW server side? The WeConnect service has been shut down by VW some days ago, but my account on void.vwgroup.io and Volkswagen-App are working fine.

### Configuration details

```yaml
vehicles:
  - name: ulmenup
    type: template
    template: vw
    title: ulmenup # Wird in der Benutzeroberfläche angezeigt # Optional
    user: myMail # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: myPassword # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: WVW--------------- # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 32 # Akkukapazität in kWh # Optional 
    mode: minpv
    minsoc: 70
    targetSoc: 90
    phases: 2
```

### Log details

```text
[main  ] ERROR 2023/11/19 14:26:09 creating vehicle ulmenup failed: cannot create vehicle 'template': cannot create vehicle 'vw': unexpected status: 401 (Unauthorized)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.122.1

VolkerK62, 2023-11-19:
try `template: id` 
https://github.com/evcc-io/evcc/issues/10832

Bullischorsch, 2023-11-19:
Great! Works for e-up! 2023.

closed 2023-11-19 by Bullischorsch