evcc-io/evcc #10839: SEMP_BASE_URL unspecified

oskarfessel, 2023-11-19:
### Discussed in https://github.com/evcc-io/evcc/discussions/4708

<div type='discussions-op-text'>

<sup>Originally posted by **flo9127** October  3, 2022</sup>
### Describe the bug

Hallo,
ich bekomme immer die Meldung 
`SEMP_BASE_URL unspecified, using http://172.27.128.1:8181 instead`
beim Start vom EVCC

### Version

0.104.2</div>

While this does not affect most people, i think it is just wrong to rely on environment variables when there is config file at work for other parameters.  
And, in my case it makes the SEMP not really usable, because the SEMP needs to use the network interface with the Home-Manager in it and evcc wisely chooses the other interface.  Everytime.  Since evcc needs to be started automatically, this is a problem - it should be configurable in the yaml, maybe just through an interface: statement.

####
[main  ] INFO 2023/11/19 17:33:01 evcc 0.122.1 (e7e78bea)
[main  ] INFO 2023/11/19 17:33:01 using config file: /etc/evcc.yaml
[main  ] INFO 2023/11/19 17:33:01 starting ui and api at :7070
[db    ] INFO 2023/11/19 17:33:01 using sqlite database: ****/.evcc/evcc.db
[lp-1  ] WARN 2023/11/19 17:33:04 vehicles option is deprecated
[site  ] WARN 2023/11/19 17:33:04 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[semp  ] WARN 2023/11/19 17:33:04 SEMP_BASE_URL unspecified, using http://83.*.*.135:7070 instead

andig, 2023-11-20:
> While this does not affect most people, i think it is just wrong to rely on environment variables when there is config file at work for other parameters.

Du schlägst vor, statt ENV ein Setting zu spendieren? Spricht nichts dagegen, PR welcome.

> evcc wisely chooses the other interface. Everytime. 

Genau deshalb gibt es diese Warning die Dir die möglichkeit gibt die richtige URL zu vergeben.

> maybe just through an interface: statement.

Wie würde man die IP fürs Interface bestimmen? Wie würde das bei einem Docker Setup helfen?

oskarfessel, 2023-11-20:
Yes, PR.  ich fürchte, ich muss das hier noch lernen, wie das hier geht.

> > evcc wisely chooses the other interface. Everytime.
> Genau deshalb gibt es diese Warning die Dir die möglichkeit gibt die richtige URL zu vergeben.

Genau das ist ja das Problem, dass die URL aus dem falschen Interface generiert wird.  Und ich müßte im Environment die zugeteilte IP-Adresse herausfinden, um die Variable richtig zu setzen.
Ich gebe zu, der Usecase ist speziell, weil ich evcc auf NetBSD laufen lasse, ohne Docker.  Und da kann ich das Environment derzeit nicht dynamisch anpassen, denn das ist pro Prozess statisch einzustellen.
Das wäre mir nicht aufgefallen, wenn mich die Mac-Version nicht mit jedem Update zur Verzweiflung treiben würde...
Ich schau mal, wie des die anderen Teile mit "interface:" Spezifikation im YAML machen.  sollte doch hier genau so funktionieren.

andig, 2023-11-20:
Gerne PR wenn Du soweit bist.

closed 2023-11-20 by andig

litithio, 2023-12-28:
> > evcc wisely chooses the other interface. Everytime.
> 
> Genau deshalb gibt es diese Warning die Dir die möglichkeit gibt die richtige URL zu vergeben.

Sorry, dass ich das nochmal auf mache.
Bei mir und meinem Bruder tritt dieselbe Fehlermeldung/dasselbe Problem auf. 
Wir haben evcc mittels Docker-Compose auf einer Synology laufen. Da wir beide SMA-Komponenten Nutzen, läuft der Container im Host Modus.
Auf der Synology habe ich noch mehrere andere Container in einem macvlan laufen, welche über das zweite Netzwerkinterface in ein VLAN gehen.
Wie bei @oskarfessel wählt evcc stets das falsche Interface für die SEMP_BASE_URL. 
An welcher Stelle kann ich denn die SEMP_BASE_URL festlegen? In der Doku konnte ich dazu nichts finden.

andig, 2023-12-28:
Das ist einfach eine Umgebungsvariable, keine Magie dahinter.

litithio, 2023-12-28:
Danke für deine schnelle Antwort.
Ich setze also "SEMP_BASE_URL" als ENV für den Container und als Wert muss ich was setzen?
http://IpDerSynology:7070 oder was wäre hier korrekt?

andig, 2023-12-28:
Da muss der Wert rein, unter dem der SHM evcc erreicht. Was auch immer das sein mag.