evcc-io/evcc #10851: make doc fails with error regarding enumer

PeterPablo, 2023-11-20:
### Describe the bug

```
 make docs
go generate github.com/evcc-io/evcc/util/templates/...
util/templates/class.go:5: running "enumer": exec: "enumer": executable file not found in $PATH
```

### Steps to reproduce

1. install brew
2. install go and node, as described in readme.md
3. compile branch #10847 
4. try to compile documents, observe error

### Configuration details

```yaml
Mac OS
```

### Log details

```text
not relevant
```

### What type of operating system are you running?

Linux

### Version

https://github.com/evcc-io/evcc/pull/10847/commits/82c36b376dd4d83c4db06523faa920a0d672f4a1

PeterPablo, 2023-11-20:
To clarify: Compilation of the application and running `./evcc` works fine. It is building the documents (which I tried out of curiosity) that fails.

andig, 2023-11-20:
Erst

    make install

closed 2023-11-20 by andig