evcc-io/evcc #10854: Klimatisierung (Climater) gibt Ladepunkt nicht frei

mdkeil, 2023-11-21:
### Describe the bug

Ich habe heute erstmalig wieder die Vorklimatisierung bei meinem e208 aktiviert und musste feststellen, dass evcc  die Vorklimatisierung nicht erkannt hat und daher auch den Lp nicht freigegeben hat.

Eine Abfrage via `evcc vehicle` zeigt aber `Climate active: true` .. Ich habe mal testweise das logging auf debug gesetzt, konnte aber keine relevanten Einträge finden.

Ich nutze für das Fahrzeug das [flobz-template](https://github.com/evcc-io/evcc/blob/master/templates/definition/vehicle/flobz.yaml).

### Steps to reproduce

1. Auto an Lp anschließen (Status B)
2. (Vor) Klimatisierung einschalten
3. Abwarten, ob evcc die erkannte Klimatisierung im UI anzeigt

### Configuration details

```yaml
nicht relevant
```

### Log details

```text
keine relevanten Einträge bis auf:

[site  ] INFO 2023/11/21 06:55:25     vehicle 3: range ✓ finish ✗ status ✓ climate ✓ wakeup ✓
```

### What type of operating system are you running?

Docker container

### Version

0.122.1

premultiply, 2023-11-21:
Poll-Intervall & -Modus berücksichtigt und ggf. entsprechend angepasst?
`evcc vehicle` startet eine neue Instanz...

mdkeil, 2023-11-21:
Poll-Intervall steht auf default, Modus auf PV.. Fahrzeug wurde in der Nacht via Ladeplaner geladen-- Ich habe aber auch kurzfristig heute morgen (während der Vorklimatisierung) auf 0.122.1 geupdated und dementsprechend evcc neugestartet.. spätestens da hätte die Klimatisierung erkannt werden müssen.