evcc-io/evcc #10864: Add Delta Wallbox

premultiply, 2023-11-22:
Not sure whether its still of interest, but I recently integrated my Delta Basic WB into Home Assistant via ModBus: https://www.photovoltaikforum.com/thread/209699-delta-ac-max-basic-wallbox-modus-rtu-rs485-wie

You dont need the small board which in addition gives you less options than the control via ModBus:

[Installation_introduction_ ACMax_basic_v0.2.1.pdf](https://github.com/evcc-io/evcc/files/13249371/Installation_introduction_.ACMax_basic_v0.2.1.pdf)

_Originally posted by @MonsieurZorglub in https://github.com/evcc-io/evcc/discussions/2860#discussioncomment-7465865_

MonsieurZorglub, 2023-11-22:
direct link to the yaml: https://www.photovoltaikforum.com/thread/209699-delta-ac-max-basic-wallbox-modus-rtu-rs485-wie/?postID=3446900#post3446900

HTH

TweetsOfNiklas, 2023-12-22:
Since I ordered 2 of these Wallboxes last week, I would be interested in an integration in EVCC. Is there a similar currently supported Wallbox? Because then I could maybe try to get them to work and test them.

premultiply, 2023-12-22:
Maybe you can share the Modbus port of at least one of the boxes to us?
That would speed up implementation and testing. 🎅🏻🎄

Please send an email with the connection details to info@evcc.io or send a pm to me at our Slack channel. Thx.

mirgonet, 2023-12-22:
I also own two AC MAX Basics and already started implementing a plugin for EVCC. Was able to sucessfully charge my EV yesterday while managing power limits via EVCC but there is still some work to do. I think it can be improved by integrating additional registers that are not described in the documentation I have.

Will send a wip pull request shortly. Feel free to comment or help to improve it (my first Go adventure ;) )

premultiply, 2023-12-23:
https://www.photovoltaikforum.com/core/attachment/408989-evcs-modbus-register-table-v1-1-working-1-pdf/

xpcone, 2023-12-27:
I got an  AC MAX Smart here, which i want to use with evcc.
Trying OCCP first until end of the year. Cabling will only provide 13A 1p until the electrican brings the new cable.

mirgonet, 2023-12-30:
@premultiply: I think this issue can be closed because the PR got merged today.
@xpcone: Would be nice if you could try out the plugin with your AC MAX Smart as I can only test with the AC MAX Basic here.

closed 2023-12-30 by premultiply