evcc-io/evcc #10881: Ecactus Inverter implementation for evcc

Duelmotor, 2023-11-23:
Hey ,how are you going,just hoping to ask for implementation for the ecactus inverter ,I have attached the API documentation 

[ecactus api_au_20231007.pdf](https://github.com/evcc-io/evcc/files/13448376/ecactus.api_au_20231007.pdf)

andig, 2023-11-24:
Looks like you should be able to do this using the custom meter with http plugin?

andig, 2023-12-06:
@Duelmotor do you want to try a PR for a new template?

Duelmotor, 2023-12-06:
What is a pr for a new template mean

VolkerK62, 2023-12-10:
@Duelmotor you can try to build a custom meter with the HTTP plugin.
If this works, you should post the config of this meter. With this information, a template can be build.

closed 2024-01-06 by github-actions[bot]

Seaniefield, 2024-04-22:
> @Duelmotor you can try to build a custom meter with the HTTP plugin. If this works, you should post the config of this meter. With this information, a template can be build.

@Duelmotor, Did you manage to complete this?