evcc-io/evcc #10885: Hardy Barth WB (template hardybarth-salia) - API Changes mit Fw 2.0?

mdkeil, 2023-11-23:
### Describe the bug

Seitdem ich die WB von Fw. 1.87 auf 2.0 upgedatet habe, wird das Log während aktiver Ladevorgänge mit folgendem Warning zugespamt: `charger logic error: disabled but charging`

Der Ladevorgang wird davon aber nicht negativ beeinflusst,.

Ich vermute, dass es ggfs. eine kleine API-Änderungen gegeben hat.

Anbei mal eine aktuelle API-Abfrage ohne aktiven Ladevorgang:

```
{
  "device": {
    "product": "2310007",
    "modelname": "Salia PLCC Slave",
    "hardware_version": "1.0",
    "software_version": "2.0.0",
    "vcs_version": "V0R5c",
    "hostname": "salia",
    "mac_address": "00:01:87:0c:7c:41",
    "serial": "100818241",
    "uuid": "5491ad62-022a-4356-a32c-0001870c7c41",
    "internal_id": "391222",
    "ip_lo": "127.0.0.1",
    "ip_br0:fallback": "169.254.12.53",
    "ip_br0": "192.168.1.11"
  },
  "secc": {
    "port0": {
      "salia": {
        "chargemode": "manual",
        "keyswitch": "0",
        "mem": "296792",
        "thermal": "51837",
        "uptime": "1 day",
        "thermalmax": "64689",
        "load": "0.33",
        "thermalmin": "0",
        "heartbeat": "alive",
        "modbus": {
          "status": "running"
        },
        "socketminamp": "6",
        "socketmaxamp": "16",
        "LM": {
          "internal": {
            "status": "internal lm not activated"
          }
        },
        "intctrl_limit": "16",
        "chargedata": "-",
        "ecoplus": "0",
        "mains_type": "none",
        "authmode": "free",
        "tc_enabled": "0",
        "firmwareprogress": "0",
        "firmwarestate": "idle",
        "status": "manual -\/-\/r -\/-\/- 0(6-16) ",
        "limit_tc": "32",
        "phase_switching": {
          "actual": "3",
          "status": "idle"
        },
        "extctrl_limit": "0",
        "pausecharging": "1"
      },
      "ci": {
        "evse": {
          "basic": {
            "grid_current_limit": {
[... 353 lines cut ...]
[lp-1  ] DEBUG 2023/11/23 03:01:09 charge currents: [14.6 14.7 15]A
[lp-1  ] DEBUG 2023/11/23 03:01:09 detected active phases: 3p
[lp-1  ] DEBUG 2023/11/23 03:01:09 charge total import: 5690.990kWh
[lp-1  ] DEBUG 2023/11/23 03:01:09 charger status: C
[lp-1  ] DEBUG 2023/11/23 03:01:09 vehicle status: C
[lp-1  ] DEBUG 2023/11/23 03:01:09 vehicle soc: 36%
[lp-1  ] DEBUG 2023/11/23 03:01:09 vehicle range: 66km
[lp-1  ] WARN 2023/11/23 03:01:09 charger logic error: disabled but charging
[salia ] TRACE 2023/11/23 03:01:09 PUT http://192.168.1.11/api/secc
[salia ] TRACE 2023/11/23 03:01:09 {"grid_current_limit":"16"}
--
result=ok 
[salia ] TRACE 2023/11/23 03:01:09 PUT http://192.168.1.11/api/secc
[salia ] TRACE 2023/11/23 03:01:09 {"salia/pausecharging":"0"}
--
result=ok
```

### What type of operating system are you running?

Docker container

### Version

0.122.1

dklinger, 2023-11-23:
+1
I'm having the same issue since upgrade to v2.

https://github.com/evcc-io/evcc/discussions/10875

andig, 2023-11-24:
Ich kann da nur raten. Wie muss `grid_current_limit` jetzt gesetzt und ausgelesen werden? Aus der API-Struktur alleine lässt sich das nicht erraten. Bitte Doku beim Hersteller anfragen.

dklinger, 2023-11-24:
Gesetzt wird es immer noch so wie vorher: INT-Wert setzt die zu nutzende Ampere. 0 deaktiviert die Ladung.
Wenn ich mich recht entsinne, konnte man über port0/ci/evse/basic/grid_current_limit/actual aber bisher den gesetzten Wert auch wieder auslesen. Das geht jetzt wohl nicht mehr.
Dafür gibt es jetz unter port0/salia zwei neue Werte: extcrtl_limit und manctrl_limit.
extctrl_limit => Wert, der mittels POST-request gesetzt wurde
manctrl_limit => Wert, der per Web-UI gesetzt wurde

So wie es aussieht, gewinnt der exctrl_limit. Um das zu prüfen, sollte man aber einfach den "offered_current_limit" unter port0/evse/basic/offered_current_limit auslesen.

[image]

Hier ein API-Auszug. 9 wurde per REST-API gesetzt, 10 per Web-UI. Geladen wird mit 9.

`{
    "port0": {
        "salia": {
            "chargemode": "manual",
            "keyswitch": "0",
            "mem": "359860",
            "thermal": "48097",
            "uptime": "8 min",
            "thermalmax": "50967",
            "load": "0.40",
            "thermalmin": "46375",
            "modbus": {
                "status": "running"
            },
            "socketminamp": "6",
            "socketmaxamp": "16",
            "intctrl_limit": "16",
            "chargedata": "480|6159|8403.22|",
            "ecoplus": "0",
            "mains_type": "none",
            "authmode": "free",
            "LM": {
                "internal": {
                    "status": "internal lm not activated"
                }
            },
            "phase_switching": {
                "actual": "3",
                "status": "idle"
            },
            "tc_enabled": "0",
            "firmwareprogress": "0",
            "firmwarestate": "idle",
            "status": "manual con/chg/r -/-/- 9(6-16) con;chg;.;Ovc(-)::",
            "pausecharging": "0",
            "limit_tc": "32",
            "extctrl_limit": "9",
            "manctrl_limit": "10"
        },
        "ci": {
            "evse": {
                "basic": {
                    "grid_current_limit": {
                        "actual": "16"
                    },
                    "phase_count": "3",
                    "physical_current_limit": "16",
                    "offered_current_limit": "9.0",
                    "current": {
                        "offered": "27.0",
                        "l1": {
                            "offered": "9.0"
                        },
                        "l2": {
                            "offered": "9.0"
                        },
                        "l3": {
                            "offered": "9.0"
[... 139 lines cut ...]
        },
        "availability": {
            "actual": "operative"
        },
        "cp": {
            "pwm_state": {
                "actual": "1"
            },
            "state": "C",
            "duty_cycle": "15.00"
        },
        "diode_present": "1",
        "cable_current_limit": "-1",
        "ready_for_slac": "0",
        "ev_present": "1",
        "charging": "1",
        "ventilation": {
            "state": {
                "actual": "0"
            },
            "available": "0"
        },
        "charging_stop": "0"
    }
}`

andig, 2023-11-24:
Bisher:

    	return res.Secc.Port0.GridCurrentLimit > 0 && res.Secc.Port0.Salia.PauseCharging == 0, err

jetzt:

	return res.Secc.Port0.Ci.Evse.Basic.OfferedCurrentLimit > 0 && res.Secc.Port0.Salia.PauseCharging == 0, err

PauseCharging bleibt am Platz?

andig, 2023-11-24:
Test wäre Klasse!

dklinger, 2023-11-24:
Wenn du mir sagst wie ich bei meinem docker-setup nen dev-branch testen kann, dann gerne :D
PauseCharging sollte drin und am Platz bleiben können. Sofern es auf 1 steht, hat jemand nen hartes Stop-Signal geschickt. In den Moment geht aber offered_current_limit auch direkt auf 0:

```json
{
    "port0": {
        "ci": {
            "evse": {
                "basic": {
                    "grid_current_limit": {
                        "actual": "16"
                    },
                    "phase_count": "3",
                    "physical_current_limit": "16",
                    "offered_current_limit": "0.0",
                    "current": {
                        "offered": "0.0",
                        "l1": {
                            "offered": "0.0"
                        },
                        "l2": {
                            "offered": "0.0"
                        },
                        "l3": {
                            "offered": "0.0"
                        }
                    },
                    "power": {
                        "offered": "0.0"
                    }
                },
                "phase": {
                    "actual": "3"
                }
            },
            "ev": {
                "soc": {
                    "actual": "-1",
                    "timestamp": "2023-11-24T17:59:56.326Z"
                }
            },
            "charge": {
                "cp": {
                    "status": "B"
                },
                "plug": {
                    "status": "locked"
                },
                "contactor": {
                    "status": "closed"
                },
                "pwm": {
                    "status": "100.00"
                }
            }
        },
        "salia": {
            "chargemode": "manual",
            "keyswitch": "0",
            "mem": "361024",
            "uptime": "4 min",
            "thermal": "48097",
            "load": "0.77",
            "thermalmax": "51541",
            "thermalmin": "47523",
            "modbus": {
                "status": "running"
            },
            "socketminamp": "6",
            "socketmaxamp": "16",
[... 131 lines cut ...]
            },
            "error": "0"
        },
        "cp": {
            "pwm_state": {
                "actual": "1"
            },
            "state": "B",
            "duty_cycle": "100.00"
        },
        "diode_present": "0",
        "cable_current_limit": "-1",
        "ready_for_slac": "0",
        "ev_present": "1",
        "charging": "1",
        "ventilation": {
            "state": {
                "actual": "0"
            },
            "available": "0"
        },
        "charging_stop": "1"
    }
}
```

StefanSchoof, 2023-11-24:
> Wenn du mir sagst wie ich bei meinem docker-setup nen dev-branch testen kann, dann gerne 

Ungefähr so:

- Auschecken
- `docker build --plattform linux/arm64 -t evcc .` (Plattform nach deinen Bedarf anpassen)
- Image auf Zielsystem übertragen, z.B mit https://stackoverflow.com/a/26226261

andig, 2023-11-25:
…und?

closed 2023-11-28 by andig

dklinger, 2023-11-28:
> …und?

Sorry, die aktuelle Kita-Krankheitswelle kam leider dazwischen.

andig, 2023-11-28:
Ab morgen im nightly- danke fürs testen!

mdkeil, 2023-11-30:
@andig 
hat funktioniert, Log wird nicht mehr mit den Fehlermeldungen zugmüllt.