evcc-io/evcc #10893: Bug in Ladeplanung ohne speziellen Tarif?

VolkerK62, 2023-11-24:
### Describe the bug

Ich hatte es bisher so verstanden, dass die Ladeplanung immer so spät wie möglich beginnt.
 
Wenn man einen CO2 Tarif konfiguriert hat, startet die Ladeplanung so spät wie möglich. 
Im Beispiel ist tagsüber durchgängig 51g und die Ladeplanung nimmt richtigerweise die letzten ("billigen") 5 Std.

[image]

Wenn ich den CO2 Tarif auskommentiere und nur noch der fixe Tarif konfiguriert ist, startet die Ladeplanung sofort.
**Sollte hier nicht auch die letzten 5 Std. ausgewählt werden?**

[image]

### Steps to reproduce

Vorgehen ist oben beschrieben

### Configuration details

```yaml
wird nachgereicht, wenn notwendig
```

### Log details

```text
wird nachgereicht, wenn notwendig
```

### What type of operating system are you running?

Linux

### Version

0.122.1

naltatis, 2023-11-24:
Ja, das war zumindest mein letzter Stand. Dass wir, wenn wir kein Preissignal haben immer möglichst kurz vor der Abfahrt laden.

N1c081, 2023-11-28:
Hallo,
das Problem (frühzeitiger Start der Ladung mit fixen Tarifen) von VolkerK62 kann ich zu 100% bestätigen. War bis vor Version 0.120 (denke ich) auch sehr verlässlich. Hatte das gestern Abend (21Uhr)nochmals getestet. Leider wieder sofortiger Start obwohl der Abfahrtszeitpunkt bei 7 Uhr am nächsten Tag geplant war. Ladedauer wären ca. 4h gewesen.

Sonst find ich die Software der Hammer und vielen Dank für eure Arbeit!

Version 0.122.1 läuft in einem Docker Container auf ner Synology

andig, 2023-11-28:
> War bis vor Version 0.120 (denke ich) auch sehr verlässlich.

@N1c081 könntest Du nochmal prüfen ob 120 da tatsächlich anders war?

andig, 2023-11-28:
@VolkerK62 lustigerweise kann ich das nicht nachvollziehen:

[image]

[image]

Stellt sich die Frage: was ist bei Dir/Euch anders?

RTTTC, 2023-11-28:
I think this is the difference?

> Wenn ich den CO2 Tarif auskommentiere und nur noch der fixe Tarif konfiguriert ist, startet die Ladeplanung sofort.

andig, 2023-11-28:
Dann probier doch mal bitte einen einfachen "fixed".

N1c081, 2023-11-28:
> > War bis vor Version 0.120 (denke ich) auch sehr verlässlich.
> 
> @N1c081 könntest Du nochmal prüfen ob 120 da tatsächlich anders war?

Mache ich die Tage mal und melde mich.

N1c081, 2023-11-28:
> Dann probier doch mal bitte einen einfachen "fixed".

Auch wenn das für VolkerK62 gedacht war.
Hier nochmal aus der evcc.yaml der tariffs Part bei mir:

[image]

andig, 2023-11-28:
Du hast co2, also ein ganz anderes Thema. Bitte nicht immer alles durcheinander werfen! Volker hat genau den (mit dem es geht) nicht!

VolkerK62, 2023-11-28:
>@andig mein fixer Tarif ist als Custom konfiguriert.

das ist es. 
Mit dem normalen fixed tariff werden die letzten Stunden genommen.
Wird klar, wenn man `evcc tariff` aufruft.
Mit dem custom traiff gibt es keine Liste, sondern nur einen Einzeiler.

```
pi@raspberrypi:~ $ evcc tariff
[main  ] INFO 2023/11/28 13:00:35 evcc 0.122.1
[main  ] INFO 2023/11/28 13:00:35 using config file: /etc/evcc.yaml
[db    ] INFO 2023/11/28 13:00:36 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/11/28 13:00:36 connecting evcc-1311805648 at tcp://localhost:1883
grid:
[mqtt  ] DEBUG 2023/11/28 13:00:36 tcp://localhost:1883 connected
[mqtt  ] DEBUG 2023/11/28 13:00:36 tcp://localhost:1883 subscribe Haus/Preis_Bezug
[mqtt  ] DEBUG 2023/11/28 13:00:36 Haus/Preis_Bezug wait for initial value
From                   To                     Price/Cost
2023-11-28 13:00:00    2023-11-30 13:00:00    0.354
```

andig, 2023-11-28:
Yep. Wir könnten aus dem einen Slot jetzt einfach 48 mit identischem Preis machen, das würde das Problem lösen.

VolkerK62, 2023-11-28:
wäre doch nur konsistent, oder?   Ist doch nix anderes als ein fixed tariff, der von außen gefüttert werden kann

RTTTC, 2023-11-28:
`evcc tariff` I did not know we had this tool available!

andig, 2023-11-30:
Closed in https://github.com/evcc-io/evcc/commit/c360cabdce0a678e6aef43996c323f9c2ddd4cf7

@RTTTC ever tried

    evcc --help

closed 2023-11-30 by andig