evcc-io/evcc #10896: Porsche API: Status: 502 (Bad Gateway)

MarkusGH, 2023-11-24:
### Describe the bug

Beim Initialisieren des Porsche API kommt:

`creating vehicle ev1 failed: cannot create vehicle 'template': cannot create vehicle 'porsche': unexpected status: 502 (Bad Gateway)`

### Steps to reproduce

Porsche Fahrzeug konfigurieren.
evcc (neu) starten.

### Configuration details

```yaml
Kann nachgereicht werden.
```

### Log details

```text
Nov 24 09:42:53 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:53 GET https://identity.porsche.com/authorize?audience=https%3A%2F%2Fapi.porsche.com&client_id=UYsK00My6bCqJdbQhTQ0PbWmcSdIAMig&code_challenge=lIvU6pJB2DbdxwB9r-GDp8qM_kp6WU5zIQU_CN0ydno&code_challenge_method=S256&redirect_uri=https%3A%2F%2Fmy.porsche.com%2F&response_typ
e=code&scope=openid+offline_access&state=MzxOXLkLCszeQBvc&ui_locales=de-DE
Nov 24 09:42:54 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:54 Found. Redirecting to /u/login/identifier?state=hKFo2SA3cUlqd1Y0bDJUSHVZcFhWMWVLaDRJNXZ2d0F4ekdNQaFur3VuaXZlcnNhbC1sb2dpbqN0aWTZIGdOdEtmUHJuQmR6Ym5PVXlKaVhtcURsV3h4Tlhpbm9Ho2NpZNkgVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWc&ui_locales=de
Nov 24 09:42:54 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:54 POST https://identity.porsche.com/u/login/identifier?state=hKFo2SA3cUlqd1Y0bDJUSHVZcFhWMWVLaDRJNXZ2d0F4ekdNQaFur3VuaXZlcnNhbC1sb2dpbqN0aWTZIGdOdEtmUHJuQmR6Ym5PVXlKaVhtcURsV3h4Tlhpbm9Ho2NpZNkgVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWc&ui_locales=de
Nov 24 09:42:54 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:54 action=default&is-brave=false&js-available=true&state=hKFo2SA3cUlqd1Y0bDJUSHVZcFhWMWVLaDRJNXZ2d0F4ekdNQaFur3VuaXZlcnNhbC1sb2dpbqN0aWTZIGdOdEtmUHJuQmR6Ym5PVXlKaVhtcURsV3h4Tlhpbm9Ho2NpZNkgVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWc&ui_locales=de&username=***&webauthn-av
ailable=false&webauthn-platform-available=false
Nov 24 09:42:54 homebridge evcc[2940132]: --
Nov 24 09:42:54 homebridge evcc[2940132]: Found. Redirecting to /u/login/password?state=hKFo2SA3cUlqd1Y0bDJUSHVZcFhWMWVLaDRJNXZ2d0F4ekdNQaFur3VuaXZlcnNhbC1sb2dpbqN0aWTZIGdOdEtmUHJuQmR6Ym5PVXlKaVhtcURsV3h4Tlhpbm9Ho2NpZNkgVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWc&ui_locales=de
Nov 24 09:42:54 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:54 POST https://identity.porsche.com/u/login/password?state=hKFo2SA3cUlqd1Y0bDJUSHVZcFhWMWVLaDRJNXZ2d0F4ekdNQaFur3VuaXZlcnNhbC1sb2dpbqN0aWTZIGdOdEtmUHJuQmR6Ym5PVXlKaVhtcURsV3h4Tlhpbm9Ho2NpZNkgVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWc&ui_locales=de
Nov 24 09:42:54 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:54 action=default&password=***&state=hKFo2SA3cUlqd1Y0bDJUSHVZcFhWMWVLaDRJNXZ2d0F4ekdNQaFur3VuaXZlcnNhbC1sb2dpbqN0aWTZIGdOdEtmUHJuQmR6Ym5PVXlKaVhtcURsV3h4Tlhpbm9Ho2NpZNkgVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWc&ui_locales=de&username=***
Nov 24 09:42:54 homebridge evcc[2940132]: --
Nov 24 09:42:54 homebridge evcc[2940132]: Found. Redirecting to /authorize/resume?state=gNtKfPrnBdzbnOUyJiXmqDlWxxNXinoG
Nov 24 09:42:54 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:54 GET https://identity.porsche.com/authorize/resume?state=gNtKfPrnBdzbnOUyJiXmqDlWxxNXinoG
Nov 24 09:42:54 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:54 Found. Redirecting to https://my.porsche.com/?code=O8aLZC0hgaAzqjvEhBx9JwSsIKfpgxlBkWQ0l9AOerB5X&state=MzxOXLkLCszeQBvc
Nov 24 09:42:54 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:54 POST https://identity.porsche.com/oauth/token
Nov 24 09:42:54 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:54 client_id=UYsK00My6bCqJdbQhTQ0PbWmcSdIAMig&code=O8aLZC0hgaAzqjvEhBx9JwSsIKfpgxlBkWQ0l9AOerB5X&code_verifier=OdmZJ2R1Z9i2nyxXTjfZ0mIlGhwVUbV65EAOkzpURUc&grant_type=authorization_code&redirect_uri=https%3A%2F%2Fmy.porsche.com%2F
Nov 24 09:42:54 homebridge evcc[2940132]: --
Nov 24 09:42:54 homebridge evcc[2940132]: {"access_token":"eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCIsImtpZCI6ImlLVTg4Mml1LWJBc1NmZTBJYmlNZCJ9.eyJjaWFtSWQiOiJmOGh4bHBlMXk3Y3h3d2JiIiwicG9yc2NoZV9qdGkiOiIwZTI0MjZhMDlhYmY4ZWUxMzcwNjFiZDlmZjRlN2YxNCIsImlzcyI6Imh0dHBzOi8vaWRlbnRpdHkucG9yc2NoZS5jb20vIiwic3ViIjoiYXV0aDB8NjQ3NWU2Y2NmNzUzZmY1NWFmYzljMWRiIiwiYX
VkIjpbImh0dHBzOi8vYXBpLnBvcnNjaGUuY29tIiwiaHR0cHM6Ly9wb3JzY2hlLXByb2R1Y3Rpb24ucG9yc2NoZS1wcm9kdWN0aW9uLmF1dGgwYXBwLmNvbS91c2VyaW5mbyJdLCJpYXQiOjE3MDA4MTUzNzQsImV4cCI6MTcwMDg1ODU3NCwiYXpwIjoiVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWciLCJzY29wZSI6Im9wZW5pZCBvZmZsaW5lX2FjY2VzcyJ9.CC7Uf5EWxkWwGZ96Vp2ZSl1IjPJ61PUOBKsljC2GRgI8Vj3rBJfEY9i5KEX3KF1Mj6G
GhzmnJ3LwkrTc3oHd39mJ5z7WqDdaD8ETOhTA5ViZ8MQig6uVy50pJ1qNZ7EmSE3zOAHzxT9_6CRCnQe2zPg5-V75QWlU6sLGANsYtLaw7T49YKR_mVShRKCqvLDlnnA9Uzi0DXFhu0MiavGOIlHpAdCIJ7JpOXlBW1rm6TvaNGUjF7YTXHDZKp_1KkSGv1fwOguROBWHUBxozOpw0qxxof0FbSVmtyYxly5LsbA8Xp6MmJgDhf4wwnzOK7gIG2C3JlSFXR55x_R9u3h_nQ","refresh_token":"v1.Mc9_oDH3pTnh1-Qr7LEHTblIqOVUJgm-60nZlJMhCXYuMiTmX
dkJB1W6WJ6QPn5DPXOqseAbBPNXXvm25gZGV1A","id_token":"eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCIsImtpZCI6ImlLVTg4Mml1LWJBc1NmZTBJYmlNZCJ9.eyJjaWFtSWQiOiJmOGh4bHBlMXk3Y3h3d2JiIiwiaXNzIjoiaHR0cHM6Ly9pZGVudGl0eS5wb3JzY2hlLmNvbS8iLCJhdWQiOiJVWXNLMDBNeTZiQ3FKZGJRaFRRMFBiV21jU2RJQU1pZyIsImlhdCI6MTcwMDgxNTM3NCwiZXhwIjoxNzAwODE1Njc0LCJzdWIiOiJhdXRoMHw2NDc1ZTZjY
2Y3NTNmZjU1YWZjOWMxZGIiLCJzaWQiOiJNWGdhOXd0T1N5aTk1bmZzVERTSHg2c1NURWtVd2tOMSJ9.Mpat9_iOpxjg0GVmwF93GCxv5H1fuj38qTeuRR0yxRt10RNzxbSbU0JY1QulY0-j20_3gvUcY_Z_wtS6PAysNIu-2T0_SMnDnZWH7Ew7f_PwMUb6O1G0-eAARJuPx4VXDj5EqOGsFNeBCMafAx5rOVdpP6TlScld0p4hOWXdOE0_rN9xsFbOoxcBhfZhfrRRbbgAFIwb3Yldx0gJ10Cld6mv0asIoEoudAdVZhGUAEo0OgnFMH-Ycu4_DMRP4saDMmJM6G8lph
B_ADIRO0zh9gXMPNhsnC6cpb-7J150tVYbrnxVmJdWEXhjRlG-4qI5Ts72keZP3jWEpHgiFMEBdQ","scope":"openid offline_access","expires_in":43200,"token_type":"Bearer"}
Nov 24 09:42:54 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:54 GET https://api.porsche.com/core/api/v3/de/de_DE/vehicles
Nov 24 09:42:55 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:55 [ {
Nov 24 09:42:55 homebridge evcc[2940132]:   "vin" : "***",
Nov 24 09:42:55 homebridge evcc[2940132]:   "isPcc" : true,
Nov 24 09:42:55 homebridge evcc[2940132]:   "relationship" : "OWNER",
Nov 24 09:42:55 homebridge evcc[2940132]:   "modelDescription" : "Cayenne E-Hybrid",
Nov 24 09:42:55 homebridge evcc[2940132]:   "modelType" : "9YAAE1",
Nov 24 09:42:55 homebridge evcc[2940132]:   "modelYear" : "2022",
Nov 24 09:42:55 homebridge evcc[2940132]:   "exteriorColor" : "tiefschwarzmetallic/tiefschwarzmetallic",
Nov 24 09:42:55 homebridge evcc[2940132]:   "exteriorColorHex" : "#000000",
Nov 24 09:42:55 homebridge evcc[2940132]:   "spinEnabled" : true,
Nov 24 09:42:55 homebridge evcc[2940132]:   "loginMethod" : "PORSCHE_ID",
Nov 24 09:42:55 homebridge evcc[2940132]:   "pendingRelationshipTerminationAt" : null,
Nov 24 09:42:55 homebridge evcc[2940132]:   "attributes" : [ {
Nov 24 09:42:55 homebridge evcc[2940132]:     "name" : "licenseplate",
Nov 24 09:42:55 homebridge evcc[2940132]:     "value" : "XXX"
Nov 24 09:42:55 homebridge evcc[2940132]:   } ],
Nov 24 09:42:55 homebridge evcc[2940132]:   "otaActive" : true,
Nov 24 09:42:55 homebridge evcc[2940132]:   "validFrom" : "2022-01-02T22:00:30.000Z"
Nov 24 09:42:55 homebridge evcc[2940132]: } ]
Nov 24 09:42:55 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:55 GET https://api.porsche.com/core/api/v3/de/de_DE/vehicles/***/pairing
Nov 24 09:42:55 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:55 {
Nov 24 09:42:55 homebridge evcc[2940132]:   "vin" : "***",
Nov 24 09:42:55 homebridge evcc[2940132]:   "pairingCode" : "***",
Nov 24 09:42:55 homebridge evcc[2940132]:   "status" : "PAIRINGCOMPLETE",
Nov 24 09:42:55 homebridge evcc[2940132]:   "canSendPairingCode" : true
Nov 24 09:42:55 homebridge evcc[2940132]: }
Nov 24 09:42:55 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:55 GET https://api.porsche.com/e-mobility/vcs/capabilities/***
Nov 24 09:42:55 homebridge evcc[2940132]: [porsche] TRACE 2023/11/24 09:42:55 {
Nov 24 09:42:55 homebridge evcc[2940132]:   "pcckErrorKey" : "GRAY_SLICE_ERROR_UNKNOWN_MSG",
Nov 24 09:42:55 homebridge evcc[2940132]:   "pcckErrorMessage" : null,
Nov 24 09:42:55 homebridge evcc[2940132]:   "pcckErrorCode" : null,
Nov 24 09:42:55 homebridge evcc[2940132]:   "pcckIsBusinessError" : true
Nov 24 09:42:55 homebridge evcc[2940132]: }
Nov 24 09:42:55 homebridge evcc[2940132]: [main  ] ERROR 2023/11/24 09:42:55 creating vehicle ev1 failed: cannot create vehicle 'template': cannot create vehicle 'porsche': unexpected status: 502 (Bad Gateway)
```

### What type of operating system are you running?

Linux

### Version

evcc 0.122.1 (adcdba04)

MarkusGH, 2023-11-24:
pyporscheconnect funktioniert. 
Wird der konkrete Aufruf an api.porsche.com benötigt, oder kommt man auch ohne den aus und es reicht das Errorhandling um die Behandlung von Status 502 zu erweitern?

andig, 2023-11-24:
Schön wäre ein Trace eines MITM Proxys.

MarkusGH, 2023-11-24:
> Schön wäre ein Trace eines MITM Proxys.

Ist für mich aufwendig, weil ich es noch nie gemacht habe.

Verständnisfrage: Warum braucht man das? 
Protokolliert evcc nicht sowieso alle Aufrufe?

andig, 2023-11-24:
> Warum braucht man das?

Weil ich dann sehe, wir der "richtige" Flow ist.

MarkusGH, 2023-11-24:
> > Warum braucht man das?
> 
> Weil ich dann sehe, wir der "richtige" Flow ist.

Ach so - Du meinst einen MITM zwischen der Porsche APP und dem Porsche Server?

andig, 2023-11-24:
Einfacher: zwischen pyconnect und dem Server! Aber es gibt ja schon einen PR- probier doch den erstmal?

closed 2023-11-25 by andig

snapscanii, 2023-12-12:
Ich habe genau den gleichen Fehler bei meinem Taycan. Habe ca 400 Fehlermeldungen am Tag. Leider bin ich noch nicht so drin in evcc. Was muss ich tun, damit alles wieder geht. Habe einen Raspberry Pi. Danke sehr

Fehlermeldung evcc: Keba P30X: vehicle odometer: unexpected status: 502 (Bad Gateway)

Aufruf von api.porsche.com bringt:

{"fault":{"faultstring":"Unable to identify proxy for host: secure and url: \/","detail":{"errorcode":"messaging.adaptors.http.flow.ApplicationNotFound"}}}