evcc-io/evcc #10911: Add APSystems Inverter with local API

rene-mueller, 2023-11-24:
An API was recently made available for the APSystems inverter. This makes it possible to read out data and set the inverter power.

andig, 2023-11-25:
Das sieht aus wie eine PythonBibliothek, aber nixht wie eine Api-Beschreibung. Könntest du mal fragen, obs auch ein Api gibt damit wir uns ncht durch den Python Quelltext wühlen müssen? In Python kenne ich mich kaum aus…

goebelmeier, 2023-11-25:
> Das sieht aus wie eine PythonBibliothek, aber nixht wie eine Api-Beschreibung. Könntest du mal fragen, obs auch ein Api gibt damit wir uns ncht durch den Python Quelltext wühlen müssen? In Python kenne ich mich kaum aus…

Ich habe Leo Tiedt mal direkt in den Kommentaren vom Vorstellungsvideo gefragt

rene-mueller, 2023-11-27:
@andig ja das ist eine Python Bibliothek. Anbei die Dokumentation der API: 
[APsystems EZ1 Local API User Manual.pdf](https://github.com/evcc-io/evcc/files/13473827/APsystems.EZ1.Local.API.User.Manual.pdf)

andig, 2023-11-27:
Soweit ich sehe bekommen wir die Leistung für 2 Kanäle, aber sonst nicht viel? Das kannst du sogar selber mit dem http Plugin umsetzen 👍🏻

rene-mueller, 2023-11-27:
@andig es ist mit der API möglich Daten auszulesen. Zusätzlich kann auch der maximale Output gesetzt werden. Spannend wäre das für eine Nulleinspeisung oder in Verbindung mit Anker SOLIX e1600 (Stromspeicher).

andig, 2023-11-28:
Yep. Willst Du mal einen PR für ein Template fürs auslesen probieren? Rest können wir später machen.

rene-mueller, 2023-11-28:
@andig kann ich machen :)

andig, 2023-11-28:
Die README bzgl. Build und ein beliebiges Zählertemplate aus https://github.com/evcc-io/evcc/tree/master/templates/definition/meter, am Beste mal nach jq suchen!

closed 2023-11-28 by andig

MaZ181, 2024-01-02:
@rene-mueller hallo gibt es eine Beispiel bzw. Ein Template für die Integration des APSystems Wechsleichter in EvCC  würde das bedeuten du gern in mein evcc integrieren danke vorab

andig, 2024-01-02:
Hat bisher noch niemand erstellt. Kannst Du als "custom" Charger aber selber machen :)

MaZ181, 2024-01-02:
@andig danke für die Info wenn ich das wüsste wie das geht. Gibt es hierfür ein bsp ?