evcc-io/evcc #10912: Home Battery Control: Fronius Gen24

MikeTheTux, 2023-11-25:
**Add Home Battery Control for Fronius Gen24**

Die entsprechenden Fronius Gen24 Register:

https://www.fronius.com/~/downloads/Solar%20Energy/Operating%20Instructions/42,0410,2649.pdf

[image]

Siehe auch [https://openwb.de/forum/viewtopic.php?t=6585](https://openwb.de/forum/viewtopic.php?t=6585)

Offen:

- Wie das Laden nicht nur ermöglicht, sondern erzwungen werden könnte.

Referenz:

- [#10379](https://github.com/evcc-io/evcc/issues/10379#issuecomment-1806906917)

andig, 2023-11-25:
We can consider this at a later stage. At this time, we don‘t want to control battery power directly. You can still,listen for the published `batteryMode` to do that externally, following evcc signaling.

andig, 2023-11-27:
Siehe auch https://github.com/evcc-io/evcc/discussions/10805#discussioncomment-7676975 Für SMA.

andig, 2023-11-27:
Also:
- Normal: StorCtl_Mod = 0
- Hold: StorCtl_Mod = 0b10, OutWRte = 0%
- Laden: ChaGriSet muss erlaubt sein (=Anwender), MinRsvPct=95% maxsoc

Die Ladesettings müssen bei Normal/Hold wieder deaktiviert werden, mindestens aber MinRsvPct=20% minsoc

Ergibt das Sinn? /cc @premultiply 

@MikeTheTux kannst du selbst compilieren? Dann könnten wir das in einem separaten Branch mal ausprobieren.

andig, 2023-11-27:
Alternative:
- Hold: MinRsvPct=Soc

Das entspräche der aktuellen Umsetzung für die anderen WR und braucht kein StorCtl_Mod

premultiply, 2023-11-27:
Da es hier nicht nur um Fronius geht sondern generisch um SunSpec via Model 124 ändere ich mal den Titel 😉

premultiply, 2023-11-27:
normal: ChaGriSet = 0, StorCtl_Mod = 0, OutWRte = 100
hold: ChaGriSet = 0, StorCtl_Mod = 2, OutWRte = 0
charge: ChaGriSet = 1, StorCtl_Mod = 2 (oder 3?), OutWRte = -100

Auch hier scheint es einen Watchdog zu geben. Ob der hier zwingend erforderlich ist konnte ich aber noch nicht klar erkennen. Ich gehe aber davon aus.

andig, 2023-11-27:
@premultiply was hast du gegen `MinRsvPct`? Da kann der WR sein Ding machen und trotzdem den Soc halten?

premultiply, 2023-11-27:
Vor allem dass dies in der Regel persistente Vorgaben sind und das System im Fehlerfall so hängen bleibt ohne dass der Anwender oder ein Externer dies einfach erkennen und beheben kann (ohne evcc!).

Zudem gehen damit u.U. dann bei unbedachter Vorgehensweise und evcc-Konfiguration systemspezifische Grundeinstellungen verloren.

Bei einigen Systemen dürfte sowas zudem bis in den Flash durchschlagen und somit auch noch komplette Neustarts überdauern.

Das sollte daher immer nur die Notlösung sein wenn sonst keine schönen und dafür gedachten Lösungen verfügbar sind.

andig, 2023-11-27:
Maybe.

> Ich sehe aber nichts, was den Wechselrichter dann zum laden zwingen würde. Es sei denn ChaGriSet = 1 ist verbindlich und nicht nur eine Möglichkeit für den WR? Ich würde ohnehin erwarten, dass das erlaubt ist um Tiefentladung zu verhindern.

premultiply, 2023-11-27:
> Ergänzung: ich denke Du vergisst beim laden InWRte, denn nur das dürfte relevant sein?

Nein, das ist Default auf 100 und hier irrelevant.
`OutWRte` mit **-100** forciert hier das Laden.

andig, 2023-11-27:
Dann bastel ich mal was...

andig, 2023-11-27:
Für Sunspec brauche ich ein Testgerät. Bitte Portweiterleitung oder ssh für info@evcc.io.

premultiply, 2023-11-27:
> Dann bastel ich mal was...

🤔 Könnten wir ggf. sogar mit meinem SMA SBS direkt testen können.

andig, 2023-11-27:
Supi- schaue ich morgen!

premultiply, 2023-11-27:
Anbei mal ein Auszug von meinem SMA SBS 3.7 im Normalzustand, schlafend mit leerer Batterie (bis auf Notstromreserve).
```
--------- Model 124 storage ---------
[main  ] TRACE 2023/11/27 18:32:11 modbus: send 00 1c 00 00 00 06 7e 03 9d b3 00 18
[main  ] TRACE 2023/11/27 18:32:11 modbus: recv 00 1c 00 00 00 33 7e 03 30 0e 60 ff ff ff ff 00 00 ff ff ff ff 00 17 ff ff 3b e2 00 03 80 00 80 00 ff ff ff ff ff ff ff ff 00 00 80 00 80 00 80 00 00 00 80 00 ff fe 80 00
WChaMax                     3680.00    uint16
WChaGra                         n/a    uint16
WDisChaGra                      n/a    uint16
StorCtl_Mod                  0x0000    bitfield16
VAChaMax                        n/a    uint16
MinRsvPct                       n/a    uint16
ChaState                      23.00    uint16
StorAval                        n/a    uint16
InBatV                       153.30    uint16
ChaSt                             3    enum16
OutWRte                         n/a    int16
InWRte                          n/a    int16
InOutWRte_WinTms                n/a    uint16
InOutWRte_RvrtTms               n/a    uint16
InOutWRte_RmpTms                n/a    uint16
ChaGriSet                       n/a    enum16
WChaMax_SF                        0    sunssf
WChaDisChaGra_SF                n/a    sunssf
VAChaMax_SF                     n/a    sunssf
MinRsvPct_SF                    n/a    sunssf
ChaState_SF                       0    sunssf
StorAval_SF                     n/a    sunssf
InBatV_SF                        -2    sunssf
InOutWRte_SF                    n/a    sunssf
```

andig, 2023-11-28:
Kann der noch mehr? V.a. Gibts ein Battery Model ohne die blöden Scale Factors?

andig, 2023-11-28:
@premultiply der Kostal hat kein 124, dafür aber das Battery Base Model 802 (danke @deadrabbit87):

```
--------- Model 1 common ---------
Mn             KOSTAL    string
Md   PLENTICORE plus7    string
--------- Model 103 inverter ---------
A                    1.15    uint16
AphA                 0.35    uint16
AphB                 0.37    uint16
AphC                 0.41    uint16
A_SF                   -2    sunssf
PPVphAB               n/a    uint16
PPVphBC               n/a    uint16
PPVphCA               n/a    uint16
PhVphA             219.80    uint16
PhVphB             221.00    uint16
PhVphC             223.40    uint16
V_SF                   -1    sunssf
W                   47.00    int16
W_SF                    0    sunssf
Hz                  50.00    uint16
Hz_SF                  -2    sunssf
VA                 255.00    int16
VA_SF                   0    sunssf
VAr                251.00    int16
VAr_SF                  0    sunssf
PF                   1.00    int16
PF_SF                  -3    sunssf
WH            37325890.00    acc32
WH_SF                   0    sunssf
DCA                   n/a    uint16
DCA_SF                  0    sunssf
DCV                   n/a    uint16
DCV_SF                  0    sunssf
DCW                 70.00    int16
DCW_SF                  0    sunssf
TmpCab              28.80    int16
TmpSnk              16.60    int16
TmpTrns             17.30    int16
TmpOt                0.00    int16
Tmp_SF                 -1    sunssf
St                      4    enum16
StVnd                   0    enum16
Evt1           0x00000000    bitfield32
Evt2           0x00000000    bitfield32
EvtVnd1        0x00000000    bitfield32
EvtVnd2        0x00000000    bitfield32
EvtVnd3        0x00000000    bitfield32
EvtVnd4        0x00000000    bitfield32
--------- Model 113 inverter ---------
A               1.1535645    float32
AphA           0.35705566    float32
AphB           0.37841797    float32
AphC            0.4180908    float32
PPVphAB                 0    float32
PPVphBC                 0    float32
PPVphCA                 0    float32
PhVphA           219.8174    float32
PhVphB          221.05095    float32
PhVphC          223.44754    float32
W                      47    float32
Hz              50.006104    float32
VA              255.38342    float32
VAr             251.00957    float32
PF                      1    float32
WH          3.7325892e+07    float32
DCA                     0    float32
DCV                     0    float32
DCW              70.50428    float32
[... 139 lines cut ...]
CellVMinStr                   0.00    uint16
CellVMinMod                   0.00    uint16
CellVAvg                      0.00    uint16
A                             0.00    int16
AChaMax                      25.60    uint16
ADisChaMax                   25.60    uint16
W                             0.00    int16
ReqInvState                      0    enum16
ReqW                          0.00    int16
SetOp                            1    enum16
SetInvState                      0    enum16
AHRtg_SF                         0    sunssf
WHRtg_SF                         0    sunssf
WChaDisChaMax_SF                 0    sunssf
DisChaRte_SF                     0    sunssf
SoC_SF                           0    sunssf
DoD_SF                           0    sunssf
SoH_SF                           0    sunssf
V_SF                            -1    sunssf
CellV_SF                        -1    sunssf
A_SF                            -2    sunssf
AMax_SF                         -2    sunssf
W_SF                             0    sunssf

```

andig, 2023-11-28:
@premultiply es gelingt mir allerdings nicht, auf `SoCRsvMin` (das sollte 40318 sein) zu schreiben:

```
[modbus] TRACE 2023/11/28 15:39:54 modbus: send 00 20 00 00 00 09 47 10 9d 7e 00 01 02 00 14
[modbus] TRACE 2023/11/28 15:39:54 modbus: recv 00 20 00 00 00 03 47 90 02
[main  ] FATAL 2023/11/28 15:39:54 set battery mode: modbus: exception '2' (illegal data address), function '16'
```

Vielleicht hast du ja eine Idee. Sonst erstmal dead end...

andig, 2023-11-28:
Ich denke du versuchst da 2 Words zu schreiben. Mit WriteSingle geht das nicht.

premultiply, 2023-11-28:
> Kann der noch mehr? V.a. Gibts ein Battery Model ohne die blöden Scale Factors?

Nein, leider nicht. Auch via 124 offensichtlich überhaupt nicht dementsprechend steuerbar. Aber bei SMA gibt es ja [andere](https://github.com/evcc-io/evcc/discussions/10805#discussioncomment-7685939) Wege. Können wir hier daher aussen vor lassen.

> es gelingt mir allerdings nicht, auf SoCRsvMin (das sollte 40318 sein) zu schreiben:

Irgendwas ist da wohl in deinem Request faul. WriteMultiple mit 5 Bytes?

premultiply, 2023-11-28:
> der Kostal hat kein 124, dafür aber das Battery Base Model 802

Aber der hier initial beschriebene Fronius Gen24 hat Model 124.
Wollen wir nicht lieber erstmal das probieren? Das ist eigentlich auch generell gängiger - wobei ich natürlich nicht weiß wie umfangreich dort ggf. die Schreibroutinen verbreitet sind.

andig, 2023-11-29:
Können wir beides probieren, sollte inhaltlich gleich sein. Ich bräuchte halt ein Testexemplar.

andig, 2023-11-29:
@premultiply Idee: wir sollten die "Batterien" für 124/802 zum experimentieren mal in eigene Templates verlagern bis wir wissen welcher WR was bietet. PR kann ich aktualisieren, hab allerdings nix mehr zum testen.

deadrabbit87, 2023-11-29:
Ich könnte noch einen Sungrow organisieren. Wäre das was?

premultiply, 2023-11-30:
Nein, SunGrow kann m.W.n. kein SunSpec.

deadrabbit87, 2023-11-30:
Interessant, der Kostal ist auf der Sunspec Website nicht drauf, der Sungrow schon, aber nur der SG125HV.

andig, 2023-11-30:
Kostal wiederum kann Sungrow, auch wenn sie keine Zertifizierung bestehen würden... die Bugs sind haarsträubend.

andig, 2023-11-30:
> > es gelingt mir allerdings nicht, auf SoCRsvMin (das sollte 40318 sein) zu schreiben:
> 
> Irgendwas ist da wohl in deinem Request faul. WriteMultiple mit 5 Bytes?

Was meinst Du genau?

    modbus: send 00 20 00 00 00 09 47 10 9d 7e 00 01 02 00 14

Das ist FC 16 (10h), Reg 0x9d7e, 0x0001 Register, Wert 0x200. Oder lese ich das völlig falsch? Frage mich allerdings, wo die 0x200 her kommt. Mhmmmmm.

deadrabbit87, 2023-11-30:
Ich kann gerne bezüglich Bugs mich an Kostal wenden. Die waren bisher sehr kooperativ.

andig, 2023-11-30:
@deadrabbit87 dann wäre es super, wenn sie ihre SunSpec Hierarchie komplett aufbauen und am Ende sauber terminieren würden damit man keine hässlichen Worarounds einbauen muss: https://github.com/andig/gosunspec/commit/3759cec7aae0007e31fe12804cf08ffbdab207d0

andig, 2023-11-30:
@deadrabbit87 wenn der Support gut ist könntest Du auch mal fragen, warum Sunspec Model 802, Block 0, Point SoCRsvMin sich nicht beschreiben lässt:

```
[modbus] TRACE 2023/11/30 11:51:58 modbus: send 00 20 00 00 00 09 47 10 9d 7e 00 01 02 00 14
[modbus] TRACE 2023/11/30 11:51:58 modbus: recv 00 20 00 00 00 03 47 90 02
[main  ] FATAL 2023/11/30 11:51:58 set battery mode: modbus: exception '2' (illegal data address), function '16'
```

deadrabbit87, 2023-12-01:
Ich geb mein Bestes da was raus zu kriegen....

andig, 2023-12-08:
Yep. Wieder ein halb umgesetzter Standard :/ Dann müssen wir- bevor es hier weiter gehen kann- erstmal auf ein passendes Testexemplar warten.

premultiply, 2023-12-08:
Ist bei SMA ja leider auch so 😔

Errator, 2023-12-13:
> Yep. Wieder ein halb umgesetzter Standard :/ Dann müssen wir- bevor es hier weiter gehen kann- erstmal auf ein passendes Testexemplar warten.

Ich bin über die FB Gruppe von EVCC hier gelandet. Ich habe einen Speicher von Alpha ESS mit Hybrid Batterie WR. Laut dem was ich gelesen habe ist der in der sunspec Alliance und die Register sind auch frei verfügbar und Beschreibbar. Zumindest gibt es da auch für IOBroker und Home Assistant zig Mods die das umsetzen.Ich suche auch schon länger die Möglichkeit die 3-4 KWH lieber im Speicher zu lassen und zu sperren. Ich lade dann lieber schnell in der Zeit mit günstigen Tibber Strom.

 Ich bin jedoch nur Anwender, kann ich mit meinem WR behilflich sein?

premultiply, 2023-12-13:
Ob eine Firma Mitglied in der SunSpec Alliance ist sagt leider nicht über die Implementierung der SunSpec-Modelle in deren Produkten aus. Von AlphaESS habe ich diesbezüglich noch nie etwas gehört. Da gibt es wie üblich nur die eigene proprietäre Soße.

andig, 2023-12-21:
Kurzes Zwischenfazit: hier geht nix da kein bekannter Hersteller die Register beschreibbar macht. Ich schließe daher vorläufig. Schade eigentlich...

closed 2023-12-21 by andig

premultiply, 2023-12-21:
Ja, wirklich schade. Hätte ja sein können dass einfach mal irgendwas standardkonform funktioniert.

MikeTheTux, 2023-12-22:
Beim Fronius scheint laut Doku (initialer Post) doch auch das Schreiben mgl..
Scheitert die Umsetzung an dem Zugang zu einem Test-Gerät?

andig, 2023-12-22:
Hatte bisher nur Kostal zum testen. Reden wir von Sunspec? Dann gerne her mit einem Testgerät.

MikeTheTux, 2023-12-22:
[https://www.fronius.com/~/downloads/Solar%20Energy/Operating%20Instructions/42,0410,2649.pdf](https://www.fronius.com/~/downloads/Solar%20Energy/Operating%20Instructions/42,0410,2649.pdf)

Passt das?

PN bzgl. Freischaltung?

andig, 2023-12-22:
Sieht gut aus. Gerne an info@evcc.io

StevieC121176, 2024-02-25:
> > Für Sunspec brauche ich ein Testgerät. Bitte Portweiterleitung oder ssh für [info@evcc.io](mailto:info@evcc.io).
> 
> Mein Zugang zum Kostal mit Sunspec wäre noch offen...

@deadrabbit87 Könntest du mir einen Tip geben wie ich so eine Portweiterleitung zu meinem Kostal einrichte. Ich habe gerade ein Modbus Problem. Vielen Dank.

deadrabbit87, 2024-02-25:
Ports einfach so zu öffnen ist keine gute Idee.

Tailscale würde ich empfehlen. Damit das aber funktioniert, musst du einen subnet Router einrichten. Ist aber vermutlich für dich auch nicht so leicht....

Was ist denn das Problem?

StevieC121176, 2024-02-26:
Ich bekomme mit den nightly Versionen ab Version `0.124.4 (3bf57f87)` immer Modbus Fehler (sh.log) Wechsle ich aber auf die 124.4 stable bekomme ich keine Fehler. Daher schließe ich ein Hardwareproblem aus. Die Batteriesteuerung funktioniert mit der 124.4 stable tadellos, bei den nightlys wird die Entladung der Batterie bei Modus Schnell nicht gesperrt.

```
● evcc.service - evcc
     Loaded: loaded (/lib/systemd/system/evcc.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2024-02-24 16:21:52 CET; 16min ago
   Main PID: 3264 (evcc)
      Tasks: 11 (limit: 3933)
        CPU: 22.844s
     CGroup: /system.slice/evcc.service
             └─3264 /usr/bin/evcc

Feb 24 16:38:18 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:38:18 charge voltages: [234 232 232]V
Feb 24 16:38:18 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:38:18 detected connected phases: 3p
Feb 24 16:38:18 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:38:18 charge currents: [0 0 0]A
Feb 24 16:38:18 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:38:18 charge total import: 4704.800kWh
Feb 24 16:38:18 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:38:18 charger status: B
Feb 24 16:38:19 raspberrypi evcc[3264]: [site  ] ERROR 2024/02/24 16:38:19 battery mode: modbus: exception '4' (server device failure), function '16'
Feb 24 16:38:19 raspberrypi evcc[3264]: [site  ] DEBUG 2024/02/24 16:38:19 ----
Feb 24 16:38:19 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:38:19 charge power: 0W
Feb 24 16:38:19 raspberrypi evcc[3264]: [lp-2  ] DEBUG 2024/02/24 16:38:19 charge power: 0W
Feb 24 16:38:20 raspberrypi evcc[3264]: [site  ] DEBUG 2024/02/24 16:38:20 pv power: 466W
pi@raspberrypi:~ $ sudo journalctl -u evcc --since "16:20:00"
-- Journal begins at Mon 2024-02-19 14:51:00 CET, ends at Sat 2024-02-24 16:41:56 CET. --
Feb 24 16:21:42 raspberrypi evcc[543]: [main  ] FATAL 2024/02/24 16:21:42 evcc was stopped. OS should restart the service. Or restart manually.
Feb 24 16:21:42 raspberrypi systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Feb 24 16:21:42 raspberrypi systemd[1]: evcc.service: Failed with result 'exit-code'.
Feb 24 16:21:42 raspberrypi systemd[1]: evcc.service: Consumed 1.095s CPU time.
Feb 24 16:21:52 raspberrypi systemd[1]: evcc.service: Scheduled restart job, restart counter is at 1.
Feb 24 16:21:52 raspberrypi systemd[1]: Stopped evcc.
Feb 24 16:21:52 raspberrypi systemd[1]: evcc.service: Consumed 1.095s CPU time.
Feb 24 16:21:52 raspberrypi systemd[1]: Started evcc.
Feb 24 16:21:52 raspberrypi evcc[3264]: [main  ] INFO 2024/02/24 16:21:52 evcc 0.124.4 (3bf57f87)
Feb 24 16:21:52 raspberrypi evcc[3264]: [main  ] INFO 2024/02/24 16:21:52 using config file: /etc/evcc.yaml
Feb 24 16:21:52 raspberrypi evcc[3264]: [main  ] INFO 2024/02/24 16:21:52 starting ui and api at :7070
Feb 24 16:21:53 raspberrypi evcc[3264]: [db    ] INFO 2024/02/24 16:21:53 using sqlite database: /var/lib/evcc/evcc.db
Feb 24 16:21:53 raspberrypi evcc[3264]: [mqtt  ] INFO 2024/02/24 16:21:53 connecting evcc-1482118671 at tcp://192.168.178.80:1883
Feb 24 16:21:53 raspberrypi evcc[3264]: [mqtt  ] DEBUG 2024/02/24 16:21:53 tcp://192.168.178.80:1883 connected
Feb 24 16:21:53 raspberrypi evcc[3264]: [proxy-5200] DEBUG 2024/02/24 16:21:53 modbus proxy for 192.168.178.74:502 listening at :5200
Feb 24 16:22:00 raspberrypi evcc[3264]: [lp-1  ] WARN 2024/02/24 16:22:00 deprecated: minCurrent setting is ignored, please remove
Feb 24 16:22:00 raspberrypi evcc[3264]: [lp-1  ] WARN 2024/02/24 16:22:00 deprecated: maxcurrent setting is ignored, please remove
Feb 24 16:22:00 raspberrypi evcc[3264]: [lp-1  ] WARN 2024/02/24 16:22:00 deprecated: phases setting is ignored, please remove
Feb 24 16:22:00 raspberrypi evcc[3264]: [lp-2  ] WARN 2024/02/24 16:22:00 deprecated: minCurrent setting is ignored, please remove
Feb 24 16:22:00 raspberrypi evcc[3264]: [lp-2  ] WARN 2024/02/24 16:22:00 deprecated: maxcurrent setting is ignored, please remove
Feb 24 16:22:00 raspberrypi evcc[3264]: [lp-2  ] WARN 2024/02/24 16:22:00 deprecated: phases setting is ignored, please remove
Feb 24 16:22:01 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:22:01 charge total import: 4704.700kWh
Feb 24 16:22:01 raspberrypi evcc[3264]: [lp-2  ] DEBUG 2024/02/24 16:22:01 charge total import: 0.000kWh
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] WARN 2024/02/24 16:22:01 vehicle 'E-Bike' does not support automatic detection
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01 site config:
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01   meters:      grid ✓ pv ✓ battery ✓
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01     grid:      power ✓ energy ✓ currents ✓
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01     pv 1:      power ✓ energy ✗ currents ✗
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01     pv 2:      power ✓ energy ✗ currents ✗
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01     pv 3:      power ✓ energy ✓ currents ✗
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
Feb 24 16:22:01 raspberrypi evcc[3264]: [site  ] INFO 2024/02/24 16:22:01   vehicles:
[... cut ...]
Feb 24 16:22:39 raspberrypi evcc[3264]: [lp-2  ] DEBUG 2024/02/24 16:22:39 charge total import: 0.000kWh
Feb 24 16:22:39 raspberrypi evcc[3264]: [lp-2  ] DEBUG 2024/02/24 16:22:39 charger status: A
Feb 24 16:22:39 raspberrypi evcc[3264]: [site  ] ERROR 2024/02/24 16:22:39 battery mode: modbus: exception '4' (server device failure), function '16'
Feb 24 16:22:40 raspberrypi evcc[3264]: [site  ] DEBUG 2024/02/24 16:22:40 ----
Feb 24 16:22:40 raspberrypi evcc[3264]: [lp-1  ] DEBUG 2024/02/24 16:22:40 charge power: 0W
```

andig, 2024-02-26:
@StevieC121176 vmtl. der gleiche Fehler wie in https://github.com/evcc-io/evcc/issues/12444. Der Ball fürs Debugging liegt bei Dir :/

andig, 2024-02-26:
@StevieC121176 machs nicht so kompliziert. Portweiterleitung und gut. Weitere Postings hier können entfallen :/

andig, 2024-12-17:
Siehe https://github.com/evcc-io/evcc/issues/17687#issuecomment-2547813460. Es herrscht etwas Verwirrung, wie für Netzladung ein Ladelimit zu setzen wäre falls nicht mit 100% Leistung geladen werden soll. Konkreter: warum wir dafür überhaupt `OutWRte` statt `InWRte` verwendet?