evcc-io/evcc #10920: Load Point IDs not fix

Steven-2023, 2023-11-25:
### Describe the bug

In my installation I use two different charger. They will be chown in MQTT as charger with load point id "1" and "2".
If I kommt out the first charger, then the 2nd charger reports with ID1.
This leads to integration issues with Home Assistance because the setup is mixed up.

### Steps to reproduce

1. configure two chargers and give them two different name 
2. start evcc and check MQTT messages  
3. memorize names in positions
4. comment out the first charger and restart
5. check MQTT messages. you will find the 2nd charger report with ID 1.

Two ideas:
- use charger name in MQTT report
- introduce a "disabled" attribute, to keep the list unchanged

(btw. I have to comment the charger out because EVCC will not start if it's not connected.)

### Configuration details

```yaml
- name: Mobile_Charger
  type: template
  template: vestel
  modbus: tcpip
  id: 255
  host: 192.168.0.7
  port: 502
- name: Vestel_EVC04
  type: template
  template: vestel
  modbus: tcpip
  id: 255
  host: 192.168.0.7
  port: 502
```

### Log details

```text
..
```

### What type of operating system are you running?

Linux

### Version

0.122.1

mdkeil, 2023-11-25:
Why do you have configured the same physical charger with different names?

Steven-2023, 2023-11-25:
Thanks for the hint. It was just a copy+paste error. These are two different devices.

closed 2023-11-26 by andig

Steven-2023, 2023-11-26:
Hi, did you change anything? Or why is this issue closed?