evcc-io/evcc #10927: Ladeleistung zeigt immer -10W

g-mocken, 2023-11-26:
Ich habe eine "Walli-Light Pro" Wallbox (mit EM-CP-PP-ETH Ladesteuerung, aber _ohne_ eigenem Zähler) und SMA-PV-Anlage (Sunnyboy-Wechselrichter und Energymeter). Grundsätzlich scheint alles mit der Grundkonfiguration aus "evcc configure" zu funktionieren, nur kriege ich im Webinterface dauerhaft -10W Ladeleistung angezeigt, in der Datenbank werden alle Vorgänge mit 0kWh abgelegt und im Webinterface deshalb anscheinend ausgeblendet.
Mir ist klar, dass mangels extra Messeinrichtung keine exakte Ladeleistung angezeigt werden kann, aber so richtig sinnvoll erscheint mir das aktuelle Verhalten trotzdem nicht. Habe ich hier noch irgendwas falsch konfiguriert?

Im Log findet sich sowas wie (keine Ladung):
```
[lp-1  ] DEBUG 2023/11/26 12:03:20 charge power: -10W
[site  ] DEBUG 2023/11/26 12:03:20 pv power: 374W
[site  ] DEBUG 2023/11/26 12:03:20 grid power: -26W
[site  ] DEBUG 2023/11/26 12:03:20 grid powers: [-358 149 182]W
[site  ] DEBUG 2023/11/26 12:03:20 grid currents: [-1.53 0.811 1.02]A
[site  ] DEBUG 2023/11/26 12:03:20 site power: -26W
[lp-1  ] DEBUG 2023/11/26 12:03:20 charge voltages: [-0.001 -0.001 -0.001]V
[lp-1  ] DEBUG 2023/11/26 12:03:21 charge currents: [-0.001 -0.001 -0.001]A
[lp-1  ] DEBUG 2023/11/26 12:03:21 charge total import: -0.010kWh
[lp-1  ] DEBUG 2023/11/26 12:03:21 charger status: B
```
oder (mit Ladung):
```
[lp-1  ] DEBUG 2023/11/26 12:16:30 charge power: -10W
[site  ] DEBUG 2023/11/26 12:16:30 pv power: 456W
[site  ] DEBUG 2023/11/26 12:16:30 grid power: 3550W
[site  ] DEBUG 2023/11/26 12:16:30 grid powers: [788 1375 1387]W
[site  ] DEBUG 2023/11/26 12:16:30 grid currents: [3.41 5.83 5.93]A
[site  ] DEBUG 2023/11/26 12:16:30 site power: 3550W
[lp-1  ] DEBUG 2023/11/26 12:16:30 charge voltages: [-0.001 -0.001 -0.001]V
[lp-1  ] DEBUG 2023/11/26 12:16:30 charge currents: [-0.001 -0.001 -0.001]A
[lp-1  ] DEBUG 2023/11/26 12:16:30 charge total import: -0.010kWh
[lp-1  ] DEBUG 2023/11/26 12:16:30 charger status: C
```
Hier meine Konfiguration:

```
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug 
levels:
  cache: error

# unique installation id
plant: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

interval: 10s # control cycle interval

meters:
- type: template
  template: sma-energy-meter 
  usage: grid  
  host: sma222222222.fritz.box  
  name: grid1
- type: template
  template: sma-inverter 
  usage: pv  
  host: sma111111111.fritz.box  
  password: xxxxxxxxxx 
  name: pv2

chargers:
- type: template
  template: phoenix-em-eth 
  id: 180  
  host: wallbox.fritz.box 
  port: 502  
  modbus: tcpip  
  name: wallbox3

loadpoints:
- title: Garage
  charger: wallbox3
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
- title: Meine Anlage
  meters:
    grid: grid1
    pv:
    - pv2
```

andig, 2023-11-26:
@premultiply hast du irgendeine Idee wo die -10W her kommen sollen? Das erscheint mir extrem spanisch?

andig, 2023-11-26:
@g-mocken woher kommen Ladestrom und -spannung? Das wird ja inner verrückter???

andig, 2023-11-26:
Mach mal bitte ein

    evcc meter wallbox3 —log trace

g-mocken, 2023-11-26:
Mache ich gleich, vorab noch eine Idee: der "phoenix-em-eth" hat grundsätzlich ein Web+Modbus-Interface für Zählerwerte, aber das ist ein separates HW-Modul, welches bei mir nicht vorhanden ist. Kann es sein, dass evcc trotzdem die Werte von da auslest und damit irgendwie auf diese dauerhaften -10W kommt? 
(Es müsste aber auch irgendwie auslesbar sein, dass gar kein solches Modul angeschlossen ist.)

g-mocken, 2023-11-26:
Ich kriege nur "Error: accepts at most 1 arg(s), received 2", wenn ich das "-log trace" anhänge. Aber ohne ist auch interessant:
```
evcc meter wallbox3
[main  ] INFO 2023/11/26 12:42:45 evcc 0.122.1
[main  ] INFO 2023/11/26 12:42:45 using config file: /etc/evcc.yaml
[db    ] INFO 2023/11/26 12:42:50 using sqlite database: /home/pi/.evcc/evcc.db
[main  ] FATAL 2023/11/26 12:42:50 wallbox3 not found
```

g-mocken, 2023-11-26:
Ach, das ist ein "--log", damit geht's:

```
evcc --log trace meter wallbox3
[main  ] INFO 2023/11/26 12:50:06 evcc 0.122.1
[main  ] INFO 2023/11/26 12:50:06 using config file: /etc/evcc.yaml
[db    ] INFO 2023/11/26 12:50:16 using sqlite database: /home/pi/.evcc/evcc.db
[db    ] TRACE 2023/11/26 12:50:16 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/11/26 12:50:16 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/11/26 12:50:16 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/26 12:50:16 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2023/11/26 12:50:16 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/11/26 12:50:16 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/11/26 12:50:16 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/11/26 12:50:16 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/11/26 12:50:16 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/26 12:50:16 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/11/26 12:50:16 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/11/26 12:50:16 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/26 12:50:16 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details%") -1 <nil>
[db    ] TRACE 2023/11/26 12:50:16 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/11/26 12:50:16 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details%") -1 <nil>
[db    ] TRACE 2023/11/26 12:50:17 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[main  ] FATAL 2023/11/26 12:50:17 wallbox3 not found

```

andig, 2023-11-26:
Machs bitte nochmal ohne das wallbox3. Ich denke der Grund ist eh schon klar:

	if b, err := wb.conn.ReadInputRegisters(phxEMEthRegVoltages, 2); err == nil && encoding.Int32LswFirst(b) > 0 {
		currentPower = wb.currentPower
		totalEnergy = wb.totalEnergy
		currents = wb.currents
		voltages = wb.voltages
	}

Da kommt wohl was und >0. Ich wundere mich allerdings, dass da nicht ein definierter Nullwert kommt?

/cc @premultiply

g-mocken, 2023-11-26:
```
evcc --log trace meter
[main  ] INFO 2023/11/26 13:03:23 evcc 0.122.1
[main  ] INFO 2023/11/26 13:03:23 using config file: /etc/evcc.yaml
[db    ] INFO 2023/11/26 13:03:27 using sqlite database: /home/pi/.evcc/evcc.db
[db    ] TRACE 2023/11/26 13:03:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/11/26 13:03:27 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/11/26 13:03:27 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/26 13:03:27 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2023/11/26 13:03:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/11/26 13:03:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/11/26 13:03:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/11/26 13:03:27 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/11/26 13:03:27 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/26 13:03:28 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/11/26 13:03:28 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/11/26 13:03:28 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/26 13:03:28 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details%") -1 <nil>
[db    ] TRACE 2023/11/26 13:03:28 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/11/26 13:03:28 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details%") -1 <nil>
[db    ] TRACE 2023/11/26 13:03:28 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[sma   ] TRACE 2023/11/26 13:03:28 send 192.168.178.37: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2023/11/26 13:03:28 recv 192.168.178.37: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2023/11/26 13:03:28 new energy meter at 192.168.178.37 - Serial=XXXX
[sma   ] TRACE 2023/11/26 13:03:28 send 192.168.178.37: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2023/11/26 13:03:28 send discover package
[sma   ] TRACE 2023/11/26 13:03:28 recv 192.168.178.38: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2023/11/26 13:03:29 send 192.168.178.37: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2023/11/26 13:03:29 send discover package
[sma   ] TRACE 2023/11/26 13:03:29 recv 192.168.178.38: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2023/11/26 13:03:29 recv 192.168.178.37: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2023/11/26 13:03:29 new energy meter at 192.168.178.37 - Serial=XXXX
[sma   ] TRACE 2023/11/26 13:03:29 found device 1901410563 at 192.168.178.37
[sma   ] TRACE 2023/11/26 13:03:29 send 192.168.178.38: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2023/11/26 13:03:29 send discover package
[sma   ] TRACE 2023/11/26 13:03:29 recv 192.168.178.38: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2023/11/26 13:03:29 new inverter at 192.168.178.38 - Serial=XXXX
[sma   ] TRACE 2023/11/26 13:03:29 found device 1992089037 at 192.168.178.38
[sma   ] TRACE 2023/11/26 13:03:29 send 192.168.178.38: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2023/11/26 13:03:29 new inverter at 192.168.178.38 - Serial=XXXX
[sma   ] TRACE 2023/11/26 13:03:29 recv 192.168.178.38: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2023/11/26 13:03:29 recv 192.168.178.38: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2023/11/26 13:03:29 login for 192.168.178.38:9522
[sma   ] TRACE 2023/11/26 13:03:29 send 192.168.178.38: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2023/11/26 13:03:29 recv 192.168.178.38: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[... cut ...]
[sma   ] TRACE 2023/11/26 13:03:30 send 192.168.178.38: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[db    ] TRACE 2023/11/26 13:03:30 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
grid1
-----
Power:          -570W
Energy:         12255.1kWh
Current L1..L3: -4.15A 0.745A 1.47A
Voltage L1..L3: 240V 240V 237V
Power L1..L3:   -992W 130W 292W

pv2
---
Power:          1018W
Energy:         24840.3kWh
Current L1..L3: 4.28A 0A 0A
Voltage L1..L3: 239V 0V 0V
Power L1..L3:   0W 0W 0W
```

premultiply, 2023-11-26:
evcc charger --log trace

g-mocken, 2023-11-26:
```
evcc charger --log trace
[main  ] INFO 2023/11/26 14:51:55 evcc 0.122.1
[main  ] INFO 2023/11/26 14:51:55 using config file: /etc/evcc.yaml
[db    ] INFO 2023/11/26 14:51:59 using sqlite database: /home/pi/.evcc/evcc.db
[db    ] TRACE 2023/11/26 14:51:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/11/26 14:51:59 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/11/26 14:51:59 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/26 14:51:59 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2023/11/26 14:51:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/11/26 14:51:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/11/26 14:51:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/11/26 14:52:00 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/11/26 14:52:00 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/26 14:52:00 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/11/26 14:52:00 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/11/26 14:52:00 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/26 14:52:00 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details%") -1 <nil>
[db    ] TRACE 2023/11/26 14:52:00 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/11/26 14:52:00 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details%") -1 <nil>
[db    ] TRACE 2023/11/26 14:52:00 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[db    ] TRACE 2023/11/26 14:52:00 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[em-eth] TRACE 2023/11/26 14:52:00 modbus: send 00 01 00 00 00 06 b4 04 00 6c 00 02
[em-eth] TRACE 2023/11/26 14:52:00 modbus: recv 00 01 00 00 00 07 b4 04 04 ff ff ff ff
[em-eth] TRACE 2023/11/26 14:52:00 modbus: send 00 02 00 00 00 06 b4 04 00 78 00 02
[em-eth] TRACE 2023/11/26 14:52:00 modbus: recv 00 02 00 00 00 07 b4 04 04 ff ff ff ff
[em-eth] TRACE 2023/11/26 14:52:00 modbus: send 00 03 00 00 00 06 b4 04 00 80 00 02
[em-eth] TRACE 2023/11/26 14:52:00 modbus: recv 00 03 00 00 00 07 b4 04 04 ff ff ff ff
[em-eth] TRACE 2023/11/26 14:52:00 modbus: send 00 04 00 00 00 06 b4 04 00 72 00 06
[em-eth] TRACE 2023/11/26 14:52:00 modbus: recv 00 04 00 00 00 0f b4 04 0c ff ff ff ff ff ff ff ff ff ff ff ff
[em-eth] TRACE 2023/11/26 14:52:00 modbus: send 00 05 00 00 00 06 b4 04 00 6c 00 06
[em-eth] TRACE 2023/11/26 14:52:00 modbus: recv 00 05 00 00 00 0f b4 04 0c ff ff ff ff ff ff ff ff ff ff ff ff
[em-eth] TRACE 2023/11/26 14:52:00 modbus: send 00 06 00 00 00 06 b4 04 00 64 00 01
[em-eth] TRACE 2023/11/26 14:52:00 modbus: recv 00 06 00 00 00 05 b4 04 02 00 42
[em-eth] TRACE 2023/11/26 14:52:00 modbus: send 00 07 00 00 00 06 b4 01 01 90 00 01
[em-eth] TRACE 2023/11/26 14:52:00 modbus: recv 00 07 00 00 00 04 b4 01 01 00
[em-eth] TRACE 2023/11/26 14:52:00 modbus: send 00 08 00 00 00 06 b4 04 00 66 00 02
[em-eth] TRACE 2023/11/26 14:52:00 modbus: recv 00 08 00 00 00 07 b4 04 04 44 59 00 00
Power:          -10W
Energy:         -0.0kWh
Current L1..L3: -0.001A -0.001A -0.001A
Voltage L1..L3: -0.001V -0.001V -0.001V
Charge status:  B
Enabled:        false
Duration:       4h51m37s

```

andig, 2023-11-27:
Muss ja- ich sehs aber auch gerade nicht. Schaue ich mir in Ruhe an.

andig, 2023-11-27:
@g-mocken könntest du für info@evcc.io eine Portweiterleitung zum Charger einrichten? Sieht alles richtig aus…

g-mocken, 2023-11-27:
Schwierig ... was für ein Port/Protokoll genau?

premultiply, 2023-11-27:
TCP/502

g-mocken, 2023-11-27:
Also Modbus? Aber wie schränkt man da den Zugriff auf einen Benutzer wie info@evcc.io ein?
Komplett offen ins Netz stellen will ich das ehrlich gesagt nicht. (Ich kann aber auch selbst einzelne Modbus-Register bedienen, falls es da nur was Low-level auszuprobieren gilt)

andig, 2023-11-27:
Ohne Zugang keine Analyse :/

g-mocken, 2023-11-27:
Hätte ja nichts gegen einen privaten Zugang, aber ich weiß nur, wie man das bei HTTP (Reverse-proxy mit HTTPS und Auth) macht (also z.B. das evcc-WebInterface könnte ich vermutlich so rausreichen). Für Modbus-TCP weiß ich nicht, wie es gehen sollte, außer den Port komplett freizugeben, was mir aber großes Unbehagen bereiten würde.
Gibt's kein log-level, dass die benötigten Daten generieren kann? Ich kann auch den Modbus per Wireshark aufzeichnen, wenn es hilft.

andig, 2023-11-27:
Bleibt nur ssh oder Tailscale

g-mocken, 2023-11-27:
Wegen ssh müsste ich mal schauen, geht aber nicht so schnell.
Das Register phxEMEthRegVoltages liefert (mit QModMaster ausgelesen) in der Tat FFFFFFFF. 
Vielleicht einfach mal vor der condition im Code hart "err = nil" und "b = FFFFFFFF" setzen und schauen was passiert?
Ich kenne die operator precedence usw. in Go nicht, aber in anderen Sprachen würde ich immer Klammern setzen, damit kein Unsinn passiert, also: (err == nil) && (encoding.Int32LswFirst(b) > 0)

andig, 2023-11-28:
Lass und keine Zeit verschwenden. Wenn wir mit dem Debugger rein schauen können dauert das nur 2min. Dafür brauchts Zugang. Ich mache hier mal zu bis das gelingt.

closed 2023-11-28 by andig

andig, 2023-12-01:
Mein Ergebnis:

```
[em-eth] TRACE 2023/12/01 09:59:27 modbus: send 00 01 00 00 00 06 b4 04 00 6c 00 02
[em-eth] TRACE 2023/12/01 09:59:28 modbus: recv 00 01 00 00 00 07 b4 04 04 ff ff ff ff
[em-eth] TRACE 2023/12/01 09:59:28 modbus: send 00 02 00 00 00 06 b4 04 00 64 00 01
[em-eth] TRACE 2023/12/01 09:59:28 modbus: recv 00 02 00 00 00 05 b4 04 02 00 42
[em-eth] TRACE 2023/12/01 09:59:28 modbus: send 00 03 00 00 00 06 b4 01 01 90 00 01
[em-eth] TRACE 2023/12/01 09:59:29 modbus: recv 00 03 00 00 00 04 b4 01 01 01
[em-eth] TRACE 2023/12/01 09:59:29 modbus: send 00 04 00 00 00 06 b4 04 00 66 00 02
[em-eth] TRACE 2023/12/01 09:59:29 modbus: recv 00 04 00 00 00 07 b4 04 04 28 a2 00 00
Charge status: B
Enabled:       true
Duration:      2h53m22s
```

andig, 2023-12-01:
Die Lösung ist hier:

[image]

...also einfach nightly nehmen.