evcc-io/evcc #10940: Meter: Victron Battery kann nicht konfiguriert werden

steve0564, 2023-11-27:
### Describe the bug

Nach Installation der neuesten Nightly von heute kann meine Victron Batterie nicht mehr konfiguriert werden.
Fehlermeldung s. u.

### Steps to reproduce

1. Neustart
2.
3.
...

### Configuration details

```yaml
yaml unverändert seit Wochen

Betroffener Abschnitt:

- type: template  
  template: victron-energy   
  usage: battery    
  host: 192.168.178.103    
  port: 5021    
  name: battery3  
  capacity: 6.6
```

### Log details

```text
Auszug:
[main  ] FATAL 2023/11/27 18:15:56 cannot create meter 'battery3': cannot create meter 'template': cannot create meter 'custom': battery limit soc: convert: invalid conversion: float2int
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.122.1 (65e470bc)

andig, 2023-11-28:
Schuld ist https://github.com/evcc-io/evcc/commit/65e470bc7c5a0522ec0b90d31e842f143394d8e0. Eigentlich hätten die Tests das finden müssen. Ich versuche mal heraus zu finden, warum sie das nicht getan haben.

andig, 2023-11-28:
Closed in https://github.com/evcc-io/evcc/commit/de049f7084f43b72d4fd90fce1b4f3f38e86ce79

closed 2023-11-28 by andig