evcc-io/evcc #10949: Ladevorgang startet erst nach Änderung des Modus

ExecuteGK, 2023-11-28:
### Describe the bug

EVCC ist so konfiguriert, das es bei Erkennung des Fahrzeuges von mode: PV auf mode: minpv wechselt. Dies hat bis etwa Version 0.120.0 problemlos funktioniert und der ID.4 hat nach anstecken mit minimaler Leistung zuzüglick vorhandenem Überschuss geladen.

In der aktuellen Version 0.122.1 wird das Fahrzeug korrekt erkannt und laut Logfile der Modus auf minpv gestellt, allerdings beginnt kein Ladevorgang und im GUI steht weiterhin: Warten auf Fahrzeug. Erst bei einem Wechsel des Modi startet der Ladevorgang. Hierzu reicht ein kurzer Wechsel auf now und zurück nach minpv um wie gewünscht zu laden.

### Steps to reproduce

1. LP konfigurieren das der defualt Modus PV ist.
2. Fahrzeug konfigurieren das der Lademodus minpv ist
3. Fahrzeug verbinden und warten bis es erkannt wurde. Logfile zeigt nun Wechsel auf minpv, Ladvorgang startet aber nicht
4. Wechsel auf Modus now und sofort zurück auf minpv -> Ladevorgang startet.
...

### Configuration details

```yaml
- title: Vorne
  charger: daheimladen
  #  vehicles:
  #  - ev4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

vehicles:
- type: template
  template: id
  title: ID.4
  user: #################
  password: #########
  vin: ################
  capacity: 77
  phases: 3
  mode: minpv
  targetSoC: 80
  minCurrent: 6
  maxCurrent: 16
  cloud: false
  name: ev4
```

### Log details

```text
Nov 28 17:03:45 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 17:03:45 ----
Nov 28 17:03:46 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 17:03:46 charge power: 0W
Nov 28 17:03:46 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 17:03:46 pv power: 0W
Nov 28 17:03:46 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 17:03:46 battery soc: 2%
Nov 28 17:03:46 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 17:03:46 battery power: 0W
Nov 28 17:03:46 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 17:03:46 grid power: 642W
Nov 28 17:03:46 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 17:03:46 site power: 642W
Nov 28 17:03:55 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 17:03:55 ----
Nov 28 17:03:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 17:03:56 charge power: 0W
Nov 28 17:03:56 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 17:03:56 pv power: 0W
Nov 28 17:03:56 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 17:03:56 battery soc: 2%
Nov 28 17:03:56 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 17:03:56 battery power: 0W
Nov 28 17:03:56 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 17:03:56 grid power: 635W
Nov 28 17:03:56 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 17:03:56 site power: 635W
Nov 28 17:03:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 17:03:56 charge voltages: [224 223 221]V
Nov 28 17:03:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 17:03:56 detected connected phases: 3p
Nov 28 17:03:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 17:03:56 charge currents: [0 0 0]A
Nov 28 17:03:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 17:03:56 charge total import: 3423.000kWh
Nov 28 17:03:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 17:03:56 charger status: B
Nov 28 17:03:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 17:03:56 charge start energy: 3.42e+03kWh
Nov 28 17:03:56 flowcontrol evcc[676734]: [lp-2  ] INFO 2023/11/28 17:03:56 car connected
Nov 28 17:03:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 17:03:56 vehicle api refresh
Nov 28 17:03:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 17:03:56 pv timer elapse
[... 160 lines cut ...]
Nov 28 19:07:56 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 19:07:56 pv power: 0W
Nov 28 19:07:56 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 19:07:56 battery soc: 1%
Nov 28 19:07:56 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 19:07:56 battery power: 0W
Nov 28 19:07:56 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 19:07:56 grid power: 11132W
Nov 28 19:07:56 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 19:07:56 site power: 11132W
Nov 28 19:07:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 19:07:56 charge voltages: [226 220 214]V
Nov 28 19:07:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 19:07:56 detected connected phases: 3p
Nov 28 19:07:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 19:07:56 charge currents: [14.8 15 15]A
Nov 28 19:07:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 19:07:56 detected active phases: 3p
Nov 28 19:07:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 19:07:56 charge total import: 3423.000kWh
Nov 28 19:07:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 19:07:56 charger status: C
Nov 28 19:07:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 19:07:56 soc estimated: 35.00% (vehicle: 35.00%)
Nov 28 19:07:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 19:07:56 vehicle soc: 35%
Nov 28 19:07:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 19:07:56 vehicle soc limit: 80%
Nov 28 19:07:56 flowcontrol evcc[676734]: [lp-2  ] DEBUG 2023/11/28 19:07:56 vehicle range: 135km
Nov 28 19:08:05 flowcontrol evcc[676734]: [site  ] DEBUG 2023/11/28 19:08:05 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.122.1

andig, 2023-11-30:
According to your log, charging starts just fine in minpv:

> DEBUG 2023/11/28 17:04:56 charger status: C

ExecuteGK, 2023-11-30:
Yes, but it does not charge unless you switch back and forth.This is what I do not understand. Might be a charger issue as well.I will try to replicate this with my other wallbox in order to see if it's box related once the car is back.Best Regards,SebastianVon meinem/meiner Galaxy gesendet

closed 2023-11-30 by andig

ExecuteGK, 2023-11-30:
> The wb says charging- so thats what it thinks it does. Closing as likely wb issue.

Quickly closing with no proper troubleshooting does not help.As I wrote, it worked with an older release. I would be surprised if the Box broke suddenly.Will test further with other box and older release on same box.SebastianVon meinem/meiner Galaxy gesendet

andig, 2023-11-30:
Da haste recht. Da der Charger aber „lädt“ sagt kann der Fehler nir da liegen. Bitte trace log für den Charger, dann können wir nochmal schauen. Bitte gleiches Log auch für die funktionierende alter Version zum Vergleich.

PeterPablo, 2023-12-01:
Ich hatte das mit einer Daheimladen-Wallbox auch schon mehrfach beobachtet, aber bisher keine zuverlässige Reproduktion hinbekommen und deswegen auch keine Logs parat. Mein Kommentar hilft wenig, außer zu zeigen, dass das Phänomen nicht nur bei @ExecuteGK auftritt.

andig, 2023-12-01:
> mehr infos aus dem charger zu bekommen, scheint aber so nicht zu gehen.....

Charger auf trace, nicht Ladepunkt! Also `daheimladen`.

andig, 2023-12-02:
> Als es nicht gestartet ist, habe ich folgende meldungen bekommen:

Der relevante Teil ist, dass der Charger sagt "lädt". Den hast Du weg gelassen. Fakt ist, dass DL im ersten Log oben Ladung angibt. Wenn das nicht stimmt ists ein Implementierungs- oder Serverproblem bei DL. Ich würde vorschlagen, den Support zu kontaktieren und/oder einfach mal die Variante mit Modbsu auszuprobieren.

Mangels trace Log gibts hier weiter nichts zu tun.

ExecuteGK, 2023-12-02:
Hi,Das ist die Variante mit Modbus.Die läuft erheblich besser als die Server Variante. Die Server Variante müsste mittlerweile auch abgeschaltet sein.Wenn ich das Versuche weiter zu korreliert, dann haben die Probleme erst angefangen als bei Ladebegin kein PV Überschuss mehr da war. Im Sommer hatte ich das, soweit ich mich erinnern kann garnicht.Gestern hab ich das wieder nicht reproduziert bekommen. Bis auf ein Problem mit dem VW Login mal wieder (was auch echt nervt..... kann VW keine ordentliche API anbieten....)Ich behalte das weiter im Auge. SebastianVon meinem/meiner Galaxy gesendet

andig, 2023-12-02:
> Hi,Das ist die Variante mit Modbus

Wie kommst du darauf?

ExecuteGK, 2023-12-02:
Weil ich kein daheimladen Token geladen hab und die Box direkt per IP anspreche.GrüßeSebastianVon meinem/meiner Galaxy gesendet