evcc-io/evcc #10950: Ladeplanung im Winter mit CO2 Tarif min. 1h vor Abfahrt Laden starten

N1c081, 2023-11-28:
Erklärung:
Da ich nun verstanden habe, dass sobald man den CO2 Tarif in der evcc.yaml aktiviert hat immer darauf geachtet wird bei der Ladeplanung so wenig CO2 Entstehung wie möglich zu verursachen bis zum Abfahrtszeitpunkt.  Oder aber den vom Nutzer  eingestellt max. CO2 Wert.
Das resultiert in teils unterbrochenen Ladezeit, die auch nicht zwingend bis zum Abfahrtszeitpunkt andauern müssen.
Bsp.:
Zeitpunkt der Einstellung 19:00Uhr
Abfahrt Tags darauf 7:00Uhr
Ladedauer bis gewünschter SoC 2h
niedrigster kumilierter CO2 Auststoss 1-2Uhr und 4-5Uhr

Daraus ergibt sich das EVCC das BEV 1x von 1-2 Uhr belädt und nochmal 4-5Uhr eine Ladung durchführt. Obwohl der Abfahrtszeitpunkt auf 7 Uhr gesetzt ist.

Alles gut soweit weil ja so vorgesehen.

Nun wäre es aber gerade im Winter effizienter wenn vor der Abfahrt der Ladevorgang immer auf ca. 1h vor diesem begonnen wird.
Grund: Da ich zumindest mein BEV vorheize bzw. es ab Fahrtbeginn wärme würde es effizienter sein die Ladung 1h vor Abfahrt zu starten da bei der Ladung der Akku vorgeheizt wird. Diese Wärme die für den Akku erzeugt wird kann man so effizienter nutzen als wenn diese erzeugt Wärme nach 2h wie im Bsp. verpuffen würde und bei Abfahrt erneut von "Null" angefangen wird.

Ich hoffe ich konnte das einigermaßen erklären.

Eine GUI Anmerkung hätte ich da auch noch. 
Ein Hinweis im Ladeplanungsfenster bei aktiviertem CO2 Tarif das darauf geachtet wird möglichst wenig CO2 zu erzeugen und dadurch der Ladezeitpunkt nicht immer mit dem Zeitpunkt der Abfahrt übereinstimmen muss wäre bestimmt für viele Nutzer aufschlussreich. Denn das war mein Verständnisproblem. Dachte immer es wäre ein Fehler in der Software.

LG Nico

andig, 2023-11-30:
@N1c081 um so spät wie möglich zu starten könntest/müsstest du im Winter den Co2 Tarif raus nehmen. Ich glaube einen Sonderfall („immer 1h vor Abfahrt“) wollen wir egtl. Nicht einbauen.

/cc @schenlap was meinst du?

StefanSchoof, 2023-11-30:
Ich hatte auch schon den Fall, dass ich gerne einen Plan ohne CO2 Tarif machen wollte. Bei mir ging es vor einer längeren Fahrt, das Auto nicht länger mit 100% stehen zu haben und das Laden Just in Time fertig zu haben. Dafür die config anpassen, Auto abstecken (meine Elli Box mag leider kein evcc restart während das Auto verbunden ist) evcc neu starten ist mein Workaround. Der ist aber recht aufwendig. Außerdem wird dann für die Ladung das CO2 nicht erfasst.

Für mich wäre es ein Mehrwert, wenn ich den Planner von CO2 auf so spät wie möglich ohne umstellen könnte ohne die config anzupassen zu müssen.

andig, 2023-11-30:
Dafür müssten wir erlauben, entweder "spät" auszuwählen oder die Plangrundlage zu ändern (was aber- falls du einen flexiblen Tarif hast)- das Problem nicht löst.

Ich hänge eher an der Anforderung. Das Auto 8h früher oder später zu laden ändert für den Akku gar nichts. Spät laden wäre für mich dann ein Vorteil, wenn wir die ohnehin schwierige Erkennung der Vorklimatisierung dafür raus werfen könnten. 😝

premultiply, 2023-11-30:
> Das Auto 8h früher oder später zu laden ändert für den Akku gar nichts.

Das kann man so nicht sagen. Bei eisigen Temperaturen kühlt der Akku schon gut aus.
Wenn es ohnehin knapp mit der Reichweite ist ist das schon relevant - jedenfalls bei mir gelegentlich.

Daher stelle ich dann momentan so spät wie möglich auf "Sofort" bis 100% um sicherzugehen dass wirklich geladen wird und der Strom für die Vorklima und Akkuheizung an bleibt.
Denn nicht voller Akku im eiskalten Auto zur Abfahrt wäre ganz ungünstig.

N1c081, 2023-11-30:
> @N1c081 um so spät wie möglich zu starten könntest/müsstest du im Winter den Co2 Tarif raus nehmen. Ich glaube einen Sonderfall („immer 1h vor Abfahrt“) wollen wir egtl. Nicht einbauen.
> 
Das habe ich nun auch so gemacht und habe den CO2 Tarif deaktiviert.
Für mich ist es nicht so sehr relevant da ich möglichst PV Übschussladen mache. 

Ausschlaggebend für den Post waren für mich die Punkte:

- Wärmeenergie gut nutzen
-  wenn Akku gewärmt 
       -  max. mögliche Reichweite ab Fahrtbeginn da keine Energie aus dem Akku für Akkuerwärmung aufgewendet werden
          muss
       - Rekuperation funktioniert wie gewohnt und daher kein"anderes" Ansprechverhalten des Fzgs (gerade für meine Frau
          wichtig)
       - volle Wärmeerzeugung steht für den Fahrgastraum zur Verfügung

Akkugesundheit kümmert sich das BMS drum und das macht es gut.

Evtl. kann der Nutzer bei aktiviertem CO2 Tarif aktiv darauf hingewiesen werden wie das Ladeverhalten von EVCC ist und ob ggf. der CO2geführte Tarif temp. deaktiviert werden soll für diesen einen Ladevorgang. So hat der Nutzer immer die Auswahl und bekommt einen Einblick wie der CO2 Tarif funktioniert. Denn wie schon geschrieben war mir das nicht klar.

Es hat übrigens keine Auswirkungen wenn man in der Ladeplanungs-GUI den CO2 Wert auf "keine" setzt. Dann wird nur nicht ein spezieller eingestellter Grenzwert genommen, aber weiterhin den möglichst niedrigsten GesamtWert bis Abfahrt.

Mir ist der Post ja nur wichtig da es immer mehr Nutzer werden die nicht den technischen Einblick haben und sich evtl. von Freunden  EVCC einrichten lassen und sich dann auf die Einstellungen der GUI verlassen, was ja an sich auch OK so ist. Weil jeder sollte eine möglichst einfache Ladeplanung haben mit all diesen echt coolen Features die EVCC bietet.
Bitte auch nicht als nörgeln verstehen denn Ihr habt hier schon einiges geleistet und seit immer dabei EVCC weiter zu entwickeln. Danke nochmal dafür.

> Ich bezog mich auf die Akkugesundheit. Betriebstemperatur ist natürlich ein Thema. Also: wäre schön, das dem Planner als "spät" mitteilen zu können. Der Plan dafür wäre ja sehr einfach :)

Dem würde ich so zusagen. Klasse.

LG Nico

andig, 2023-11-30:
> Bitte auch nicht als nörgeln verstehen

absolut nicht. Die use cases kommen immer aus der Praxis und jeder von uns hat da seine eigenen Anforderungen. Ist doch eine gute Symbiose 👍🏻. Manchmal dauerts halt bis der Groschen fällt 🥳

andig, 2023-11-30:
@naltatis hast du eine UI-Idee wie wir das dem Planner verkaufen wollen?

StefanSchoof, 2023-12-01:
> Selbst wenn wir demnächst mehrere Ladepläne haben #5492 glaube ich, dass das ok ist, das nicht pro Ladepan zu speichern. Wie seht ihr das?

Ich denke es gebe bestimmt einige interessante Möglichkeiten, das pro Ladeplan zu haben, wie z.B nachts CO2 laden und dann über Tag zum Abend hin so viel Sonne wie möglich nehmen und den Rest also so spät wie möglich. Aber wenn das so viel Komplexität rein bringt ist eine Einstellung auch schon was tolles.

mafde2023, 2023-12-02:
Interessante Diskussion, das Thema beschäftigt mich auch gerade. Ich lade aktuell CO2 optimiert, Ziel 6Uhr um das Netz zu schonen und um flexibel früh starten zu können, würde aber gerne zum Vorklimatisieren aus dem Netz Strom beziehen um beim Hybrid die Reichweite zu schonen. Ich würde daher persönlich bevorzugen, wenn die Erkennung der Vorklimatisierung sauber funktioniert, da in dem Fall eh das Auto bestimmt wann und wie die Vorklimatisierung läuft. --> AdHoc bzw. je nach Außentemperatur und Abfahrtszeitpunkt variable Vorlaufzeiten, die vom Auto bestimmt werden. Macht also wenig Sinn da irgendwelche pauschalen Zeiten einzubauen und dann in EVCC und im Auto Einstellungen vornehmen zu müssen und zu hoffen das das irgendwie zusammenpasst.

N1c081, 2023-12-02:
Hallo mafde2023 wenn ich das richtig verstanden habe machst du es so das du unabhängig von dem tatsächlichen Abfahrtszeitpunkt bis 6Uhr vollgeladen willst. Da du dann aber später losfährst willst du die Vorklimatisierung nutzen?

Dann ist natürlich der Punkt den ich wichtig fand und mit der Auswahlmöglichkeit die naltatis vorgeschlagen hat nicht in Übereinstimmung und du bräuchtest explizit die Erkennung der Vorklimatisierung. 

Die Idee dahinter von mir mit der festen Ladung 1h vor Abfahrt war ja, da die Batterie beim Ladevorgang sowieso auf Temperatur gebracht wird, diese eingesetzte Energie nicht wieder an die Umgebung abzugeben nur weil der reale Zeitpunkt der Abfahrt später liegt und die Batterie dann wieder auskühlt. Dann würde ja bei Abfahrt wieder die Batterie zusätzlich von „kalt“ auf Betriebstemperatur gebracht werden müssen was wieder zusätzliche Energie erfordert.

Oder habe ich dich falsch verstanden?

LG

andig, 2023-12-02:
> sondern eher AdHoc, sodass ich 15-20min vor Abfahrt die Klima starte und genau dann soll die Wallbox in den Lademodus gehen

Das ist dann hier aber völlig OT...

mafde2023, 2023-12-02:
Ok, um nochwas beizutragen, einen Switch für die Ladestrategie (normal/co2/preis) pro Ladepunkt würde ich auch begrüßen, würde mir helfen. Im Menü für die Phasenwahl wäre cool :-)

closed 2023-12-28 by github-actions[bot]

StefanSchoof, 2023-12-29:
Ist das ein Feature, dass es in die backlog schafft oder fällt es aufgrund von anderen Prioritäten raus?

Ich würde mich über die Lösung aus https://github.com/evcc-io/evcc/issues/10950#issuecomment-1836340840 freuen.

schwarmik, 2024-03-07:
> Selbst wenn wir demnächst mehrere Ladepläne haben #5492 glaube ich, dass das ok ist, das nicht pro Ladepan zu speichern. 
> 

ich würde mich da gerne nochmal dranhängen - ich fahre seit letzter Woche ein "eigenes" Model Y SR und will mich gerade fit schießen was den schonenden Umgang mit dem Akku betrifft. 

Im [teslamag](https://teslamag.de/news/daten-tausende-tesla-batterien-lange-leben-faktoren-62454) hab ich dazu ein interessantes Interview gelesen mit dem Entwickler vom Teslalogger, ein Ausschnitt daraus:

*Im Winter sollte man möglichst mindestens eine halbe Stunde vor der Fahrt das Auto laden, sodass durch den Ladevorgang die Batterie (kostenlos) schonend gewärmt wird. Das steigert die Lebensdauer und senkt den Verbrauch. Ich stecke abends bei mir das Ladekabel ein und gebe eine geplante Ladung ab 7 Uhr vor. Wenn ich dann vor 8 Uhr losfahre, habe ich meist eine warme Batterie.*

Ich hab es ehrlich gesagt nicht ganz geschnallt ob und wie solch ein Feature jetzt geplant ist - ich lade (neben Überschuss) bis jetzt nur preis-optimiert via tibber. Jetzt fehlt mir eben die Möglichkeit die letzte Stunde in der Ladeplanung immer als "fix und gesetzt" auszuwählen. Geht das über eine zweite Ladeplanung (wenn diese umgesetzt sind) oder bitte um kurze Erklärung wie das gedacht ist.
Danke!!

naltatis, 2024-03-08:
Wie genau wir aufeinanderfolgende Ladepläne (bspw. 70% bis 7 Uhr, 80% bis 8 Uhr) abbilden ist noch nicht ganz geklärt.
Eine Variante wäre zu sagen, dass wir erst, wenn die Zielzeit eines Plans (70% 7 Uhr) erreicht ist kümmern wir uns um den nächsten Plan (80% 8 Uhr). Das würde dann in deinem Anwendungsfall genau das gewünschte Ergebnis liefern. Möglicherweise gibts aber auf Fälle, in denen das nicht gewünscht ist.

Ich nehm das aber mit in die konkrete Ausgestaltung in https://github.com/evcc-io/evcc/issues/5492