evcc-io/evcc #10961: Ford (Mustang Mach-E): 403 forbidden

miniluma, 2023-11-30:
### Describe the bug

The connection to the Ford API stopped working for a couple of days.
No changes in my config, no changes in my login data. Ford App works as expected.

Syslog states:
`creating vehicle ev5 failed: cannot create vehicle 'template': cannot create vehicle 'ford': login failed: unexpected status: 403 (Forbidden)`

I attached the trace output which shows "Service unavailable" HTML page as a response to the GET.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
- name: ev5
  type: template
  template: ford
  title: Mustang Mach-E # Wird in der Benutzeroberfläche angezeigt (optional)
  user: email # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
  password: "password" # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
  vin: myVIN # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
  capacity: 75.7 # Akkukapazität in kWh (optional)
```

### Log details

```text
ov 30 08:07:18 pi4 evcc[139796]: [ford  ] TRACE 2023/11/30 08:07:18 GET https://sso.ci.ford.com/v1.0/endpoint/default/authorize?...
Nov 30 08:07:18 pi4 evcc[139796]: [ford  ] TRACE 2023/11/30 08:07:18 <html>
Nov 30 08:07:18 pi4 evcc[139796]: <title>Service unavailable</title>
Nov 30 08:07:18 pi4 evcc[139796]: <head>
Nov 30 08:07:18 pi4 evcc[139796]: #011<link rel="stylesheet" type="text/css" href="https://unpkg.com/carbon-components/css/carbon-components.min.css">
Nov 30 08:07:18 pi4 evcc[139796]: #011<script src="https://unpkg.com/carbon-components/scripts/carbon-components.min.js"></script>
Nov 30 08:07:18 pi4 evcc[139796]: #011<meta http-equiv="Content-Type" content="text/html; charset=utf-8"/>
Nov 30 08:07:18 pi4 evcc[139796]: #011<style>
...
```

### What type of operating system are you running?

Linux

### Version

0.122.1

andig, 2023-11-30:
Ford hat mal wieder den Mechnismus geändert- leider in etwas sehr Kompliziertes... Workaround: Integration per ioBroker.

miniluma, 2023-11-30:
Naja, in HA wurde es auch adaptiert. Wirkt jetzt auf mich nicht so sehr viel komplexer als vorher.
https://github.com/itchannel/fordpass-ha/commit/306900b6b713ed4759533fc26aed4a58261803f4

GrimmiMeloni, 2023-12-03:
I have an old Ford Login - I will see if I can get this adjusted based on the changes they did in HA.

andig, 2023-12-05:
Could someone please share login credentials with info@evcc.io?

closed 2023-12-05 by andig

GrimmiMeloni, 2023-12-07:
@appcoders mal im Browser komplett durchgespielt? Müssen da vielleicht neue TOS oder ähnliches akzeptiert werden?

PR #10986 (2023-12-05): Ford: adjust login flow from password to code grant
fix #10961