evcc-io/evcc #10974: Energy Ausgabe von Shelly Steckdose stimmt nicht

VolkerK62, 2023-12-02:
### Describe the bug

Ich habe einen Shelly PlugS als Charger für meine eBike.
Die Anzeige der geladenen Energie pro Session erscheint schlüssig.
Aber der Gesamtverbrauch (Zählerstand) stimmt nicht mit dem überein, was der Shelly im JSON ausgibt.
Der Shelly meldet 23,389 kWh.
`evcc charger` gibt aber nur 0,4 kWh aus.

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
-------
[shelly] TRACE 2023/12/02 20:08:33 GET http://192.168.178.83/status
[shelly] TRACE 2023/12/02 20:08:33 {"wifi_sta":{"connected":true,"ssid":"KAS1","ip":"192.168.178.83","rssi":-80},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":false},"time":"20:08","serial":1,"has_update":false,"mac":"BCFF4D4381CD","relays":[{"ison":false,"has_timer":false,"overpower":false}],"meters":[{"power":0.00,"is_valid":true,"timestamp":1701547714,"counters":[0.000, 0.000, 0.000],"total":23389}],"temperature":13.09,"overtemperature":false,"update":{"status":"unknown","has_update":false,"new_version":"","old_version":"20190516-073020/master@ea1b23db"},"ram_total":50832,"ram_free":40268,"fs_size":233681,"fs_free":171182,"uptime":1045575}
[shelly] TRACE 2023/12/02 20:08:33 GET http://192.168.178.83/status
[shelly] TRACE 2023/12/02 20:08:33 {"wifi_sta":{"connected":true,"ssid":"KAS1","ip":"192.168.178.83","rssi":-80},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":false},"time":"20:08","serial":1,"has_update":false,"mac":"BCFF4D4381CD","relays":[{"ison":false,"has_timer":false,"overpower":false}],"meters":[{"power":0.00,"is_valid":true,"timestamp":1701547714,"counters":[0.000, 0.000, 0.000],"total":23389}],"temperature":13.09,"overtemperature":false,"update":{"status":"unknown","has_update":false,"new_version":"","old_version":"20190516-073020/master@ea1b23db"},"ram_total":50832,"ram_free":40256,"fs_size":233681,"fs_free":171182,"uptime":1045575}
[shelly] TRACE 2023/12/02 20:08:33 GET http://192.168.178.83/status
[shelly] TRACE 2023/12/02 20:08:33 {"wifi_sta":{"connected":true,"ssid":"KAS1","ip":"192.168.178.83","rssi":-80},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":false},"time":"20:08","serial":1,"has_update":false,"mac":"BCFF4D4381CD","relays":[{"ison":false,"has_timer":false,"overpower":false}],"meters":[{"power":0.00,"is_valid":true,"timestamp":1701547714,"counters":[0.000, 0.000, 0.000],"total":23389}],"temperature":13.09,"overtemperature":false,"update":{"status":"unknown","has_update":false,"new_version":"","old_version":"20190516-073020/master@ea1b23db"},"ram_total":50832,"ram_free":40268,"fs_size":233681,"fs_free":171182,"uptime":1045575}
[shelly] TRACE 2023/12/02 20:08:33 GET http://192.168.178.83/relay/0
[shelly] TRACE 2023/12/02 20:08:33 {"ison":false,"has_timer":false,"overpower":false}
Power:         0W
Energy:        0.4kWh
Charge status: B
Enabled:       false
Features:      [IntegratedDevice]
```

### What type of operating system are you running?

Linux

### Version

0.122.1

andig, 2023-12-03:
/cc @thierolm

VolkerK62, 2023-12-03:
hat sich vermutlich erledigt.
Der Shelly meldet  `"total":23389}`. Ich dachte das sind `Wh`. 
Wenn das `Wm` sind, ergibt es umgerechnet "390 Wh", dann stimmt "0,4 kWh".

Trotzdem die Frage an @thierolm : 
In der vorletzten Session war der Endstand "4,3 kWh". Die letzte startet dann bei "0 kWh".
Wird das resettet, wenn man den Shelly vom Strom nimmt?

closed 2023-12-03 by VolkerK62

mdkeil, 2023-12-03:
`Wird das resettet, wenn man den Shelly vom Strom nimmt?`

Das ist der Nachteil fast aller Shelly Geräte, dass der Energiewert nur im flüchtigen Speicher liegt.. sobald Strom weg, ist auch der Wert resetet.

VolkerK62, 2023-12-03:
Danke.

thierolm, 2023-12-03:
> Wird das resettet, wenn man den Shelly vom Strom nimmt?

Ja, so ist das leider. Auch bei einem Firmware-Update wird der Wert zurückgesetzt.