evcc-io/evcc #10976: configUI: Saving vehicle takes long, leads to duplicates due to double clicks

GrimmiMeloni, 2023-12-03:
### Describe the bug

I just tried adding a vehicle via the new config UI.
After filling the form, I clicked save at the bottom. This operation did not respond quickly with a success message (could be my poor Synology being busy).
As the save button was not greyed out, I clicked it again, as I felt the system had ignored my initial click.
Afterwards I got a success message. Open closing the form, I saw I now had the vehicle added twice.

### Steps to reproduce

1. create a vehicle via config ui.
2. click save
3. click save again
4. a duplicate car is created

### Configuration details

```yaml
N/A
```

### Log details

```text
N/A
```

### What type of operating system are you running?

Docker container

### Version

nightly - d63dec95

VolkerK62, 2023-12-05:
In dem Zusammenhang. 
Wenn man ein per UI erstelltes Fahrzeug löscht, wird es danach am Loadpoint in der Fahrzeugliste immer noch angezeigt. Erst nach Aktualisierung der Seite (F5) ist es weg.

andig, 2023-12-30:
Will be closed by config ui.

closed 2023-12-30 by andig