evcc-io/evcc #10980: 1 error(s) decoding: * '' has invalid keys: resetondisconnect

lindeEV, 2023-12-03:
### Describe the bug

Nach dem Update auf 0.122.1
wird mir angezeigt: 1 error(s) decoding: * '' has invalid keys: resetondisconnect

Ich habe es jetzt ausgeklammert als workaround. Wegen Winterbetrieb im now gerade nicht so wichtig.

### Steps to reproduce

1. Update auf 0.122.1
2. restart
3. 1 error(s) decoding: * '' has invalid keys: resetondisconnect
4. Ich habe es jetzt ausgeklammert als workaround. Wegen Winterbetrieb im minpv gerade nicht so wichtig.

### Configuration details

```yaml
loadpoints:  
- title: Stellplatz  
  charger: wallbox3  
  vehicle: ev2  
  mode: minpv  
  phases: 3  
  mincurrent: 6  
  maxcurrent: 16  
  **resetOnDisconnect: true**
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.122.1

closed 2023-12-03 by andig

lindeEV, 2023-12-03:
@andig: Ja, sorry. Ich bin tatsächlich im Nightly. Dann muss ich die geänderte Syntax in der yaml hier herausfinden, ja? Betrifft ja offenbar auch den Target SOC

Danke für den Hinweis!

andig, 2023-12-06:
Da gibts wenig rauszufinden- schau Dir einfach die Fehlermeldung an, die sagt was weg muss.