evcc-io/evcc #10996: Loadpoint kann nicht eingerichtet werden

steve0564, 2023-12-04:
### Describe the bug

Bei den letzten beiden Nightly-Versionen kann der Loadpoint nicht eingerichtet werden.

### Steps to reproduce

1. Restart
2. Fehlermeldung
3.
...

### Configuration details

```yaml
unverändert
```

### Log details

```text
Auszug aus dem Startlog:

Dez 04 08:28:54 raspberrypi evcc[6652]: [main  ] FATAL 2023/12/04 08:28:54 failed configuring loadpoints: failed configuring loadpoint: 1 error(s) decoding:
Dez 04 08:28:54 raspberrypi evcc[6652]: * '' has invalid keys: resetondisconnect
Dez 04 08:28:54 raspberrypi evcc[6652]: [main  ] FATAL 2023/12/04 08:28:54 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.122.1+1701569822

VolkerK62, 2023-12-04:
BC   https://github.com/evcc-io/evcc/discussions/10978

steve0564, 2023-12-04:
Viel zu tun, danke!

TobiasHuber1980, 2023-12-04:
in deiner yaml unter loadpint den eintrag resetondisconnect entfernen oder mit # davor auskommentieren.

closed 2023-12-04 by steve0564