evcc-io/evcc #11018: 1 error(s) decoding: * '' has invalid keys: resetondisconnect

Kludi2019, 2023-12-05:
### Describe the bug

Seit dem gestrigen Update kommt obige Fehlermeldung. Sobald ich resetOnDisconnect: true herausnehme läuft evcc einwandfrei. Ich kann keinen Fehlen in meiner yaml Datei finden. 

Vielen Dank für den tollen Adapter
Gruß Gerhard

### Steps to reproduce

1. ich habe bei beiden loadpoints resetondisconnect per # zu Text gemacht und mehrfach evcc gestoppt und gesatartet.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: debug  
levels:  
  cache: error  
    
# unique installation id  
plant: *****  
  
interval: 10s # control cycle interval  
  
meters:  
#- type: template  
  #template: sonnenbatterie   
  #usage: grid    
  #host: 192.168.178.33    
  #port: 8080    
  #name: grid1  
    
- type: template  
  template: tasmota  
  usage: grid  
  host: 192.168.178.61  
  user: *****  
  password: *****  
  name: grid1  
- type: template  
  template: sonnenbatterie   
  usage: pv    
  host: 192.168.178.33    
  port: 8080    
  name: pv2  
- type: template  
  template: sonnenbatterie   
  usage: battery    
  host: 192.168.178.33    
  port: 8080    
  name: battery3  
  
- type: template  
  name: Mobilladegerat_Pwr  
  template: shelly-1pm  
  user: *****  
  password: *****  
  usage: charge  
  host: 192.168.178.26  
    
chargers:  
- type: template  
  template: ellipro   
  ski: EE38 62DB xxxx xxxx xxxx    
  ip: 192.168.178.74    
  name: wallbox  
  
- type: template  
  template: shelly  
  host: 192.168.178.26  
  standbypower: 5  
  user: *****  
  password: *****  
  name: Mobilladegeraet  
  
vehicles:  
- type: template  
  template: fiat  
  title: Paulchen  
  user: *****  
  password: *****  
  vin: *****  
  capacity: 42  
  pin: *****  
  name: ev4  
  
- type: template  
  template: id  
  title: Id5  
  user: *****  
  password: *****  
  vin: *****  
  capacity: 77  
  minSoc: 40  
  targetSoc: 80  
  name: ev1  
    
loadpoints:  
- title: Elli  
  charger: wallbox  
  mode: pv  
  phases: 3  
  mincurrent: 6  
  maxcurrent: 16  
  resetOnDisconnect: true  
  
- title: mobil  
  charger: Mobilladegeraet  
  meter: Mobilladegerat_Pwr  
  mode: off  
  phases: 1  
  mincurrent: 6  
  maxcurrent: 10  
  #resetOnDisconnect: true  
    
mqtt:  
  broker: 192.168.178.32:1887  
  topic: evcc  
  clientid: foo  
  user: *****  
  password: *****  
  
tariffs:  
  currency: EUR  
  grid:  
    type: fixed  
    price: 0.368  
  
  feedin:  
    type: fixed  
    price: 0.082  
  
site:  
  - title: Im Schapenkamp 7  
    meters:  
      grid: grid1  
      pv: pv2  
      battery: battery3  
    residualPower: 300
```

### Log details

```text
FEHLER BEIM STARTEN
failed configuring loadpoints
failed configuring loadpoint
1 error(s) decoding: * '' has invalid keys: resetondisconnect
Bitte überprüfe deine Konfigurationsdatei. Sollte dir die Fehlermeldung nicht weiterhelfen, suche in unseren GitHub Diskussionen nach einer Lösung.

Hinweis: Ein weiterer Grund könnte ein fehlerhaftes Gerät (Wechselrichter, Zähler, …) sein. Überprüfe deine Netzwerkverbindungen.

Konfiguration
Verwendete Konfigurationsdatei: /etc/evcc.yaml
Behebe das Problem und starte den Server neu.
```

### What type of operating system are you running?

Linux

### Version

122.1

VolkerK62, 2023-12-05:
https://github.com/evcc-io/evcc/discussions/10978

Kludi2019, 2023-12-05:
Vielen Dank für die schnelle Antwort,  sorry, das hatte ich nicht gesehen.