evcc-io/evcc #11020: Nach dem Einstecken des Fahrzeugs an der Wallbox wird der falsche Ladezustand angezeigt

JamyMartini, 2023-12-05:
### Describe the bug

Wenn das Fahrzeug "eC4" nach einer Fahrt an der Wallbox angeschlossen wird, zeigt die UI von evcc den falschen Ladezustand des Fahrzeugs an, im Beispiel 80% statt 66%. Abhilfe: Nach einem Neustart des evcc-Dienstes wird der korrekte Ladezustand (im Beispiel also 66%) angezeigt.

Beim anderen Fahrzeug "eBerlingo", das üblicherweise an einer anderen Wallbox angeschlossen wird, tritt dieser Fehler nicht auf.

[image]
[image]

### Steps to reproduce

1. Dienst evcc ist gestartet.
2. Beide Fahrzeuge sind angeschlossen.
3. Die beiden Wallboxen haben ein gemeinsames Lastmanagement von insgesamt 10kW.
4. Die Wallbox "Müllbox" / "eC4" hat Priorität 5.
5. Die Wallbox "Garage" / "eBerlingo" hat Priorität 1.
6. Die Ladeplanung wird für beide Fahrzeuge auf 7:00 nächster Tag eingestellt. (Die Ladeplanung wird jeden Abend händisch aktiviert.)
7. Beide Fahrzeuge werden nach Tibber-Tarif geladen, das Lastmanagement wird dabei eingehalten und die Reihenfolge der Ladung läuft nach der Priorität ab.
8. Die Ladeplanung ist spätestens zur eingestellten Zeit  fertig.
9. Fahrzeug "eBerlingo" wird abgetrennt und fährt.
10. Fahrzeug "eC4" wird abgetrennt und fährt.
11. Fahrzeug "eC4" wird wieder angeschlossen.
12. Fehler tritt auf. (1. Screenshot mit rotem Kringel)
13. Fahrzeug "eBerlingo" wird wieder angeschlossen.

ABHILFE:
14. Über die UI: Hilfe>Neustart oder Kommandozeile: "sudo systemctl restart evcc" den Dienst evcc neu starten.
15. Der Fehler ist behoben, der korrekte Ladezustand wird angezeigt. (2. Screenshot mit grünem Kringel)

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: ----------------------------------------------------------------

sponsortoken: 
------------------------------------.-----------------------------------------------------------------------------------------------.-------------------------------------------

interval: 10s # Definiert das zeitliche Interval, in welchem neue Werte von allen Messgeräten gelesen werden
              # und die Ladeströme der Wallboxen neu geregelt wird.

meters:
- type: template
  template: sma-energy-meter
  usage: grid
  host: xx.xx.xx.28
  name: grid_SMA_EnergyMeter
- type: template
  template: sma-inverter
  usage: battery
  host: xx.xx.xx.27
  password: -------------------
  name: battery_SMA_SunnyBoyStorage
- type: template
  template: sma-inverter
  usage: pv
  host: xx.xx.xx.52
  password: -------------------
  name: pv_SMA_TriPower
- type: template
[... 307 lines cut ...]
[lp-1  ] DEBUG 2023/12/05 22:16:25 charger status: B
[lp-1  ] DEBUG 2023/12/05 22:16:25 pv charge current: 0A = 0A + -6.65A (1529W @ 1p)
[site  ] DEBUG 2023/12/05 22:16:35 ----
[lp-1  ] DEBUG 2023/12/05 22:16:35 charge power: 0W
[lp-2  ] DEBUG 2023/12/05 22:16:35 charge power: 0W
[site  ] DEBUG 2023/12/05 22:16:35 pv power: 0W
[site  ] DEBUG 2023/12/05 22:16:35 battery soc: 59%
[site  ] DEBUG 2023/12/05 22:16:35 battery power: 0W
[site  ] DEBUG 2023/12/05 22:16:35 grid power: 1029W
[site  ] DEBUG 2023/12/05 22:16:35 grid powers: [16 869 144]W
[site  ] DEBUG 2023/12/05 22:16:35 grid currents: [0.713 4.4 1.15]A
[site  ] DEBUG 2023/12/05 22:16:35 site power: 1529W
[lp-2  ] DEBUG 2023/12/05 22:16:35 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2023/12/05 22:16:35 charge total import: 2207.178kWh
[lp-2  ] DEBUG 2023/12/05 22:16:35 charger status: B
[lp-2  ] DEBUG 2023/12/05 22:16:35 pv charge current: 0A = 0A + -6.65A (1529W @ 1p)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.122.1

andig, 2023-12-06:
Vmtl. Liefert das Citroen Api einen falschen/veralteten Wert. Bitte trace log vom Auto dafür anlegen.

closed 2023-12-06 by andig

JamyMartini, 2023-12-09:
Der Fehler ist wieder aufgetreten, dieses Mal beim Fahrzeug "eBerlingo".
evcc zeigt "80%". (Screenshot_20231209-081555_Firefox)
Die Citroen App zeigt "85%". (Screenshot_20231209-081613_My Citron)
Der evcc Dienst schien hängen geblieben zu sein. (Screenshot_20231209-073549_Firefox)
Anbei das log (vehicle-TRACE, gezippt).

**Bitte dieses log nach dem Download löschen - Danke.    :)**

[image]
[evcc.vehicleTRACE.2023-12-07_2023-12-09.zip](https://github.com/evcc-io/evcc/files/13623446/evcc.vehicleTRACE.2023-12-07_2023-12-09.zip)
[image]
[image]
[image]
[image]

andig, 2023-12-09:
Mit den ganzen Screenshots, zumal von einem evcc ohne laufende evcc Applikation lässt sich nichts anfangen. Ich glaube Dir gerne, dass evcc einen anderen Stand als die Renault App anzeigt. Davon braucht es ein trace Log fürs Fahrzeug wie oben geschrieben.

JamyMartini, 2023-12-09:
- **Das TRACE-log war bereits angehängt!** (evcc.vehicleTRACE.2023-12-07_2023-12-09.zip, siehe oben)

Das Trace-log wurde erstellt mit den Optionen:
_log: info
levels:
  cache: error
  citroen: trace_
Das log ist vom 7.12. bis heute morgen 9.12., es sollte also den Fehler umfassen.

- Es ist **keine Renault-**, sondern Citroen-App.

- **Der evcc-Dienst lief unterbrochen**: Als die Meldung kam, "unable to restart server" habe ich mit "sudo systemctl status evcc" nachgeschaut und der Dienst wurde mit "active (running)" angezeigt. Das sollte im log ersichtlich sein.

JamyMartini, 2023-12-09:
[image]

andig, 2023-12-09:
Wie auch immer- auf diesen Screenshots läuft kein evcc:

[image]

log habe ich zwischen den riesigen Screenshots gefunden.

JamyMartini, 2023-12-09:
Ja, das sind Screenshots vom Mobiltelefon; ich bin bei den Screenshot-Aufnahmen ständig zwischen Keller (evcc-Server) und Garage (Auto) hin- und hergelaufen. Es kann daher sein, dass das Mobiltelefon kurzzeitig kein WLAN hatte und daher keine Verbindung zum evcc-Server. Die Anzeige des Ladestands hat sich durch die fehlende Serververbindung nicht geändert.

andig, 2023-12-09:
Du musst halt wenns wieder passiert mal ins Logfile schauen:

```json
{
    "createdAt": "2023-12-07T02:08:47Z",
    "type": "Electric",
    "subType": "ElectricEnergy",
    "level": 43,
    "autonomy": 72,
}
```

Passt der Wert für `level` den PSA schickt? Ist der `createdAt` Timestamp aktuell oder sind das veraltete Werte?

JamyMartini, 2023-12-09:
okay.
Ist es richtig, dass "level" der Ladezustand in Prozent ist?

Dann verstehe ich das ganze überhaupt nicht...
Wenn die gesamte log-Datei durchschaue, finde ich keinen einzigen Wert für "level":80 !
Wie kann das sein?
Die beiden Autos werden ja häufig korrekt auf 80% geladen - dann müsste doch mindestens ein solcher Eintag zu finden sein, oder?
"level":81 oder "level":79 o.ä. finde ich in der gesamten letzten Woche auch nicht.
Oder was bedeutet "level"?

JamyMartini, 2023-12-09:
So, gerade eben ist es nochmal passiert.
Anbei einen vorher-Screenshot mit 80% (falsch) beim eC4.
Und einen nachher-Screenshot mit den korrekten 72% beim eC4.
Dazwischen Neustart des evcc-Dienstes.

Noch ein Fehler: Beim ersten Screenshot hat der Berlingo (!) die 72%, er sollte eigentlich 69% haben!

Dann habe ich in die TRACE-log reingeschaut:
Vor dem Neustart werden nur die 72% vom eC4 gelesen, nicht aber die vom Berlingo.
Nach dem Neustart werden die 72% und 69% sauber gelesen.
Die VINs habe ich kontrolliert, daran kann es nicht liegen.

Sorry, da läuft irgendwas anderes ganz gewaltig schief.

Anbei Screenshots und neue TRACE-logs.
[evcc.vehicleTRACE.2023-12-07_2023-12-09.NEU.zip](https://github.com/evcc-io/evcc/files/13625197/evcc.vehicleTRACE.2023-12-07_2023-12-09.NEU.zip)

[image]
[image]

andig, 2023-12-09:
> Sorry, da läuft irgendwas anderes ganz gewaltig schief.

Was willst du damit sagen? Falsche oder veraltete Daten von PSA?

JamyMartini, 2023-12-09:
Was genau schief läuft, kann ich Dir nicht sagen, ich kenne den Quellcode ja nicht. Aber "Falsche oder veraltete Daten von PSA" sind es definitiv nicht. Ich habe Dir jedesmal die TRACE-logs geschickt und nachgeschaut, ob die Werte plausibel sind. Aus meiner Sicht sind die Werte von PSA nicht falsch oder veraltet, das sieht man sehr deutlich in den TRACE-logs nach dem Neustart. Mehr kann ich als User erst mal nicht machen.

"Falsche oder veraltete Daten von PSA" sind es aus meiner Sicht aus folgenden Gründen nicht:
=> Wenn das so wäre, würden doch die 80% oder andere falsche Werte im TRACE-log auftauchen , oder?
=> Wie kann es sein, dass die Ladezustände beider Autos vertauscht werden?
=> Wie kann es sein, dass nach einem Neustart die Werte wieder korrekt sind?
Das sind die Fakten, die ich Dir mit dem issue mitgeteilt habe.
Alles andere wäre von mir "aus der Glaskugel" orakelt.

Ich finde evcc absolut top und freue mich, wenn es sich weiterentwickelt. Dazu möchte ich auch unbedingt beitragen. Als Sponsor und indem ich es meinen Freunden und Kollegen weiterempfehle - als "Fehlermelder". Also: Wenn ich Dir noch irgendwelche Informationen oder Anhaltspunkte liefern kann, damit Du den Programmehler weiter eingrenzen kannst, helfe ich Dir gerne!

andig, 2023-12-09:
Das ist doch alles Rätselraten Kannst Du im Logfile die Socs mit den beobachteten in Einklang bringen? Welche Timestamps, welche Werte? Wo passt es und wo nicht? Wo ist das was vertauscht?

JamyMartini, 2023-12-09:
Echt jetzt? Bin ich Entwickler oder Du?
Ich habe lange als Entwickler und als Supporter gearbeitet.
Dass man als User die log-Files interpretieren soll, habe ich noch nie erlebt...
Du kannst die beiden profanen Screenshots nicht mit dem log-File zusammenbringen???
Reicht es Dir ernsthaft nicht, dass deine GUI nach dem Neustart von evcc andere Werte liefert!?!?!

Na schön, ich versuche das noch ein letztes Mal. Dieses Mal die soc-Werte von der GUI mit denen aus dem log-File mit TRACE "IN EINKLANG BRINGEN..." Ich benutze mal wieder die Farben rot (Fehler) und grün (korrekt) im jeweiligen Screenshot von der GUI und des log-Files, um das in Einklang zu bringen. Verstehe mich nicht falsch - Langsam suche ich nach der Kamera von "Verstehen Sie Spaß?", nur ist es für mich nicht mehr lustig!!

Also: Der Fehler ist, dass evcc falsche Werte anzeigt, bzw. vertauscht:

[image]

Und NACH einem Neustart sind die Werte korrekt:

[image]

Im log-File sieht man das auch:

Ausschnitt log-File VOR dem Neustart:

[image]

Ausschnitt log-File NACH dem Neustart:

[image]

andig, 2023-12-09:
> Echt jetzt? Bin ich Entwickler oder Du?

Das wäre jetzt eine gute Gelegenheit wieder sachlich zu werden- Für das Lesen von Logfiles muss man sicher kein Entwickler sein.

Dass evcc andere Werte als im Logfile anzeigt ist erwartet, da Zwischenstände interpoliert werden. Das kannst Du deaktivieren. Ausserdem kannst Du konfigurieren, wie oft aktualisiert werden soll, und ob ständig oder nur beim laden.

JamyMartini, 2023-12-10:
Das wäre dann vielleicht auch eine gute Gelegenheit, das issue wieder zu schließen und darauf zu warten, bis sich irgendwann einmal User mit mehr Geduld findet und den gleichen Programmfehler nocheinmal meldet.

Ich bin raus. Unglaublich.

premultiply, 2023-12-10:
@JamyMartini https://docs.evcc.io/docs/reference/configuration/loadpoints/#soc sollte dir wahrscheinlich an der Stelle helfen.

JamyMartini, 2023-12-10:
Danke für Deinen netten Hinweis, premultiply. Ich hatte zuvor auch vermutet, dass die soc-Einstellungen helfen könnten, war aber leider nicht so. Die Vertauschung der Ladewerte kam dennoch vor. Auto(s) abstecken und anstecken half nicht, ebenso mal kurzzeitig das Laden über die GUI anstoßen half nicht - die Werte blieben falsch. Nur der Reset half.
Danach habe ich die soc-Abschnitte halt wieder gelöscht, unter anderem auch, damit mir Citroen den Zugang nicht sperrt bzw. sich der Wagen nicht entlädt und um möglichst alles für die Fehlersuche auf Standard zu haben.
Nun ja. Ich kann damit leben, wenn das ein oder andere mal der Ladestand nicht stimmt und ich deswegen resetten muss. Muss ja sowieso täglich draufschauen, um den Zeitplan einzustellen.
Alles andere ist mir der Aufwand im Moment nicht wert.