evcc-io/evcc #11023: "integrated vehicle" sollte an anderen Loapoints nicht auswählbar sein

VolkerK62, 2023-12-06:
Angeregt durch https://github.com/evcc-io/evcc/discussions/11019 habe ich etwas ausprobiert.

Ich habe 4 Loadpoints und 4 Vehicle.
3 Vehicle sind je einem Loadpoint fest zugewiesen und die entsprechenden Charger mit dem feature: integrateddevice versehen.

Am Loadpoint der Wallbox gibt es kein zugewiesenes Vehicle. Dort werden in der Auswahlliste alle Vehicle angezeigt.
Wenn ich nun dort eines der andernorts fest zugewiesenen Fahrzeuge auswähle, wird es offensichlich am ursprünglichen Loadpoint entfernt (die Soc Anzeige verändert sich).
Da bei integrateddevice keine Vehicleliste verfügbar ist, kann ich das nur durch einen Neustart wieder richtigstellen.

Meiner Erwartung wäre, das "integrated vehicle" an anderen Loadpoints nicht auswählbar sind bzw. gar nicht angezeigt werden.

andig, 2023-12-06:
Das ist ein interessanter Punkt.

> Wenn ich nun dort eines der andernorts fest zugewiesenen Fahrzeuge auswähle, wird es offensichlich am ursprünglichen Loadpoint entfernt (die Soc Anzeige verändert sich).

Ich bin aber ein bisschen verwirrt: wie kann ein integrateddevice (das per Definition kein Fahrzeug hat) ein zugewiesenes Fahrzeug haben? Was wäre das in der Realität? Ich wäre eher versucht, diesen Konfigurationsfehler abzustellen?

/cc @naltatis

naltatis, 2023-12-06:
Ja, ich hätte auch erwartet, dass wir bei integrateddevice gar kein Fahrzeug akzeptieren. Weder in Konfiguration noch via API.

VolkerK62, 2023-12-06:
OK, dann habe eine Regelungslücke genutzt
Ich wollte halt auch den Soc haben.

Charger:
```
- name: E-Bikes
  type: template
  template: shelly
  ##Shelly Plug-S in Garage
  host: 192.168.178.83
  standbypower: 10
  integrateddevice: true
  icon: bike
```
Vehicle:
```
- name: bike
  type: custom
  title: E-Bikes
  icon: bike
  capacity: 0.625
  phases: 1
  soc:
    source: mqtt
    topic: haus/mqtt/bike_soc
    timeout: 25h
```
Loadpoint:
```
- title: E-Bikes
  charger: E-Bikes
  vehicle: bike
  mode: pv
  phases: 1
  mincurrent: 0.75
  maxcurrent: 0.75
  priority: 5
  soc:
    estimate: true
  enable:
    delay: 30s
  disable:
    delay: 5m
  guardduration: 5m
```

naltatis, 2023-12-06:
Welchen Vorteil hat denn `integrateddevice: true` dann in diesem Szenario für dich? Eine alternative/kompakte Darstellung haben wir ja bislang noch nicht bzw. nur minimal umgesetzt.

VolkerK62, 2023-12-06:
Vorteil? Eigentlich keinen. Ich könnte `integrateddevice` auch rausnehmen, dann wäre das oben beschriebene Problem ohne Neustart lösbar.
Gedanke bei der Umsetzung war/ist, das es eine zusammengehörige Einheit ist und eine Vehicle-Auswahl somit unnötig.
Die gleiche Kombi habe ich auch mit meiner Wärmepumpe konfiguriert.

VolkerK62, 2023-12-06:
Ja, das wäre der Mercedes 😄 
Aber, ich kann mit der aktuellen Umsetzung gut leben. 
Das Eingangsproblem war konstruiert, da ich normalerweise kein Fahrzeug auswählen muss.
Wollte aber trotzdem auf dieses Verhalten hinweisen.
Mein erster Gedanke war, wenn "integrateddevice=true" und "Vehicle dem Loadpoint zugewiesen" dann Fahrzeug "unsichtbar".

VolkerK62, 2023-12-06:
>Mein erster Gedanke war, wenn "integrateddevice=true" und "Vehicle dem Loadpoint zugewiesen" dann Fahrzeug "unsichtbar".

oder evtl noch einfacher. `features: ["unvisible"]` beim vehicle. Dann wird es im UI nicht angezeigt. 
@MarkusGH  wäre das auch für dich eine Lösung?

MarkusGH, 2023-12-06:
> Das Kernrequirement hier ist doch eigentlich, beliebige Funktionen zu einem bestehenden Device zu "dekorieren". Das Fahrzeug "dekoriert" hier den SoC. Schön wäre ja sowas in der Art- damit verschwindet der Bedarf des zusätzlichen Vehicles:

Ich bin mir nicht sicher, ob das meinen use case abdecken würde. Denn ich hätte schon gerne den Namen des "vehicle" im UI und in der Liste der Ladevorgänge.

naltatis, 2023-12-06:
Die aktuelle Planung ist, dass wir für Integrated Device Ladepunkte keine Ladevorgänge im klassischen Sinne speichern, da diese Ladepunkt auch so etwas wie ein Einstecken/Ausstecken nicht haben. Klassische Beispiele wäre hier Heizstab, Wärmepumpe, Steckdose mit festem Verbraucher. Diese Ladepunkte wären eher ein Kandidat für zeitabhängige Statistiken (Energie pro Tag, ...).

Das was hier ja gewünscht ist, geht eher in die Richtung eines klassichen Ladepunkts mit mehr Customizing Möglichkeiten, oder?

Siehe Konzept: https://github.com/evcc-io/evcc/discussions/7235
Der Plan ist bislang, dass die "Mini" Spalte im Konzept dem "Integrated Device" entspricht.

MarkusGH, 2023-12-06:
> > Mein erster Gedanke war, wenn "integrateddevice=true" und "Vehicle dem Loadpoint zugewiesen" dann Fahrzeug "unsichtbar".

Das würde mir nicht helfen.

> oder evtl noch einfacher. `features: ["unvisible"]` beim vehicle. Dann wird es im UI nicht angezeigt. @MarkusGH wäre das auch für dich eine Lösung?

Wenn das Vehicle damit nur aus den Auswahllisten für Fahrzeuge verschwindet wäre mir damit geholfen.
Super wäre noch wenn man dann noch die Fahrzeugauswahlliste ohne weiteren Verlust an Funktionalität pro loadpoint deaktivieren könnte.

MarkusGH, 2023-12-06:
> Die aktuelle Planung ist, dass wir für Integrated Device Ladepunkte keine Ladevorgänge im klassischen Sinne speichern, da diese Ladepunkt auch so etwas wie ein Einstecken/Ausstecken nicht haben. [...]

Mindestens bei der Wärmepumpe ist es nicht unbedingt schlecht, die einzelnen Vorgänge zu sehen.

> Das was hier ja gewünscht ist, geht eher in die Richtung eines klassichen Ladepunkts mit mehr Customizing Möglichkeiten, oder?

Das ist dem Grunde nach das, was ich in https://github.com/evcc-io/evcc/discussions/11019 angefragt habe.
Was VolkerK62 hier anfragt ist etwas spezifischer (wäre für meinen use case auch vom Ansatz her nicht ganz passend).

VolkerK62, 2023-12-07:
Ist für mich richtig.

andig, 2023-12-10:
@MarkusGH abgefahrenes Setup. Du verwendest die Fahrzeuge aber "nur" fürs Logging, nicht um etwas zu steuern, richtig?

MarkusGH, 2023-12-10:
@andig: Genau

andig, 2023-12-22:
Für @VolkerK62's case könnte man die Fahrzeuge vmtl. unsichtbar machen. einfach ein `invisible` Feature und sie damit aus dem API komplett entfernen. Für @MarkusGH fällt mir keine gute Lösung ein, damit müssen wir erstmal leben...

MarkusGH, 2023-12-22:
Wenn "invisible" das Fahrzeug nur in den Auswahllisten versteckt wäre mir zumindest teilweise geholfen.
Aber warum nicht flexibler:
Definition einer Fahrzeugklasse (einfach ein String) und am Loadpoint eine Liste von akzeptierten Fahrzeugklassen.
Damit wäre auch gleich der Fall unterschiedlicher Steckertypen abgehandelt.

MarkusGH, 2023-12-22:
Das Gastfahrzeug oder Fahrzeuge ohne Klasse kriegen dabei die Fahrzeugklasse "default" oder "", Loadpoint ohne Filter für die Klassen akzeptiert alle Klassen.

andig, 2023-12-22:
> Aber warum nicht flexibler:

Weil wir einfach kein klares Bild von dem Usecase haben und keine Einzellösung bauen wollen.

MarkusGH, 2023-12-22:
> > Aber warum nicht flexibler:
> 
> Weil wir einfach kein klares Bild von dem Usecase haben und keine Einzellösung bauen wollen.

Das hat für mich keine hohe Priorität, ist also völlig OK, aber dennoch:

Das wäre doch gerade eine generalisierte Lösung und keine Einzellösung. Der Fall unterschiedlicher Steckertypen kommt in der Praxis sicher vor: 
Beispiel: Jemand hat eine Ladestation für E-Scooter an einem Shelly und eine normale Wallbox für Elektroautos. Die E-Scooter können niemals an die Wallbox und die EVs können niemals an die Ladestation für die Scooter. Macht keinen Sinn, die wechselweise auswählen zu können.

andig, 2023-12-22:
Wenn Du es so verkaufst, finde ich die Idee Klasse! Jetzt müssen wir nur rausfinden, wie sich das elegant lösen liesse so dass die Konfiguration nach Möglichkeit orthogonal bleibt.

Was wäre eine Kompatibilitätsregel für einen "Steckertyp": es darf alles verbunden werden, was entweder keinen oder den passenden Typ hat? Also wenn nur einer der Player typisiert ist dann passt es immer?

MarkusGH, 2023-12-22:
> Wenn Du es so verkaufst, finde ich die Idee Klasse! Jetzt müssen wir nur rausfinden, wie sich das elegant lösen liesse so dass die Konfiguration nach Möglichkeit orthogonal bleibt.

Freut mich dass ich durchdringe.
Es müsste aus meiner Sicht gar nicht orthogonal sein.

> Was wäre eine Kompatibilitätsregel für einen "Steckertyp": es darf alles verbunden werden, was entweder keinen oder den passenden Typ hat? Also wenn nur einer der Player typisiert ist dann passt es immer?
> 
> Die Regel erscheint sinnvoll, macht es aber schwierig weil aktuell kaum einer der Charger die Zusatzfunktionen für icon/feature enthält (...und die heute auch nicht vermisst werden). Da müssten wir uns was Schlaues überlegen.

Ich würde es einfach halten.

Nennen wir das Feature doch "group".

1) das vehicle bekommt eine  "group", Gastfahrzeug hat group ""
2) der Loadpoint bekommt eine Liste "allowedVehicleGroups", wenn "allowedVehicleGroups" nicht gesetzt ist sind alle "groups" erlaubt.

Wenn ein vehicle nicht in den "allowedVehicleGroups" auftaucht kann es im UI nicht ausgewählt werden.

andig, 2023-12-22:
Wieso der Loadpoint? Wenn hängt die group doch am charger?

MarkusGH, 2023-12-22:
> Wieso der Loadpoint? Wenn hängt die group doch am charger?

Ist am Ende egal - loadpoint und charger haben doch sowieso eine 1:1 Zuordnung - oder?
Loadpoint kommt mir plausibler vor - es kann ja auch "nicht technische" Gründe für eine Gruppierung geben...

andig, 2023-12-30:
@naltatis as discussed- please review

andig, 2024-01-10:
Bisher haben wir kein Konzept gefunden, dass sich auch elegant mit einem künftigen Config UI verbinden lassen würde.

MarkusGH, 2024-01-11:
> Bisher haben wir kein Konzept gefunden, dass sich auch elegant mit einem künftigen Config UI verbinden lassen würde.

Verstehe ich nicht - wenn der von mir vorgeschlagene "group" Ansatz implementiert würde, dann bekäme doch nur der Loadpoint optional eine zusätzliche Liste von Strings und das Verhicle optional einen String. Wenn am Loadpoint kein Filter gesetzt ist sind alle Fahrzeuggruppen erlaubt, gibt es einen Filter dann sind nur die Fahrzeuge erlaubt die einer der genannten Gruppen angehören. Das Gastfahrzeug hat eine vordefinierte Gruppe. Einfache Regel, völlig generisch, vielseitig verwendbar.

andig, 2024-01-13:
> dann bekäme doch nur der Loadpoint optional eine zusätzliche Liste von Strings und das Verhicle optional einen String.

Es geht ums Config UI. Das wäre nicht sehr intuitiv.

MarkusGH, 2024-01-13:
Ich bin nicht der Meinung, dass intuitive Bedienung in einem künftigen Config UI ein starkes Kriterium gegen ein solches "Expertenfeature" ist.

andig, 2024-01-13:
Meinungen helfen uns hier nicht. Vorschläge, wie sich das elegant lösen liesse würden das Thema schneller voran bringen.

MarkusGH, 2024-01-13:
> Meinungen helfen uns hier nicht. 

OK - verstanden. Damit macht es für mich aber auch keinen Sinn mehr, mich einzubringen.

andig, 2024-01-13:
Es gibt keinen Grund beleidigt zu sein. Wir sehen da ein Thema und das verschwindet auch dadurch nicht, dass Du es anders einschätzt.

andig, 2024-02-03:
Scheint, als hätten wir keine gute Idee, wie sich das elegant mit dem Config UI verbinden ließe- closing.

closed 2024-02-03 by andig

andig, 2024-06-24:
Siehe auch https://github.com/evcc-io/evcc/issues/14508. Damit könnten (ausser switchsockets) für custom Charger auf das zusätzliche `vehicle` (soc ist bereits erledigt) oder `meter` verzichtet werden.