evcc-io/evcc #11032: Potential Incorrect Charging Cost Calculation

changyang1230, 2023-12-07:
### Describe the bug

Hi, please see the attached photo below. 

[image]

I am wondering if the calculated "charger cost" of 2.7c is an over-optimistic calculation? 

The way I see it:
- the solar cost is 2.25c (my feed in tariff), and the current grid energy tariff is 8.2c. 
- my current solar production is 8.3kW, of which the preferential use are consumption (1.7kW) and home battery charging (3.0kW). 
- I am also importing 3.6kW from the grid currently. 
- therefore, the car charging is really using 3.6kW from the grid and 3.6kW from the solar. 
- If you weight the charging cost appropriately, my charger cost should have been (3.6*2.25+3.6*8.2)/(3.6+3.6) = 5.2c  rather than the over-optimistic 2.7c shown in this screen. 

I am guessing that the reason EVCC has produced 2.7c estimate is because it puts the car charging as the preferential user of the solar energy instead of my home battery, therefore of the 8.3kW solar production, 1.7kW went to the house, and 6.7kW went to the car. Therefore the car charging cost was calculated as (6.7*2.25+0.5*8.2)/(6.7+0.5) = 2.7c. 

I wonder if this is an incorrect way of calculating the charging cost. If I wasn't charging the car, these solar excess would have gone to the battery first, there's no way I would have imported grid energy to charge the battery as this calculation is proposing. 

### Steps to reproduce

See above. 

### Configuration details

```yaml
not relevant
```

### Log details

```text
not relevant
```

### What type of operating system are you running?

Docker container

### Version

v0.122.1

changyang1230, 2023-12-07:
Just had a look with google translate (don't speak german) - it's a different calculation issue. 

I thought it's the natural order that solar should be considered to be used household consumption > home battery > car battery; as the first two are what get the solar when the car is not plugged in. 

As it stands, the car charger expense is presumed to have used up the solar excess first (hence very cheap) before the home battery which is not really logical.

StefanSchoof, 2023-12-07:
I think in that issue there is a link to the docs. Here the englisch version:

https://docs.evcc.io/en/docs/guides/setup/#savings-calculation

> evcc assumes that the house battery is primarily used to store excess, self-produced solar power. If the house battery also discharges to satisfy other loads, or charges from grid supply, this assumption isn't always correct.

Looking in your screenshot I see that the battery is loading from the grid.

changyang1230, 2023-12-07:
By the way an additional comment here. 

The battery is a hybrid battery where it has a maximum inverter of 5kW. Therefore, even though the solar production of 8.2kW, it can only convert 5 of it to AC and the rest is supplied to charge the battery. 

Therefore there is this illusion of "i am charging the battery with grid energy" which is not really the case.

naltatis, 2023-12-07:
> The battery is a hybrid battery where it has a maximum inverter of 5kW. Therefore, even though the solar production of 8.2kW, it can only convert 5 of it to AC and the rest is supplied to charge the [...]

But this seems to be the issue here. I would have assumed that the charger would reduce power to not introduce grid import. Or did you set a min of 7.2kW?

\\cc @premultiply

changyang1230, 2023-12-07:
That bit is intentional, my solar excess is not sufficient so I chose to fast charge with the cheap day time grid energy.

changyang1230, 2023-12-08:
Any more thought thought from @StefanSchoof or @andig  now that I have explained my rationale in more detail?

RTTTC, 2023-12-12:
I do agree with @changyang1230  - for savings calculation car charging energy should be considered "last in line" to get solar energy, no matter what the reasons or settings are.

naltatis, 2023-12-13:
Depending on your configured log level you can see the individual values in `stdout`. Do build aggregates per charging session to calculate the price per charging session but these all all in memory. We only persist the result to db (sessions table in evcc.db) when charging is finished.

closed 2024-01-09 by naltatis