evcc-io/evcc #11045: Ladeplaner Zielzeit wird als 01.01.0001 errechnet

GrimmiMeloni, 2023-12-08:
### Describe the bug

Ich sehe im Log für den Planner eine Zielzeit die schon "ein wenig" länger her ist....

`[lp-2  ] DEBUG 2023/12/08 07:48:57 plan: charge 7h2m55s starting at 2023-12-08 23:57:05 +0100 CET until 0001-01-01 00:53:28 +0053 LMT (power: 11040W, avg cost: 196.000)`

### Steps to reproduce

1. evcc mit `cmd/demo.yaml` Konfig starten
2. Ein Ladeplan für das weiße Model 3 auswählen
3. Ins Log schauen und obige Zeile sehen

### Configuration details

```yaml
generated using stock demo.yaml from the repo
```

### Log details

```text
[lp-2  ] DEBUG 2023/12/08 07:48:57 plan: charge 7h2m55s starting at 2023-12-08 23:57:05 +0100 CET until 0001-01-01 00:53:28 +0053 LMT (power: 11040W, avg cost: 196.000)
```

### What type of operating system are you running?

Linux

### Version

0.122.1 (be6915cb) (built from master)

GrimmiMeloni, 2023-12-08:
Ich bin nicht sicher ob es sich hier nur um ein kosmetisches Problem handelt. Meine lokale Live Installation zeigt das gleiche Verhalten und hat zusätzlich auch letzte Nacht nicht zum errechneten Zeitpunkt die Ladung gestartet. Es _könnte_ also sein, daß hier ein generelles Problem vorliegt das die Ladeplanung unbrauchbar macht.

GrimmiMeloni, 2023-12-08:
Sieht irgendwie danach aus, als würde der Ladeplaner jetzt auf dem Vehicle arbeiten (via `vehicle.SetPlanSoc()`. Die Logmeldung kommt aber aus dem Loadpoint. Auffällig auch: Die Startzeit der Logmeldung kommt aus dem Plan, die Endzeit aber ist ein Member auf dem Loadpoint. Irgendwas passt da nicht mehr zusammen. Ich vermute die Endzeit auf dem Loadpoint wird nicht mehr gesetzt.

andig, 2023-12-08:
`lp.planTime` bildet zusammen mit `lp.planEnergy` nur noch das Energielimit ab (für Fahrzeuge ohne Soc). Hier muss zusätzlich auch `vehicle.planTime` betrachtet werden. Fix kommt.

closed 2023-12-08 by andig