evcc-io/evcc #11046: Kostal Plenticore: batteryDischargeControl battery is not charging

deadrabbit87, 2023-12-08:
### Describe the bug

Actually the Plenticore stops discharging the battery when `batteryDischargeConrol` is set to `true`. 

But it won't charge the battery again, if solar power is produced. 

In my point of view is this issue caused by writing registers from the external battery manager. The inverter doesn't fall back to it's internal regulation, as long registers are written. 

Therefore evcc should not write any registers of the external battery manager except `batteryDischargeControl` is `hold`

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.122.1

andig, 2023-12-08:
`hold` sollte weiterhin laden zulassen. Das ist nicht der Fall?

/cc @premultiply

andig, 2023-12-08:
> Ich verstehe es so dass der normale Betrieb hier nicht funktioniert, hold aber im Prinzip schon, richtig?

Auch `hold` funktioniert anscheinend nur halb. Ich würde erwarten, dass `hold` weiter laden, aber nicht mehr entladen kann. Das ist aber ggf. verschmerzbar.

Was mich aber wirklich wundert ist, dass das bei Kostal verwendete Register, also die

    1042 # limit soc

dazu führt, dass nicht mehr geladen wird. Das kann ich aus der Doku nicht entnehmen?

In Konsequenz würde das bedeuten, dass wir hier die "einfache" Steuerung über soc nicht verwenden können und uns ein komplexeres Modell überlegen müssten. Fürs Erste würde das dann revert bedeuten.

deadrabbit87, 2023-12-08:
In der Mobus Spec steht da gar nichts, in der Anleitung steht 

> Das interne Energiemanagement bleibt aktiv, wird aber
> durch die externen Vorgaben bzgl. Lade- und Entladeleistung übersteuert.
> Folgende Kommandos sind möglich:
> Ladung/Entladung der Batterie über Stromvorgabe in
> Prozent oder Watt
> Ladung/Entladung der Batterie über Leistungsvorgabe in Prozent oder Watt
> Bereichsvorgabe eines min./max. SOC in Prozent

Fraglich ist, was da mit internem Energiemanagament jetzt genau gemeint ist... Ich muss das alles nochmal testen wenn Sonne da ist. Im Moment ist das ein wenig schwierig. 

Stand jetzt aber, wenn `batteryDischargeConrol` gesetzt ist, dann wird nicht mehr geladen.

premultiply, 2023-12-08:
> Ladung/Entladung der Batterie über Leistungsvorgabe in Prozent

Das dürfte hier wohl das unkomplizierteste und minimalinvasivste Verfahren sein.

deadrabbit87, 2023-12-08:
Für meinen Use Case wäre es aktuell am Besten, ihr schreibt `1042` nur solange `hold` ist. 

Dann würde der WR nach max. 60 s wieder auf die interne Regelung zurückfallen und auch wieder laden.

andig, 2023-12-08:
Hier gehts durcheinander. Bitte nicht immer alles in einen Topf werfen. Kostal hat aktuell weder was mit Sunspec noch (aktuell) mit Leistungsvorgabe oder -steuern zu tun. Wir haben *einzig* den MinSoc angepasst. Der Dokuauszug passt nicht zur Umsetzung.

deadrabbit87, 2023-12-08:
Versteh ich nicht. Ich hab nichts von SunSpec gesagt und die Register beziehen sich auch nicht auf SunSpec. 

Hier gehts nur darum, dass sobald Modbus Register der externen Batteriesteuerung geschrieben werden, der WR hier dann auch nicht mehr PV-Überschuss in die Batterie lädt, weil er anscheinend die interne Regelung komplett deaktiviert und nur noch auf Befehle von außen wartet.

StevieC121176, 2023-12-08:
Bei mir führt `batteryDischargeConrol=true` zum völligen entladen der Batterie, auch wenn kein Auto lädt, quasi nur durch den Hausverbrauch obwohl ich am Kostal WR 10% MinSoc eingestellt habe. Sobald er dann auf 0% startet er eine Notladung.

deadrabbit87, 2023-12-08:
Du hast aber dann vermutlich keinen `minSoc` bei der Batterie gesetzt? Da drauf bezieht sich evcc dann.

StevieC121176, 2023-12-08:
doch ist gesetzt:

```
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.118  
  port: 1502  
  usage: battery  
  modbus: tcpip
  capacity: 12.8 
  minsoc: 10  
  maxsoc: 100  
  watchdog: 60s 
  name: battery2
```

deadrabbit87, 2023-12-08:
Ja, für mich auch. Ich muss das auch nochmal genauer beobachten, ich will auch nicht ausschließen, dass da das Kostal Batteriemanagament reinpfuscht. 

Das macht manchmal nämlich auch so merkwürdige Dinge, wie die Batterie nicht laden, wenn länger kein Ertrag da war.

deadrabbit87, 2023-12-09:
@StevieC121176 Kannst du mal die Ausgabe von ` sudo evcc -l trace meter battery1` posten? `battery1` muss die unter `meters` definierte Batterie sein.

StevieC121176, 2023-12-09:
> @StevieC121176 Kannst du mal die Ausgabe von ` sudo evcc -l trace meter battery1` posten? `battery1` muss die unter `meters` definierte Batterie sein.

```
pi@raspberrypi:~ $ sudo evcc -l trace meter battery2
[main  ] INFO 2023/12/09 16:18:50 evcc 0.122.1 (be6915cb)
[main  ] INFO 2023/12/09 16:18:50 using config file: /home/pi/evcc.yaml
[db    ] INFO 2023/12/09 16:18:50 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2023/12/09 16:18:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/12/09 16:18:50 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/12/09 16:18:50 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/12/09 16:18:50 SELECT * FROM `settings` 7 <nil>
[mqtt  ] INFO 2023/12/09 16:18:50 connecting evcc-1551713391 at tcp://192.168.178.80:1883
[mqtt  ] DEBUG 2023/12/09 16:18:50 tcp://192.168.178.80:1883 connected
[db    ] TRACE 2023/12/09 16:18:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/12/09 16:18:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/12/09 16:18:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/12/09 16:18:50 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/12/09 16:18:50 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/12/09 16:18:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/12/09 16:18:50 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/12/09 16:18:50 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/12/09 16:18:50 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2023/12/09 16:18:50 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/12/09 16:18:50 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2023/12/09 16:18:50 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[modbus] TRACE 2023/12/09 16:18:50 modbus: send 00 01 00 00 00 06 47 03 9c 40 00 02
[modbus] TRACE 2023/12/09 16:18:50 modbus: recv 00 01 00 00 00 07 47 03 04 53 75 6e 53
[modbus] TRACE 2023/12/09 16:18:50 modbus: send 00 02 00 00 00 06 47 03 9c 42 00 02
[modbus] TRACE 2023/12/09 16:18:50 modbus: recv 00 02 00 00 00 07 47 03 04 00 01 00 42
[modbus] TRACE 2023/12/09 16:18:50 modbus: send 00 03 00 00 00 06 47 03 9c 86 00 02
[modbus] TRACE 2023/12/09 16:18:50 modbus: recv 00 03 00 00 00 07 47 03 04 00 67 00 32
[modbus] TRACE 2023/12/09 16:18:50 modbus: send 00 04 00 00 00 06 47 03 9c ba 00 02
[modbus] TRACE 2023/12/09 16:18:50 modbus: recv 00 04 00 00 00 07 47 03 04 00 71 00 3c
[modbus] TRACE 2023/12/09 16:18:50 modbus: send 00 05 00 00 00 06 47 03 9c f8 00 02
[modbus] TRACE 2023/12/09 16:18:50 modbus: recv 00 05 00 00 00 07 47 03 04 00 78 00 1a
[modbus] TRACE 2023/12/09 16:18:50 modbus: send 00 06 00 00 00 06 47 03 9d 14 00 02
[modbus] TRACE 2023/12/09 16:18:50 modbus: recv 00 06 00 00 00 07 47 03 04 00 7b 00 18
[modbus] TRACE 2023/12/09 16:18:50 modbus: send 00 07 00 00 00 06 47 03 9d 2e 00 02
[modbus] TRACE 2023/12/09 16:18:50 modbus: recv 00 07 00 00 00 07 47 03 04 00 a0 00 44
[modbus] TRACE 2023/12/09 16:18:50 modbus: send 00 08 00 00 00 06 47 03 9d 74 00 02
[modbus] TRACE 2023/12/09 16:18:50 modbus: recv 00 08 00 00 00 07 47 03 04 03 22 00 3e
[modbus] TRACE 2023/12/09 16:18:50 modbus: send 00 09 00 00 00 06 47 03 9d b4 00 02
[modbus] TRACE 2023/12/09 16:18:50 modbus: recv 00 09 00 00 00 07 47 03 04 ff ff 00 00
[modbus] TRACE 2023/12/09 16:18:50 modbus: send 00 0a 00 00 00 06 47 03 9c 44 00 42
[modbus] TRACE 2023/12/09 16:18:50 modbus: recv 00 0a 00 00 00 87 47 03 84 4b 4f 53 54 41 4c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 50 4c 45 4e 54 49 43 4f 52 45 20 70 6c 75 73 38 2e 35 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 30 31 2e 32 39 2e 31 32 30 33 38 00 00 00 00 00 39 32 30 39 32 56 46 36 30 30 30 30 39 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 47 5a 5a
[modbus] TRACE 2023/12/09 16:18:50 modbus: send 00 0b 00 00 00 06 47 03 9c 40 00 02
[modbus] TRACE 2023/12/09 16:18:50 modbus: recv 00 0b 00 00 00 07 47 03 04 53 75 6e 53
[... cut ...]
[modbus] TRACE 2023/12/09 16:18:51 modbus: recv 00 16 00 00 00 7f 47 03 7c 00 18 39 08 3a 66 3a 66 00 00 00 00 00 00 00 00 00 00 00 46 00 00 00 00 00 00 00 64 00 00 00 00 00 00 00 00 00 01 00 04 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 14 9d 16 d0 00 00 00 21 00 00 00 00 00 20 00 00 00 00 00 00 00 67 0a 00 0a 00 02 1f 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff ff ff fe ff fe 00 00
[modbus] TRACE 2023/12/09 16:18:51 802:0:SoC: 70
Power:    543W
Soc:      70%
Capacity: 12.8kWh
```

deadrabbit87, 2023-12-09:
Sorry, da hat was gefehlt: ` sudo evcc -c /etc/evcc.yaml -l trace meter battery1 -b hold`

Da sollte dann das Register 1042 beschrieben werden: 
```
[modbus] TRACE 2023/12/09 16:36:45 modbus: send 00 16 00 00 00 0b 47 10 04 12 00 02 04 42 50 00 00
[modbus] TRACE 2023/12/09 16:36:45 modbus: recv 00 16 00 00 00 06 47 10 04 12 00 02
```

Wenn das bei dir der Fall ist, hast du big-endian (Sunspec) im WR eingestellt? 
[image]

StevieC121176, 2023-12-09:
Ich habe es gerade nochmal überprüft, ja, big Endian Sunspec ist eingestellt.

hier der Log:

```
pi@raspberrypi:~ $ sudo evcc -c /etc/evcc.yaml -l trace meter battery2 -b hold
[main  ] INFO 2023/12/09 18:39:27 evcc 0.122.1 (be6915cb)
[main  ] INFO 2023/12/09 18:39:27 using config file: /etc/evcc.yaml
[db    ] INFO 2023/12/09 18:39:27 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2023/12/09 18:39:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/12/09 18:39:27 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/12/09 18:39:27 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/12/09 18:39:27 SELECT * FROM `settings` 7 <nil>
[mqtt  ] INFO 2023/12/09 18:39:27 connecting evcc-457943864 at tcp://192.168.178.80:1883
[mqtt  ] DEBUG 2023/12/09 18:39:27 tcp://192.168.178.80:1883 connected
[db    ] TRACE 2023/12/09 18:39:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/12/09 18:39:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/12/09 18:39:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/12/09 18:39:27 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/12/09 18:39:27 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/12/09 18:39:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/12/09 18:39:27 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/12/09 18:39:27 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/12/09 18:39:27 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2023/12/09 18:39:27 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/12/09 18:39:27 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2023/12/09 18:39:27 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[modbus] TRACE 2023/12/09 18:39:27 modbus: send 00 01 00 00 00 06 47 03 9c 40 00 02
[modbus] TRACE 2023/12/09 18:39:27 modbus: recv 00 01 00 00 00 07 47 03 04 53 75 6e 53
[modbus] TRACE 2023/12/09 18:39:27 modbus: send 00 02 00 00 00 06 47 03 9c 42 00 02
[modbus] TRACE 2023/12/09 18:39:28 modbus: recv 00 02 00 00 00 07 47 03 04 00 01 00 42
[modbus] TRACE 2023/12/09 18:39:28 modbus: send 00 03 00 00 00 06 47 03 9c 86 00 02
[modbus] TRACE 2023/12/09 18:39:28 modbus: recv 00 03 00 00 00 07 47 03 04 00 67 00 32
[modbus] TRACE 2023/12/09 18:39:28 modbus: send 00 04 00 00 00 06 47 03 9c ba 00 02
[modbus] TRACE 2023/12/09 18:39:28 modbus: recv 00 04 00 00 00 07 47 03 04 00 71 00 3c
[modbus] TRACE 2023/12/09 18:39:28 modbus: send 00 05 00 00 00 06 47 03 9c f8 00 02
[modbus] TRACE 2023/12/09 18:39:28 modbus: recv 00 05 00 00 00 07 47 03 04 00 78 00 1a
[modbus] TRACE 2023/12/09 18:39:28 modbus: send 00 06 00 00 00 06 47 03 9d 14 00 02
[modbus] TRACE 2023/12/09 18:39:28 modbus: recv 00 06 00 00 00 07 47 03 04 00 7b 00 18
[modbus] TRACE 2023/12/09 18:39:28 modbus: send 00 07 00 00 00 06 47 03 9d 2e 00 02
[modbus] TRACE 2023/12/09 18:39:28 modbus: recv 00 07 00 00 00 07 47 03 04 00 a0 00 44
[modbus] TRACE 2023/12/09 18:39:28 modbus: send 00 08 00 00 00 06 47 03 9d 74 00 02
[modbus] TRACE 2023/12/09 18:39:28 modbus: recv 00 08 00 00 00 07 47 03 04 03 22 00 3e
[modbus] TRACE 2023/12/09 18:39:28 modbus: send 00 09 00 00 00 06 47 03 9d b4 00 02
[modbus] TRACE 2023/12/09 18:39:28 modbus: recv 00 09 00 00 00 07 47 03 04 ff ff 00 00
[modbus] TRACE 2023/12/09 18:39:28 modbus: send 00 0a 00 00 00 06 47 03 9c 44 00 42
[modbus] TRACE 2023/12/09 18:39:28 modbus: recv 00 0a 00 00 00 87 47 03 84 4b 4f 53 54 41 4c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 50 4c 45 4e 54 49 43 4f 52 45 20 70 6c 75 73 38 2e 35 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 30 31 2e 32 39 2e 31 32 30 33 38 00 00 00 00 00 39 32 30 39 32 56 46 36 30 30 30 30 39 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 47 5a 5a
[modbus] TRACE 2023/12/09 18:39:28 modbus: send 00 0b 00 00 00 06 47 03 9c 40 00 02
[modbus] TRACE 2023/12/09 18:39:28 modbus: recv 00 0b 00 00 00 07 47 03 04 53 75 6e 53
[... cut ...]
[modbus] TRACE 2023/12/09 18:39:29 modbus: recv 00 18 00 00 00 7f 47 03 7c 00 18 39 08 3a 66 3a 66 00 00 00 00 00 00 00 00 00 00 00 3b 00 00 00 00 00 00 00 64 00 00 00 00 00 00 00 00 00 01 00 04 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 14 90 16 d0 00 00 00 21 00 00 00 00 00 20 00 00 00 00 00 00 00 66 0a 00 0a 00 02 1a 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff ff ff fe ff fe 00 00
[modbus] TRACE 2023/12/09 18:39:29 802:0:SoC: 59
Power:    538W
Soc:      59%
Capacity: 12.8kWh
```

deadrabbit87, 2023-12-10:
Ich kann bei dir jetzt keinen Fehler sehen. 
```
[modbus] TRACE 2023/12/09 18:39:29 modbus: send 00 16 00 00 00 0b 47 10 04 12 00 02 04 42 6c 00 00
[modbus] TRACE 2023/12/09 18:39:29 modbus: recv 00 16 00 00 00 06 47 10 04 12 00 02
```
Das passt auch, es wird der minSoC von 59% gesetzt. 

Du könntest es noch mit `sudo evcc -c /etc/evcc.yaml -l trace meter battery2 -b normal` versuchen, dann sollte bei dir hier `41 20` anstatt `42 6c` geschrieben werden.

Bei mir geht es komischerweise jetzt seit zwei Tagen. Ich hab zwischendrin auch mal Schnellgeladen und da wurde auch die Entladung gesperrt. 

@andig Bitte in Diskussion umwandeln, ich glaube das war wieder ein Schnellschuss und ist gar kein Issue... Sorry dafür...