evcc-io/evcc #11055: plugin calc (Modbus und OCPP)

manuCalw, 2023-12-08:
Hallo
Ich versuche über den Calc Plugin die Leistung meines SMA Home MAnagers zu addieren mit der Leistung meiner Wallbox (ABB über OCPP) . Hintergrund ist dass mein SMA Home Manager nicht die Leistung der Wallbox sieht und da muss ich ein Custom Grid machen.
Ich habe aus der Diskussion verstanden, dass man alles selber zu Fuss machen kann, aber ich verzweifle langsam..
So sieht mein Auszug aus der Konf Datei:
(Grid 2 ist der SMA Home Manager als Referenz. Über Calc versuche ich zunächst nur die Leistung des SMA HomeManagers über Modus herauszulesen bevor ich mit OCPP anfange :-)

meters:
  - name: grid
    type: custom
    power:
      type: calc
      add:
       - source: modbus
         uri: 192.168.178.34
         model: sunspec
         value: 30867:3:DCW # mppt 3 charge         
         scale: -1

  - name: grid2
    type: template
    template: sma-home-manager
    usage: grid
    host: 192.168.178.34 

Wenn ich Grid2 lese (Ohne calc)funktioniert einwandsfrei 👍
[main  ] INFO 2023/12/08 18:30:47 evcc 0.122.0
[main  ] INFO 2023/12/08 18:30:47 using config file: D:\evcc\evcc_0.122.0_windows-amd64\evcc.yaml
[db    ] INFO 2023/12/08 18:30:47 using sqlite database: C:\Users\manub\.evcc\evcc.db
Power:          290W
Energy:         184.6kWh
Current L1..L3: 0.64A 0.637A 0.613A
Voltage L1..L3: 237V 235V 236V
Power L1..L3:   88.2W 61.3W 141W

Wenn ich über Calc lese bekomme ich folgende Fehlermeldung 
D:\evcc\evcc_0.122.0_windows-amd64>evcc meter grid
[main  ] INFO 2023/12/08 18:32:42 evcc 0.122.0
[main  ] INFO 2023/12/08 18:32:42 using config file: D:\evcc\evcc_0.122.0_windows-amd64\evcc.yaml
[db    ] INFO 2023/12/08 18:32:42 using sqlite database: C:\Users\manub\.evcc\evcc.db
[main  ] FATAL 2023/12/08 18:32:42 cannot create meter 'grid': cannot create meter 'custom': power: invalid plugin source:

Ich wäre dankbar über einen Tipp...
Vielen Dank
Emmanuel

manuCalw, 2023-12-08:
Danke Volker. 
Jetzt bin ich ein Schritt weiter. Bekomme aber folgende Meldung:
D:\evcc\evcc_0.122.0_windows-amd64>evcc meter grid
[main  ] INFO 2023/12/08 20:24:54 evcc 0.122.0
[main  ] INFO 2023/12/08 20:24:54 using config file: D:\evcc\evcc_0.122.0_windows-amd64\evcc.yaml
[db    ] INFO 2023/12/08 20:24:54 using sqlite database: C:\Users\manub\.evcc\evcc.db
[main  ] FATAL 2023/12/08 20:24:56 cannot create meter 'grid': cannot create meter 'custom': power: add[0]: dial tcp 192.168.178.34:502: connectex: No connection could be made because the target machine actively refused it.

Kein Fehler wenn ich direkt über das sma-home-manager template..

WErde mich mit MQTT auseinandersetzen....

VolkerK62, 2023-12-09:
Ich vermute, `calc` verlangt mindestens 2 Werte. Ich habe einfach den Wert 0 mittels `const` hinzugefügt. Das kannst du dann später für die ChargePower nutzen.
```
- name: grid
  type: custom
  power:
    source: calc
    add:
    - source: modbus
      uri: 192.168.178.34
      model: sunspec
      value: 30867:3:DCW # mppt 3 charge
      scale: -1
    - source: const
      value: 0
```
Was die SMA Abfrage angeht, kann ich dir leider nicht weiterhelfen.

By the way, das ist eher eine Diskussion, als ein Issue.