evcc-io/evcc #11060: LP laden alle mit voller Leistung trotze niedrigen Überschuss

xerion3800, 2023-12-09:
### Describe the bug

Bei mir kommt gerade die Sonne raus und auf einmal fängt alles mit voller Leistung an zu laden , Obwohl nicht genüg Überschuss da ist

### Steps to reproduce

1. PV Leistung vorhanden
2. Ladepunkte auf PV Modus
3. Speicher etwas geladen

[image]
residualPower ist auf 300 eingestellt.

### Configuration details

```yaml
network:
  schema: http
  host: 192.168.178.191
  port: 7070

log: debug

levels:
  # go: trace
  site: debug
  lp-1: debug
  lp-2: debug
  # mqtt: trace
  # modbus: trace

  
interval: 10s # control cycle interval

plant: ############

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: ###########

meters:
  - name: grid
    type: template
    template: sungrow-hybrid
    usage: grid     
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.141 # Hostname
    port: 502 # Port 
  - name: sungrow
    type: template
    template: sungrow-hybrid
    usage: pv     
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.141 # Hostname
    port: 502 # Port 

  - name: plenticore
    type: template
    template: kostal-plenticore
    usage: pv      
    
    # Modbus TCP
    modbus: tcpip
    id: 71
    host: 192.168.178.199 # Hostname
[... 385 lines cut ...]
[lp-2  ] DEBUG 2023/12/09 11:14:01 vehicle range: 251km
[lp-2  ] DEBUG 2023/12/09 11:14:01 limitSoc reached: 62.0% > 60%
[site  ] DEBUG 2023/12/09 11:14:10 ----
[lp-1  ] DEBUG 2023/12/09 11:14:10 charge power: 3680W
[lp-2  ] DEBUG 2023/12/09 11:14:10 charge power: 0W
[lp-3  ] DEBUG 2023/12/09 11:14:10 charge power: 0W
[lp-4  ] DEBUG 2023/12/09 11:14:10 charge power: 14W
[site  ] DEBUG 2023/12/09 11:14:10 pv power: 2561W
[site  ] DEBUG 2023/12/09 11:14:10 battery soc: 33%
[site  ] DEBUG 2023/12/09 11:14:10 battery power: 1303W
[site  ] DEBUG 2023/12/09 11:14:10 grid power: 211W
[site  ] DEBUG 2023/12/09 11:14:10 site power: 1814W
[lp-3  ] DEBUG 2023/12/09 11:14:10 charge voltages: [0 0 0]V
[lp-3  ] DEBUG 2023/12/09 11:14:10 charge currents: [0 0 0]A
[lp-3  ] DEBUG 2023/12/09 11:14:10 charge total import: 25.595kWh
[lp-3  ] DEBUG 2023/12/09 11:14:10 charger status: B
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.122.1 (74525f2d)

xerion3800, 2023-12-09:
Vollständiges Log:
[_evcc-io_logs.txt](https://github.com/evcc-io/evcc/files/13623820/_evcc-io_logs.txt)

andig, 2023-12-09:
Das sieht komisch aus:

    lp Parkplatz at prio 5 gets additional 0W from Vaillant aroTHERM plus at prio 1, 3720W from Parkplatz at prio 5, 3480W from Carport at prio 2, 0W from Tasmota Steckdose at prio 0, total 7200W

Es scheint als würde "Parkplatz" hier Ladeleistung von sich selbst bekommen und damit wunderbare Vermehrung der Energiemenge machen?

andig, 2023-12-09:
Da fehlte eine Anpassung. Closed in https://github.com/evcc-io/evcc/commit/8e7df670a883f6bc2f710ae0471514bd7413283b.

xerion3800, 2023-12-09:
> Es scheint als würde "Parkplatz" hier Ladeleistung von sich selbst bekommen und damit wunderbare Vermehrung der Energiemenge machen?

Das sollten wir als Patent anmelden wenn das funktioniert 😉