evcc-io/evcc #11062: Planner: add indicator for minimum required charge time

andig, 2023-12-09:
Trying to charge e.g. 100kWh within 2hrs is not possible. It might be helpful to give an indication, that a charging plan will not work as exptected:

[image]

@naltatis das hängt vllt. auch mit der Frage um, wie wir mit Plänen nicht erfüllbarer Dauer grundsätzlich umgehen wollen.

premultiply, 2023-12-10:
Es wäre ja möglich dafür sehr einfach eine (optimistischste) "minimale Ladezeit" zu berechnen die technisch eigentlich nicht unterschritten werden kann (vorausgesetzt alle Daten stimmen).

Eine Unterschreitung der berechneten "Normalladezeit" könnte (bzw. sollte) immer noch zu einer Warnung führen.

naltatis, 2023-12-10:
Ja, ein Warnungstext wäre sicher für das erste die einfachste Lösung. @andig Magst du in der Antwort vom Plan (`/api/loadpoints/1/plan`) nen Marker zurückliefern, der diesen Umstand abbildet? Alternativ können wir das auch rein UI-seitig lösen und den Endzeit des letzten Slots mit der aktuellen Zeit vergleichen.

closed 2023-12-19 by andig