evcc-io/evcc #11068: Amtron Compact 2.0s serial parameters

premultiply, 2023-12-10:
Hallo zusammen,

ich versuche gerade eine Amtron Compact 2.0s in EVCC einzubinden. Ist leider noch nicht funktionsfähig, aber vielleicht interessiert es ja jemanden und wir können uns weiterhelfen.

Im Moment hakt es daran, dass ich folgenden Fehler in EVCC erhalte, wenn ich das bestehende Template verwende:

`[main  ] FATAL 2023/12/10 14:51:57 cannot create charger 'wallbox5': cannot create charger 'template': cannot create charger 'mennekes': invalid comset: 8N2`

Wenn ich als Comset "8N1" verwende (entspricht aber nicht der Einstellung gemäß der Amtron Modbus Dokumentation ) startet EVCC zwar, aber es kommt keine Kommunikation über den Modbus zu Stande, Fehlermeldung:
```
[mennekes] ERROR 2023/12/10 14:54:10 heartbeat: serial: timeout
[mennekes] ERROR 2023/12/10 14:54:20 heartbeat: serial: timeout
```
Weiß hier jemand weiter? **Komisch ist ja bereits, dass EVCC das Comset 8N2 (d.h. mit zwei Stopp-Bits) von vornherein als ungültig zurückweist.**

Meine Installation sieht im Moment wie folgt aus:
In der Wallbox ist eine Hutschiene und Platz genug für einen Raspberry Zero samt Hutschienennetzteil. Der Raspberry ist im Haus-Wlan und mit der Wallbox soll er über Modbus RTU kommunizieren über einen USB Modbus Adapter. Auf dem Raspberry läuft EVCC. 

Jumper Einstellung auf S1 in der Wallbox: 
1-OFF (LED-Anzeige Farbschema), 
2-ON (Schieflastbegrenzung on), 
3-ON (=RFID on), 
4-ON (=Modbus on), 
5-ON (=Satellite), 
6-ON (=TQ EM 420 manager), 
7-ON (=Solar-Laden on), 
8-OFF

Neueste Mennekes Firmware (update mittels Konfigurationskabel) ist installiert.

Über den Modbus Adapter und Raspberry konnte ich bereits einen Eastron SDM630 Modbus V3 Zähler (auch in der Wallbox auf der Hutschiene) für EVCC auslesen -> das Modbus setup auf dem Raspi sollte also funktionieren. Bis die Kommunikation zur Mennekes Wallbox läuft, ist der Eastron Zähler aber vom Bus getrennt und in der evcc.yaml auskommentiert. Zähler und Wallbox müssten mit verschiedenen Baudraten angesprochen werden, so dass das evtl. auch Probleme geben könnte.

Als Modbusleitung verwende ich ein ca. 30cm langes Stück Cat 7 Kabel, wovon ich ein Aderpärchen verwende. Ich habe es mit und ohne Abschlusswiderstand auf der USB-Adapter-Seite versucht und auch mit und ohne GND Anschluss über über die Kabelschirmung verdrahtet versucht.

_Originally posted by @chriuso in https://github.com/evcc-io/evcc/discussions/9515#discussioncomment-7811835_

andig, 2023-12-10:
Also fehlt das Feature für 8N2? Das gibts bisher tatsächlich nicht.

andig, 2023-12-10:
Closed in https://github.com/evcc-io/evcc/commit/1a28912579f8570c1b1cc5fa270aaafe3321ebc5

chriuso, 2023-12-10:
oh - Danke für die schnelle Berücksichtigung! 
Tut mir leid wenn ich frage, aber ich kenne mich mit GitHub nicht so gut aus: kann ich 1a28912 schon verwenden und testen oder muss ich auf ein neues Release warten? Ich habe bisher evcc immer nur über apt und das Repository
```
curl -1sLf \
  'https://dl.cloudsmith.io/public/evcc/unstable/setup.deb.sh' \
  | sudo -E bash
```
installiert. Habe im Moment Version  0.122.1.

premultiply, 2023-12-10:
Musst für das neue Nightly Build bis morgen warten.

chriuso, 2023-12-11:
Heute konnte ich ein update per apt installieren. Sollte das 8N2 nun erkannt werden oder ist das noch nicht die neueste Version? Ich erhalte wieder den gleichen Fehler.

 ```
[main  ] INFO 2023/12/11 09:09:30 evcc 0.122.1 (3e1dcd34)
[main  ] INFO 2023/12/11 09:09:30 using config file: evcc.yaml
[main  ] INFO 2023/12/11 09:09:30 starting ui and api at :7070
[db    ] INFO 2023/12/11 09:09:32 using sqlite database: /root/.evcc/evcc.db
[main  ] FATAL 2023/12/11 09:09:33 cannot create charger 'wallbox5': cannot crea                                 te charger 'template': cannot create charger 'mennekes': invalid comset: 8N2
[main  ] FATAL 2023/12/11 09:09:33 will attempt restart in: 5m0s

```

andig, 2023-12-11:
Probiers bitte in 20min nochmal- da fehlte noch was, Build läuft.

chriuso, 2023-12-11:
Jetzt bekomme ich folgenden Fehler:
```
[main  ] INFO 2023/12/11 12:00:14 evcc 0.122.1 (855185b1)
[main  ] INFO 2023/12/11 12:00:14 using config file: evcc.yaml
[main  ] INFO 2023/12/11 12:00:14 starting ui and api at :7070
[db    ] INFO 2023/12/11 12:00:16 using sqlite database: /root/.evcc/evcc.db
[main  ] FATAL 2023/12/11 12:00:19 failed configuring loadpoints: failed configuring loadpoint: 1 error(s) decoding:

* '' has invalid keys: resetondisconnect
[main  ] FATAL 2023/12/11 12:00:19 will attempt restart in: 5m0s

```

chriuso, 2023-12-12:
Den obigen Fehler konnte ich beheben oder umgehen, indem ich Folgendes aus der evcc.yaml aaskommentiert habe:
``` #  resetOnDisconnect: false ```

Nun startet evcc mit comset 8N2 ohne Fehler, aber ich erhalte immer noch den timeout Fehler - siehe letzte Zeile:
```[main  ] INFO 2023/12/12 22:36:50 evcc 0.122.1 (855185b1)
[main  ] INFO 2023/12/12 22:36:50 using config file: evcc.yaml
[main  ] INFO 2023/12/12 22:36:50 starting ui and api at :7070
[db    ] INFO 2023/12/12 22:36:52 using sqlite database: /root/.evcc/evcc.db
[lp-1  ] WARN 2023/12/12 22:36:55 locking phase config to 3p for switchable charger
[lp-1  ] ERROR 2023/12/12 22:37:01 charge total import: serial: timeout
[site  ] INFO 2023/12/12 22:37:01 site config:
[site  ] INFO 2023/12/12 22:37:01   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/12/12 22:37:01     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/12/12 22:37:01     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/12/12 22:37:01   vehicles:
[site  ] INFO 2023/12/12 22:37:01     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2023/12/12 22:37:01 loadpoint 1:
[lp-1  ] INFO 2023/12/12 22:37:01   mode:        pv
[lp-1  ] INFO 2023/12/12 22:37:01   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/12/12 22:37:01   meters:      charge ✓
[lp-1  ] INFO 2023/12/12 22:37:01     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/12/12 22:37:01 phase timer inactive
[lp-1  ] DEBUG 2023/12/12 22:37:01 pv timer inactive
[lp-1  ] DEBUG 2023/12/12 22:37:01 guard timer inactive
[site  ] DEBUG 2023/12/12 22:37:01 ----
[lp-1  ] DEBUG 2023/12/12 22:37:01 charge power: 0W
[site  ] DEBUG 2023/12/12 22:37:01 pv power: 0W
[site  ] DEBUG 2023/12/12 22:37:01 grid power: 280W
[site  ] DEBUG 2023/12/12 22:37:01 grid powers: [60 10 200]W
[site  ] DEBUG 2023/12/12 22:37:01 grid currents: [0.51 0.11 1.06]A
[site  ] DEBUG 2023/12/12 22:37:01 site power: 280W
[lp-1  ] DEBUG 2023/12/12 22:37:01 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/12/12 22:37:01 charge total import: 1161664184.000kWh
[lp-1  ] ERROR 2023/12/12 22:37:06 charger: serial: timeout
[mennekes] ERROR 2023/12/12 22:37:28 heartbeat: serial: timeout
```
Wie könnte ich das noch detaillierter debuggen?

premultiply, 2023-12-12:
`evcc charger --log trace`

chriuso, 2023-12-13:
ok super. ich kopiere mal den output hier rein. Es sieht so aus als ob Daten empfangen werden. Werde mir bei Gelegenheit ansehen, ob man in der Mennekes Modbus API sich daraus einen Reim machen kann, falls nicht schon jemand vorher eine idee hat. Der Energy-Wert von 1161664184.0kWh ist offensichtlich falsch, falls es sich dabei um die insgesamt mit der Wallbox geladene Energiemenge handeln sollte. Das dürften grob nicht mehr als 15000kWh sein.

```
sudo evcc charger --log trace
[main  ] INFO 2023/12/13 08:07:40 evcc 0.122.1 (855185b1)
[main  ] INFO 2023/12/13 08:07:40 using config file: /home/pi/evcc.yaml
[db    ] INFO 2023/12/13 08:07:42 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2023/12/13 08:07:42 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%   device_id       %") -1 <nil>
[db    ] TRACE 2023/12/13 08:07:42 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[mennekes] TRACE 2023/12/13 08:07:42 modbus: send 32 03 05 12 00 02 61 01
[mennekes] TRACE 2023/12/13 08:07:42 modbus: recv 32 03 04 00 00 00 00 f9 30
[mennekes] TRACE 2023/12/13 08:07:42 modbus: send 32 03 10 00 00 02 c5 08
[mennekes] TRACE 2023/12/13 08:07:42 modbus: recv 32 03 04 45 3d 96 b8 12 22
[mennekes] TRACE 2023/12/13 08:07:42 modbus: send 32 03 05 00 00 06 c0 c7
[mennekes] TRACE 2023/12/13 08:07:42 modbus: recv 32 03 0c 00 00 00 00 00 00 00 00 00 00 00 00 e0 65
[mennekes] TRACE 2023/12/13 08:07:42 modbus: send 32 03 05 06 00 06 20 c6
[mennekes] TRACE 2023/12/13 08:07:42 modbus: recv 32 03 0c 00 00 00 00 00 00 00 00 00 00 00 00 e0 65
[mennekes] TRACE 2023/12/13 08:07:42 modbus: send 32 03 01 00 00 01 80 35
[mennekes] TRACE 2023/12/13 08:07:42 modbus: recv 32 03 02 00 01 7d 80
[mennekes] TRACE 2023/12/13 08:07:42 modbus: send 32 03 0d 05 00 01 93 64
[mennekes] TRACE 2023/12/13 08:07:42 modbus: recv 32 03 02 00 00 bc 40
Power:          0W
Energy:         1161664184.0kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 0V 0V 0V
Charge status:  A
Enabled:        false
```

Weiterhin: EVCC erkennt, wenn das Fahrzeug angeschlossen ist (weiß nicht ob die Info über das Fahrzeug oder die Wallbox bereitgestellt wird). Auch kann ich z.B. im Lademodus "Schnell" das prozentuale Ladeziel eingeben. EVCC meldet in der GUI: "Ladebereit. Warte auf Fahrzeug ...". Das Fahrzeug meldet jedoch den Fehler, dass keine Spannung bereitgestellt wird.

Hier der output von EVCC (Fahrzeugdaten sind als 'X' anonymisiert):
sudo evcc -c evcc.yaml
```
[main  ] INFO 2023/12/13 08:18:56 evcc 0.122.1 (855185b1)
[main  ] INFO 2023/12/13 08:18:56 using config file: evcc.yaml
[main  ] INFO 2023/12/13 08:18:56 starting ui and api at :7070
[db    ] INFO 2023/12/13 08:18:58 using sqlite database: /root/.evcc/evcc.db
[lp-1  ] WARN 2023/12/13 08:19:01 locking phase config to 3p for switchable charger
[lp-1  ] DEBUG 2023/12/13 08:19:01 charge total import: 1161664184.000kWh
[site  ] INFO 2023/12/13 08:19:01 site config:
[... cut ...]
[site  ] DEBUG 2023/12/13 08:20:07 pv power: 29W
[site  ] DEBUG 2023/12/13 08:20:07 grid power: 370W
[site  ] DEBUG 2023/12/13 08:20:07 grid powers: [200 10 150]W
[site  ] DEBUG 2023/12/13 08:20:07 grid currents: [1.33 0.41 0.95]A
[site  ] DEBUG 2023/12/13 08:20:07 site power: 370W
[lp-1  ] ERROR 2023/12/13 08:20:12 charge meter: serial: timeout
[lp-1  ] DEBUG 2023/12/13 08:20:12 charge total import: 1161664184.000kWh
[lp-1  ] DEBUG 2023/12/13 08:20:12 charger status: B
[mennekes] ERROR 2023/12/13 08:20:24 heartbeat: serial: timeout
```

andig, 2023-12-13:
Die Werte sind alle völlig falsch:

```
Energy:         1161664184.0kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 0V 0V 0V
```

chriuso, 2023-12-17:
Hallo Andig und premultiply,

Current und voltage könnte ja stimmen, wenn damit output-Werte gemeint sind und kein Ladevorgang läuft.

ich habe jetzt mal mit mbpoll (modbus simulator) ein bisschen gespielt und kann damit zumindest Integer register ausesen. Bei Float Registern habe ich es noch nicht geschafft. 
Ich konnte auch die Modbus-Datensatz Version der Wallbox auslesen - diese ist V 1.0.2.

Mit dem Aufruf unten sollte der Inhalt der Register 0x0D02-0xD06 (Cable Lock, Solar Charging Mode, Requested Phases, Charging Release energy Manager, Lock EVSE) zurückgegeben werden - das könnte auch stimmen. Machmal habe ich aber Probleme, Register auszulesen, bei denen 'Available at V.1.0.0' steht - evtl. kann man nicht automatisch davon ausgehen, dass alle Register aus V.1.0.0 auch in V.1.0.2 vorhanden sind.

Das heißt, dass die Kommunikation bei mir hardwareseitig wohl funktioniert und der Wurm vllt. irgendwo in der Programmierung der Kommunikation (mennekes.go?) liegt? 

Woran könnte es liegen, dass ein Auslesen des Registers 0x1000 (Energy) in evcc zwar funktioniert, aber ein falscher, viel zu hoher Wert herauskommt?

Weiterhin ist mir aufgefallen, dass als Heartbeat intervall in 'mennekes.go' 10s eingestellt sind, aber man könnte/sollte es vielleicht kürzer einstellen - z.B. alle 2s? In der Spezifikation ist beschrieben, dass **mindestens** alle 10s ein Heartbeat erwartet wird.

```
mbpoll -b 57600 -P none -s 2 -m rtu -a 50 -1 -r 3330 -t4 -c 5 /dev/ttyUSB0
mbpoll 1.0-0 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright © 2015-2019 Pascal JEAN, https://github.com/epsilonrt/mbpoll
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: Modbus RTU
Slave configuration...: address = [50]
                        start reference = 3330, count = 5
Communication.........: /dev/ttyUSB0,      57600-8N2
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 50...
[3330]:         0
[3331]:         3
[3332]:         1
[3333]:         0
[3334]:         0
```

PS: die Liste der Register gibt es z.B. hier https://api.library.loxone.com/downloader/file/1194/ModbusRTU_AmtronCompact2-0s_Description_v1-2.pdf - ist aber wahrscheinlich schon bekannt.

andig, 2023-12-18:
> Current und voltage könnte ja stimmen, wenn damit output-Werte gemeint sind und kein Ladevorgang läuft.

Die sind nicht gemeint.

> Woran könnte es liegen, dass ein Auslesen des Registers 0x1000 (Energy) in evcc zwar funktioniert, aber ein falscher, viel zu hoher Wert herauskommt?

Keine Ahnung. Du könntest die Modbuskommunikation von mbpoll und evcc vergleichen. Irgendwas ist bei Dir völlig krumm- da stimmt gar nichts.

> Weiterhin ist mir aufgefallen, dass als Heartbeat intervall in 'mennekes.go' 10s eingestellt sind, aber man könnte/sollte es vielleicht kürzer einstellen - z.B. alle 2s? In der Spezifikation ist beschrieben, dass mindestens alle 10s ein Heartbeat erwartet wird.

@premultiply dann sollten wir das auf 5s reduzieren um im Intervall zu bleiben. 

@chriuso das Auslesen sollte allerdings unabhängig davon funktionieren- das erklärt Deinen Fehler nicht. Im Zweifel müsstest Du uns mal Remotezugang geben um das analysieren zu können (info@evcc.io).

premultiply, 2023-12-18:
Beim Energie ist nur der Datentyp falsch.
Für den Rest braucht es Remotezugang, da die Doku offensichtlich nicht klar oder nicht fehlerfrei ist.

chriuso, 2023-12-18:
Vielen Dank, dass Ihr so reaktiv seid und bereit für Nachforschungen! 
Ich habe gerade eine e-mail geschickt.

Ich weiß leider im Moment auch nicht weiter. Ich habe nochmal versucht Float register auszulesen (mit mbpoll dann entsprechend -t4:float) habe es aber nicht hinbekommen. 

Was man auch beachten müsste: Big Endian Bytereihenfolge - hat auch nichts verändert bei mir mit mbpoll.

BMW999, 2023-12-19:
… plane mir ebenfalls eine Mennekes Compact zuzulegen und verfolge gespannt die Diskussion. Finde es toll, wie schnell das evcc Team reagiert.
👍 Wäre schön, wenn ich das Ergebnis des troubleshootings hier postet. Grüße BMW999

premultiply, 2023-12-19:
@chriuso Danke für den Zugang.
Ich versuche mir das im laufe des Nachmittags anzusehen.

closed 2023-12-19 by premultiply

chriuso, 2023-12-20:
Einfach Klasse! Vielen Dank, dass das so gut geklappt hat: Ich kann jetzt über das GUI den Ladevorgang starten und stoppen (vorausgesetzt man hat zuvor per RFID chip an der WB authorisiert). Ich konnte noch nicht mit Sonne testen, sondern nur Modus 'Schnell'. 

Den Mbdm Service habe ich jetzt gelöscht, so dass er nicht bei einem neustart wieder aktiv wird und dazwischenfunkt. 

Was mir aufgefallen ist: evtl. sind 1-phasiges und 3-phasiges Laden vertauscht? Gehe ich auf 1-phasiges Laden und starte, erhalte ich maximale Leistung, gehe ich auf 3-phasiges und starte, lädt er nur mit 33% Leistung. 
[image]
[image]

Zusätzliche Frage: Ist es möglich oder wird es möglich sein, den Zählerstand der Wallbox oder eines als Charge-Zähler konfigurierten Zählers in dem GUI anzuzeigen? 

Ich werde erst wieder am Wochenende testen können. Der Remotezugang ist zumindest bis dahin wieder abgestellt.

BMW999, 2024-01-04:
Hallo Leute, schönes Neues noch! 
Wollte mal hören, ob es hier noch weiter geht und/oder ob ihr den Status hier posten könntet.
Grüße BMW999

chriuso, 2024-01-08:
Hallo, 
Also es ist is soweit funktional. PV-geführtes Laden funkioniert an sich.

Was noch hakt:
a) 1-phasig / 3-phasig is vertauscht
b) automatische Phasenumschaltung habe ich bisher noch nicht beobachten können - entweder zu wenig Sonne (es kam maximal 5kWp bisher) oder aber aufgrund von a) wird das nicht richtig gesteuert
c) während des Ladens wird der Hausverbrauch nicht mehr angezeigt - er wird als 0W angezeigt, was nicht stimmt. Siehe z.B. Screenshot - es fehlen ca. 600W in der Bilanz. Wenn nicht geladen wird, wird der Verbrauch richtig angezeigt.
d) der Prozentanteil von Sonnenenergie an geladener Energie sowohl während des Ladens, als auch in der Übersicht der Ladevorgänge wird nicht richtig angezeigt. Es sind immer nur ganz niedrige Werte (0,1%) was nicht immer stimmen kann.
e) Die Ladepunktleistung wird um den Faktor 1000 zu hoch angezeigt.

c), d) und e) wären eher kosmetischer Natur; im Prinzip läuft es so schon.

Andig, premultiply, wenn es Euch interessiert und ihr mittels Zugang Euch das anschauen wollt, kann ich wieder einen aufmachen.
[image]

andig, 2024-01-08:
C/d sind entweder Folgefehler oder unabhängig hiervon. Der Rest lässt sich sicher schnell beheben.

premultiply, 2024-01-09:
1,4 MW sind ganz schön viel für so einen PKW... Kannst du mal schauen ob es draussen schon taghell leuchtet? :D

premultiply, 2024-01-09:
a) und e) mit https://github.com/evcc-io/evcc/commit/72b4a1d96d94c89979c36a57fd25953372c27cd0 umgedreht bzw. gefixed.
Somit sollte der Rest auch behoben sein.

chriuso, 2024-01-11:
Es gab zum Glück keine Supernova auf dem Stellplatz sondern es war in der Tat nur die Darstellung in der Software :-D

Jetzt konnte ich zumindest ohne Sonne testen: a) und c) und e) sind behoben. 
b) und d) werden dann wahrscheinlich auch behoben sein. Falls ich mich nicht mehr melde ist das dann auch so.
Großes Lob für diese wiederum schnelle Reaktion!!

Vielleicht noch eine generelle Information für Amtron Compact 2.0s Benutzer:
Die Jumper 6 und 7 auf der Bank S1 haben meines Erachtens keine Funktion, wenn man EVCC benutzt, da dann die Wallbox als Slave/Satellite arbeitet und komplett über EVCC gesteuert wird (-> Jumper 5 = On).

BMW999, 2024-01-11:
Schön zu verfolgen, wie hier zielstrebig eine Lösung herbeigeführt führt wurde. Ich bin beeindruckt. Eine Frage an die Entwickler (andig oder/und premultiply) bzgl. automatischer Phasenumschaltung. Würdet ihr bitte mal den genauen Ablauf der automatischen 1/3 Phasenumschaltung inkl. der dafür verwendeten Modbusregister beschreiben? … oder noch besser den Quell-Code verlinken? Würde mich sehr freuen…. Gruß BMW999

premultiply, 2024-01-16:
Klingt danach dass evcc nicht gestartet ist bzw. nicht kommunizieren kann.

BMW999, 2024-01-16:
Danke für die schnelle Antwort. EVCC ist tatsächlich noch nicht konfiguriert.
Da ich EVCC Anfänger bin, wollte ich zunächst sicherstellen, dass die Modbuskommunikation steht via TCP/RTU Gateway. Sollte das nicht gehen? Oder erwartet die Wallbox tatsächlich irgendein „Hallo hier bin ich - dein Master“ vom übergeordneten EMS?

chriuso, 2024-01-16:
Die Wallbox erwartet alle 10s einen Heartbeat von dem entsprechenden Master. Ich denke das ist das mindeste was sie erwartet und falls der nicht eintrifft wird ein Fehler ausgegeben.
Ich benutze kein TCP/RTU Gateway sondern einen USB auf Seriell Adapter - RS485 Konverter - FTDI FT232RL Chipsatz. Direkt an der USB Schnittstelle des Raspberry auf dem EVCC läuft und der sitzt in der Wallbox.

BMW999, 2024-01-16:
Danke für die Info. Ich dachte, man kann Modbus und  Satelite Mode schalten und dann die Register lesen/schreiben. In welchem Zustand war denn deine WB als du die Register 3330-3334 wie oben beschrieben ausgelesen hast? Ist da parallel ein heartbeat gelaufen?

chriuso, 2024-01-16:
Hm, stimmt, da lief evcc nicht gleichzeitig. Trotzdem würde ich mal versuchen evcc zu konfigurieren und auszuprobieren - das konfigurieren ist ziemlich straight forward.

Mein Mbpoll Aufruf samt Parameternl ist oben im Code quote zu sehen. Ich hatte damit aber wie beschrieben nicht so richtig Erfolg. 

Ich muss sowieso noch einmal in die Box und ein USB Kabel tauschen. Das werde ich morgen mal machen und die Jumper Stellungen sowie Firmware auslesen, damit hier zumindest keine Unterschiede sind.

BMW999, 2024-01-16:
Danke! Wenn ich die WB starte, rotiert das LED Feld für ca. 5min, danach geht die WB auf Störung (roter Schraubenschlüssel blinkt).
Wenn du deine WB ohnehin öffnest: Kannst du bitte mal die Modbus Verbindung kappen, die WB neu starten und prüfen, ob dein WB sich auch so verhält? Wäre sehr nett!!

andig, 2024-01-16:
Wollt ihr das mal bitte in einer Discussion besprechen? Der PR hier ist erledigt!