evcc-io/evcc #11069: Planner nimmt falsches maxcurrent zur Berechnung

VolkerK62, 2023-12-10:
Der Planner nimmt das maxcurrent vom Fahrzeug zur Berechnung der Ladedauer, auch wenn dieses größer ist, als das vom Loadpoint.
Ich würde erwarten, dass der kleiner Wert zur Berechnung herangezogen wird.

Gem. Issue #10330  scheint das so gewollt zu sein.
[image]

Erscheint mir aber nicht richtig zu sein.
Wurde im Issue auch diskutiert.

andig, 2023-12-10:
Damit lässt sich 1p realisieren. Es gibt aktuell keine schöne Lösung. Wenn jemand einen guten Vorschlag hat bitte explizit machen.

closed 2023-12-10 by andig

VolkerK62, 2023-12-10:
>Damit lässt sich 1p realisieren. 

Sorry, das versteh ich jetzt nicht.

andig, 2023-12-10:
Bei mehreren Fahrzeugen setzt Du am LP max 16A. Das 1p Fahrzeug dass 32A ziehen darf bekommt die entsprechende Konfiguration- und kann ja nicht 3p laden ;)

premultiply, 2023-12-10:
Und bei mehreren LPs mit unterschiedlichem maxCurrent? z.B. einer mit 3P 16A und einer mit 3P 32A?

Dazu der Beispielfuhrpark
- ZOE (3P 32A),
- Kona (1P 32A),
- Volkswagen (3P 16A)

Am Fahrzeug sollte ja schon dessen tatsächlichen Fähigkeiten eingetragen werden, oder?

andig, 2023-12-10:
Die Frage ist etwas dogmatisch ;) Mir gehts drum, wie wir die Szenarien abbilden können.

premultiply, 2023-12-10:
Mir doch auch. Ich denke aber dass wir hier derzeit noch eine konzeptionelle Lücke in bestimmten Szenarien haben könnten.

andig, 2023-12-10:
Isso. Der aktuelle Status entspricht i.w. dem vor dem Umbau, nur klarer und flexibler implementiert.

andig, 2023-12-11:
Ich möchte Euch mal daran erinnern, dass wir hier in einem geschlossenen Issue diskutieren.

Davon ab brauchen wir- egal welche Lösung- ein Verständnis, wie die Lösung die Szenarien aus https://github.com/evcc-io/evcc/issues/10330#issuecomment-1765203043 ermöglicht? Darauf sehe ich weiterhin keine Antwort ;)