evcc-io/evcc #11088: evcc fails to start with charger template "hardybarth-salia" if grid_current_limit is set to non-integer value

lorenzzhao, 2023-12-12:
### Describe the bug

If the grid_current_limit value of a charger using template "hardybarth-salia" (in my case I have the "eCHARGE cPµ2 - PRO Wallbox" connected) is set to a non-integer value, e. g. "9.2", evcc fails to start with the error message: 

"cannot unmarshal number 9.2 into Go struct field Port.Secc.Port0.grid_current_limit of type int"

### Steps to reproduce

1. Configure charger using template "hardybarth-salia" 
2. Set grid_current_limit to "9.2", e. g. using MQTT topic "port0/grid_current_limit"
3. start evcc

### Configuration details

```yaml
chargers:
  - name: wallbert
    type: template
    template: hardybarth-salia
    host: 192.168.178.16
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/12/12 10:03:17 evcc 0.122.1
[main  ] INFO 2023/12/12 10:03:17 using config file: /config/evcc.yaml
[main  ] INFO 2023/12/12 10:03:17 starting ui and api at :7070
[main  ] FATAL 2023/12/12 10:03:18 cannot create charger 'wallbert': cannot create charger 'template': cannot create charger 'hardybarth-salia': json: cannot unmarshal number 9.2 into Go struct field Port.Secc.Port0.grid_current_limit of type int
[main  ] FATAL 2023/12/12 10:03:18 will attempt restart in: 5m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.122.1

mdkeil, 2023-12-12:
Did you try setting the grid-current-limit via web-ui..? I think it will not work, because I think a float-value is not supported.. therefor the template expected an int-value.

andig, 2023-12-13:
Closed in https://github.com/evcc-io/evcc/commit/3499d99e5893ce65f4c3b60b9aedfb402bb877d6. This should accept float limits when unmarshaling, but we would still only set integer limits. If setting float limits is supported please let us know.

closed 2023-12-13 by andig

lorenzzhao, 2023-12-16:
> Did you try setting the grid-current-limit via web-ui..? I think it will not work, because I think a float-value is not supported.. therefor the template expects an int-value.

Maybe you are right, anyway, trying to start evcc after setting a non-integer value via MQTT is still a valid use case IMO. So this scenario should be supported.