evcc-io/evcc #11094: eGolf connection does not work anymore

AJ1979, 2023-12-13:
### Describe the bug

Seitdem VW von der WeConnect App auf die Volkswagen App umgestellt hat, kann evcc keine Verbindung zum VW Server mehr herstellen und das Auto wird in evcc als offline angezeigt.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: eGolf
    type: VW
    title: eGolf
    capacity: 30 # kWh
    user: 
    password:
    vin: WVWZZZAUZJ8903623
    onIdentify: # set defaults when vehicle is identified
      minSoC: 20
      targetSoC: 70
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/12/13 07:07:35 evcc 0.122.1
[main  ] INFO 2023/12/13 07:07:35 using config file: /config/evcc.yaml
[main  ] INFO 2023/12/13 07:07:35 starting ui and api at :7070
[db    ] INFO 2023/12/13 07:07:35 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2023/12/13 07:07:35 connecting evcc-1336263846 at tcp://localhost:1883
[mqtt  ] DEBUG 2023/12/13 07:07:35 tcp://localhost:1883 connected
[main  ] ERROR 2023/12/13 07:07:36 creating vehicle eGolf failed: cannot create vehicle 'VW': unexpected status: 401 (Unauthorized)
[lp-1  ] WARN 2023/12/13 07:07:36 poll mode '{always 1h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.122.1

oberschmuh, 2023-12-13:
Schau mal hier: [https://github.com/evcc-io/evcc/discussions/10820#discussioncomment-7605416](url)

AJ1979, 2023-12-13:
Danke, das wars -> funktioniert wieder!

StefanSchoof, 2023-12-13:
Dann schließe das Issue bitte. Das Team muss ja nicht unnötig mit diesem Issue beschäftigen.

closed 2023-12-13 by AJ1979