evcc-io/evcc #11095: Sungrow Modbus down

xerion3800, 2023-12-13:
### Describe the bug

Seit dem heutigen Nightly Update bekommt EVCC keine Verbindung mehr per Modbus zu den Sungrow Komponenten

### Steps to reproduce

1. Update auf Nightly https://github.com/evcc-io/evcc/commit/855185b1 durchführen
2. Sungrow Komponenten laut config anlegen
3. EVCC starten und Log beobachten
4. Vermutlich hat das mit dem Commit zu tun? https://github.com/evcc-io/evcc/commit/855185b186c8ea6421f40ed236d4471360fddecb

### Configuration details

```yaml
network:
  schema: http
  host: 192.168.178.191
  port: 7070

log: debug

levels:
  # go: trace
  site: debug
  lp-1: debug
  lp-2: debug
  # mqtt: trace
  # modbus: trace

  
interval: 10s # control cycle interval

plant: ############

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: ###########

meters:
  - name: grid
    type: template
    template: sungrow-hybrid
    usage: grid     
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.141 # Hostname
    port: 502 # Port 
  - name: sungrow
    type: template
    template: sungrow-hybrid
    usage: pv     
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.141 # Hostname
    port: 502 # Port 

  - name: plenticore
    type: template
    template: kostal-plenticore
    usage: pv      
    
    # Modbus TCP
    modbus: tcpip
    id: 71
    host: 192.168.178.199 # Hostname
    port: 1502 # Port 
 
[... 649 lines cut ...]
[site  ] DEBUG 2023/12/13 07:58:34 battery power: 0W
[site  ] ERROR 2023/12/13 07:58:35 grid meter: read failed: read tcp 172.17.0.3:36380->192.168.178.141:502: read: connection reset by peer
[site  ] ERROR 2023/12/13 07:58:35 grid meter: read failed: read tcp 172.17.0.3:36380->192.168.178.141:502: read: connection reset by peer
[site  ] DEBUG 2023/12/13 07:58:43 ----
[lp-1  ] DEBUG 2023/12/13 07:58:43 charge power: 0W
[lp-2  ] DEBUG 2023/12/13 07:58:43 charge power: 0W
[lp-3  ] DEBUG 2023/12/13 07:58:43 charge power: 0W
[lp-4  ] DEBUG 2023/12/13 07:58:43 charge power: 14W
[site  ] ERROR 2023/12/13 07:58:44 pv 1 power: read failed: read tcp 172.17.0.3:47736->192.168.178.141:502: read: connection reset by peer
[site  ] DEBUG 2023/12/13 07:58:44 pv power: 0W
[site  ] ERROR 2023/12/13 07:58:44 battery 1 power: mul[0]: read failed: read tcp 172.17.0.3:47764->192.168.178.141:502: read: connection reset by peer
[site  ] ERROR 2023/12/13 07:58:44 battery 1 soc: read failed: read tcp 172.17.0.3:47776->192.168.178.141:502: read: connection reset by peer
[site  ] DEBUG 2023/12/13 07:58:44 battery soc: 0%
[site  ] DEBUG 2023/12/13 07:58:44 battery power: 0W
[site  ] ERROR 2023/12/13 07:58:45 grid meter: read failed: read tcp 172.17.0.3:47806->192.168.178.141:502: read: connection reset by peer
[site  ] ERROR 2023/12/13 07:58:45 grid meter: read failed: read tcp 172.17.0.3:47806->192.168.178.141:502: read: connection reset by peer
```

### What type of operating system are you running?

Docker container

### Version

evcc 0.122.1 (855185b1)

naltatis, 2023-12-13:
Ich kann das nicht reproduzieren. Mit meinem Sungrow SH10RT läuft das auch mit dem aktuellen Nightly.

xerion3800, 2023-12-13:
Komisch, habe nur ein Update gemacht und danach war das Problem vorhanden. Über mein SmartHome System kann ich nach wie vor die WR erreichen. Ich schaue mal.

xerion3800, 2023-12-13:
Zufälle gibt es. Habe gerade den Netzwerk Anschluss einmal neu verbunden und es geht wieder. Vorher bis zum Update lief das ohne Probleme. Dann habe ich auf das aktuelle Nightly upgedatet und in den commits fand ich dann auch noch Modbus Änderungen....also lag das für mich sehr nahe zusammen. 
Aber jetzt läuft es wieder sorry für die Verwirrung.

closed 2023-12-13 by xerion3800

premultiply, 2023-12-13:
172.17.0.3:47806->192.168.178.141:502

Fragwürdige IP-Konfiguration.

xerion3800, 2023-12-13:
Läuft bei mir in Docker Container, hängt das damit vielleicht zusammen?